FormerLurker/Octolapse #363: 1060 layer print, 81 snapshots only

tvl83, 2019-07-06:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Just started the print

#### What should have happened?
Created a timelapse of the print taking a snapshot every layer

#### What happened instead?
created a timelapse of a print with 1065 layers and only took 81 snapshots

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Original Prusa MK3
Printer Firmware Version: 3.7.2-2363

#### Browser and version of browser, operating system running browser

Browser: Chrome 75.0.3770.100
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/tvl83/53955fd82095964c38e96180709ea832

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/tvl83/162af3cbd8f3d3019973a3b74c6d7d9d

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://drive.google.com/file/d/1JlZjsvYVlScvwVndIiFJXiZDk-Sr7U7O/view?usp=sharing

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/tvl83/33c4f9eb7116dd72b04825f6721ebcac

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://drive.google.com/open?id=1dlz6VvkZ0P9wcBcPN2xaV5SnELzukGsb

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

---------------

Thank you for your help.

Video is upside down because I didn't have the right settings.

FormerLurker, 2019-07-06:
Thanks for filling out the ticket so well!  Ill get on it tomorrow.  Is probably an easy fix.

FormerLurker, 2019-07-14:
OMG, sorry about this taking so long!  I solved an extremely similar issue for someone else, and conflated the two!

The fix for this is to switch to the layer change - compatibility snapshot profile.  You are hitting the 'minimum layer time' based on your slicer settings, which slows down the print speed, and Octolapse can't figure out what is printing (infill, interior/exterior perimeters, etc), and decides not to take any snapshots due to the 'high quality' settings.  This is known as feature detection, and you'll see that it's enabled within your snapshot profile.

Fyi, this has been removed from the next release for exactly this reason.  The smart layer trigger has similar functionality, and is actually reliable.

Again, SO SORRY for the delay!

FormerLurker, 2019-07-23:
So, I think this one is solved.  Can you confirm?

tvl83, 2019-07-23:
Yea, i havent had issues since then and using the every .2mm setting it works perfectly. 

thanks for your reponse and advice. sorry for late reply. i'll close it :)

closed 2019-07-23 by tvl83

FormerLurker, 2019-07-23:
No problem, thanks!  Feel free to add another issue if you ever have any more trouble!