FormerLurker/Octolapse #364: gcode doesn't trigger snapshot

rossome2648, 2019-07-13:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OctoPrint: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

I changed the settings to get my Raspi webcam to work and now Gcode doesn't take the snapshot. I have tried multiple prints. It takes a snapshot after the skirt for some reason but the rest of the time, it acts like the snapshot gcode isn't there. I have tried both snap and G4 P1 with the same results.
I also changed it to Layer Change Compatibility and it does take snapshots. It doesn't come out the greatest as the extruder will not be in the same spot - I have it to center left but some snaps have the extruder right above the print.

Layer Change High Quality was being used but was doing the same as gcode.

#### What should have happened?
Snapshot on gcode trigger

#### What happened instead?
No snapshot on gcode trigger - only after skirt

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: CR-10S Pro
Printer Firmware Version: TM3D B7

#### Browser and version of browser, operating system running browser

Browser: Version 75.0.3770.100
Browser OS: Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/rossome2648/9fd47ca9fac2a4d4841c675d6f526480

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/rossome2648/9fd47ca9fac2a4d4841c675d6f526480

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Completed timelapse with Layer Change Compatibility - can see extruder in wrong position
Screenshot/Video Links: https://drive.google.com/open?id=1bpBcZQmV1TBZs5dIDOMSeQrsBJuk-SIf

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-07-13:
Hey, it looks like you posted the gcode twice by mistake.  Can you upload the settings.json file again?  Thanks!

rossome2648, 2019-07-13:
Hm, I thought I added them both to that link. It said add file!
Here is the json:
https://gist.github.com/rossome2648/e1fb87e8acf47d08561add6e2830dc2d

FormerLurker, 2019-07-13:
LOL!  You did, it was just WAY WAY at the bottom.

FormerLurker, 2019-07-13:
So, it's the issue I initially suspected.  Edit your 'Gcode trigger' and turn off feature detection.

Also, I noticed G4 P1 custom snapshot command in your printer profile.  I probably know where you got that from, perhaps a certain youtube tutorial :)

Feature detection has been removed in the next version, so I won't bother debugging why it didn't work (probably because you are running into the minimum layer time, and your slicer is slowing down the layers).  I REALLY should have prevented that setting from working with the gcode trigger, but for some reason I didn't think about it.  It doesn't make much sense to have a gcode trigger that won't take a snapshot right where the gcode is placed, does it?

Anyway, I hope this solves your issue!  Let me know how it goes.

rossome2648, 2019-07-13:
Haha, yup! From his tutorial. I go back and forth between snap and G4 P1 to see which one works better.

And it seems to be working with Feature detection turned off. I thought about turning that off but didn't think it had much to do with it. Being in tech, I should know to always undo what I did before it stopped working.

Now, I just need to get the timing right for the snaps. And figure out why my camera is zoomed in even though zoom is set to 0.

Thanks for your help!

FormerLurker, 2019-07-13:
Your pi cam is probably zoomed in because of image stabilization.  I've been pulling my hair out over that dumb check box all day (writing a custom control page for the PI, since it looks better than the default).  It will turn on and zoom in, but it won't turn off.  I have found one way to disable it:

1.  Uncheck image stabilization.
2.  Click on either of the flip check boxes
3.  Click on the flip check box again to undo the changes.

Viola!

Regarding feature detection, I spent probably an entire month trying to get that working.  I thought I was successful, but turns out the slicers have so many features that were thwarting my plans (for example, detecting when infill is being printed by watching the extrusion speed/print speed).  The minimum layer time is the one I could never fix.. :(

rossome2648, 2019-07-13:
We're glad that we have this awesome plugin! And it's fairly easy to get it working correctly. We can't expect for you do to everything for us! Haha.
Keep up the great work!

closed 2019-07-13 by rossome2648

FormerLurker, 2019-07-13:
Let me know if that image stabilization trick worked.  I'm trying to figure out how to make every setting work for the raspi cam.  Some of the controls are connected, meaning one drop down or check box might need to be set before another control works (auto exposure and the exposure settings drop down, for example).  I'm probably going to post a 'help wanted' issue for that.

rossome2648, 2019-07-13:
It doesn't look like that helped. Still looks zoomed in. I even checked image stabilization, unchecked then did the trick and it still seems zoomed in. I don't believe I moved the camera but I'll mess with that and see if that could be the issue.