FormerLurker/Octolapse #365: *** Error (-60: 'Could not lock the device') ***

ffried78, 2019-07-18:
#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4rc1.dev3

#### Version of OctoPrint

Latest

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

I am using a Sony A6000 with Octolapse.

Scenario 1 - On the most successful attempt:
1. Launch octolapse
2. One picture gets taken
3. When the second picture comes, the printer times-out.

Scenario 2 - All other attempts.
1. Launch octolapse.
2. The camera is triggered on every layer but the shutter cannot be heard. The following error pops up on every shot: *** Error (-60: 'Could not lock the device') ***
3. The print keeps running fine, every layer the error pops up again.
note: restarting the pi and camera will occasionally allow the camera to take one picture as in Scenario 1 above.

#### What should have happened?
Run Octolapse as normal with my Sony A6000.

#### What happened instead?
The best attempts have given me one picture. All other attempts show the error code *** Error (-60: 'Could not lock the device') ***

#### Operating System running OctoPrint and Octolapse

OS Name: Mac OS
Os Version: Mojave 10.14.2

#### Printer model & used firmware incl. version

Printer Model: MK3
Printer Firmware Version: 

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Mac

#### Link to the gcode file you were printing when the problem occurred

----Start g-code on mK3 ----
M83  ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G90
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
M221 S{if layer_height==0.05}100{else}95{endif}

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-07-18:
Is your Sony A6000 one of those new mirror-less cameras?  I've gotten lots of reports of issues using the newer sony models with ghpoto2.  They probably have unique firmware, and the gphoto2 dev hasn't had time to get everything working (or it's not possible).  I have found that installing the latest version (not the apt-get version) seems to fix some of the issues.  Take a look at [this updater](https://github.com/gonzalo/gphoto2-updater).  OctoPi changes the behavior of git somewhat, so a slight change to the script may be required.  I will hunt around because I think I've got a gist somewhere with a modified version that works on OctoPi.

FormerLurker, 2019-07-18:
I should have looked first, but I expected some trouble finding the script.  [Here is the gist](https://gist.github.com/FormerLurker/d379cf5bd520b63fba26bd2f9846c7a5).  You just need to create a .sh file on your pi, grant execute permissions (just like in the DSLR guide) and run the script with sudo.  You can install the latest release or dev version by providing parameters.  There is even integrated help.  

I hope that helps.  Let me know if you get it working and how so I can add a guide for the sony cams.  That should save folks future frustrations.

FormerLurker, 2019-10-07:
Going to go ahead and close this down.  Please post a new issue if you'd like me to reexamine it.  Thanks for your post!

closed 2019-10-07 by FormerLurker