FormerLurker/Octolapse #369: In test mode works fine but not when printing

Caballete007, 2019-08-08:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-08-08:
I'll need additional info to solve your issue.  Please fill in the form above.

Caballete007, 2019-08-09:
Hi
I already submit the information. If you need something else let me know. Have a great day.

FormerLurker, 2019-08-10:
HI!

Your printer profile settings look fine, fyi.  I see you added G90 to the top of your gcode at some point, that was a good call.

There were some really odd errors about threading exceptions in your log.  It might be a good idea to disable your other plugins while running a test of Octolapse (in live mode, not test mode).  If it works you can re-enable the other plugins and see if you have any issues with live print then.

However, what I see mostly where you cancel prints in the log (presumably because it's stalled out) is that your printer is reporting communication problems (I think):

```
2019-08-07 18:59:56,631 Queuing Command: Command Type:None, gcode:M400, cmd: M400, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-08-07 18:59:56,633 Queuing Command: Command Type:None, gcode:M114, cmd: M114, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-08-07 18:59:56,774 Received from printer: line:  *******RTS_HandleData********
2019-08-07 18:59:56,775 Received from printer: line: 
2019-08-07 18:59:56,784 Received from printer: line:
2019-08-07 18:59:56,785 Received from printer: line:   recdat.data[0] ==4
2019-08-07 18:59:57,274 Queuing Command: Command Type:temperature_poll, gcode:M105, cmd: M105, tags: set(['trigger:comm.poll_temperature'])
2019-08-07 18:59:57,539 Received from printer: line:   recdat.addr ==4098== Checkkey==0  *******RTS_HandleData********
2019-08-07 18:59:57,541 Received from printer: line: 
2019-08-07 18:59:57,552 Received from printer: line:
2019-08-07 18:59:57,559 Received from printer: line:   recdat.data[0] ==3
2019-08-07 18:59:58,524 Received from printer: line:   recdat.addr ==4158== Checkkey==11echo:busy: processing
2019-08-07 19:00:00,338 Received from printer: line:  *******RTS_HandleData********
2019-08-07 19:00:00,347 Received from printer: line: 
2019-08-07 19:00:00,355 Received from printer: line:
2019-08-07 19:00:00,375 Received from printer: line:   recdat.data[0] ==102
2019-08-07 19:00:00,614 Received from printer: line:   recdat.addr ==4172== Checkkey==15  *******RTS_HandleData********
2019-08-07 19:00:00,624 Received from printer: line: 
2019-08-07 19:00:00,635 Received from printer: line:
2019-08-07 19:00:00,652 Received from printer: line:   recdat.data[0] ==202
2019-08-07 19:00:00,654 Received from printer: line:   recdat.addr ==4172== Checkkey==15echo:busy: processing
2019-08-07 19:00:00,893 Received from printer: line:  *******RTS_HandleData********
2019-08-07 19:00:00,896 Received from printer: line: 
2019-08-07 19:00:00,900 Received from printer: line:
2019-08-07 19:00:00,906 Received from printer: line:   recdat.data[0] ==302
2019-08-07 19:00:01,330 Received from printer: line:   recdat.addr ==4172== Checkkey==15  *******RTS_HandleData********
2019-08-07 19:00:01,335 Received from printer: line: 
2019-08-07 19:00:01,339 Received from printer: line:
2019-08-07 19:00:01,350 Received from printer: line:   recdat.data[0] ==402
```
I've never seen output quite like that before.  I suspect this is something specific to the B7 firmware you installed, but that's really just a guess.  Whatever it's doing there it took quite a while to complete.  Note that when this happened, Octolapse was sending some snapshot movements, but nothing unusual at all.  Then later during the same snapshot process this starts to happen:

```
2019-08-07 19:00:02,609 Sent to printer: Command Type:None, gcode:M400, cmd: M400, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-08-07 19:00:04,606 Received from printer: line:echo:busy: processing
2019-08-07 19:00:06,603 Received from printer: line:echo:busy: processing
2019-08-07 19:00:08,605 Received from printer: line:echo:busy: processing
2019-08-07 19:00:10,095 Received from printer: line:  *******RTS_HandleData********
2019-08-07 19:00:10,110 Received from printer: line: 
2019-08-07 19:00:10,121 Received from printer: line:
2019-08-07 19:00:10,131 Received from printer: line:   recdat.data[0] ==3
2019-08-07 19:00:10,138 Received from printer: line:   recdat.addr ==4166== Checkkey==15echo:enqueueing "G28"
2019-08-07 19:00:10,146 Received from printer: line:echo:enqueueing "G1 F100 Z10.2"
2019-08-07 19:00:10,152 Received from printer: line:
2019-08-07 19:00:10,163 Received from printer: line: RTSUpdate Waitway4
2019-08-07 19:00:10,171 Received from printer: line: RTSUpdate Waitway4
2019-08-07 19:00:10,180 Received from printer: line: RTSUpdate Waitway4
2019-08-07 19:00:10,191 Received from printer: line: RTSUpdate Waitway4
```
That is REALLY REALLY strange, and it continues with the Waitway4 stuff for many seconds until you canceled the print (it would have been interesting to see how long that would have gone, and what the printer would have reported had you NOT cancelled.  Might be worth trying again).  

So, as you can see from the log above, Octolapse sends an M400 (wait for all movements to finish), which it uses to determine when the printer is in position, then some normal looking 'busy: processing' messages are sent, then some odd stuff after.  Do you notice the ```echo:enqueueing "G1 F100 Z10.2"```?  That command is NOT in your gcode file and was NOT generated by Octolapse.  Did that come from Octoprint somehow or from the control panel on your printer?  I've looked at a lot of logfiles in my day, and have never seen anything like this.

Much later in the log I see this:
```
[... cut ...]
I'm not quite sure where to go from here.  Most of the snapshots I see in the file look normal enough (though it's hard to tell since I'm not sure what the gcode queue within the printer looks like when Octolapse starts), but a few are really wacky.  Do you have linear advance enabled?  I do see that most of the crazy stuff happens after M400 is sent, and on my MK2 I occasionally see longer than usual pauses with linear advance enabled.  I didn't see it in your gcode, but maybe it is defaulted somewhere.

Caballete007, 2019-08-10:
Hi
I'm going to run a live mode printing without the plugins now. where I can find if I have enable linear advance? "Did that come from Octoprint somehow or from the control panel on your printer?" how I look for this? Thank you

FormerLurker, 2019-08-10:
It is controlled via firmware or gcode.  You can add a ```M900 K0``` command to your start gcode to ensure it is off.

Caballete007, 2019-08-10:
Already start the live printing and add the commands please take a look at the picture and let me know if it's ok . Still doing the same if you want I will finish it and send you the log.  Thank you

https://github.com/Caballete007/In-test-mode-works-fine-but-not-when-printing-369

Caballete007, 2019-08-10:
[image]

FormerLurker, 2019-08-10:
Yes, it looks good.  Just make sure you allow the print to continue through a couple of these long delays.

Caballete007, 2019-08-10:
Send: N1505 G1 X146.701 Y149.195 E28.38126*99
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok N1499 P0 B0
Send: N1506 G1 X146.988 Y148.435 E28.40693*103
Recv: ok N1500 P0 B0
Send: N1507 G1 X147.455 Y147.755 E28.43299*109
Recv: ok N1501 P0 B0
Send: N1508 M105*27
Recv: ok N1502 P0 B0
Send: N1509 G1 X148.062 Y147.215 E28.45866*97
Recv: ok N1503 P0 B0
Send: N1510 G1 X148.789 Y146.83 E28.48466*86
Recv: ok N1504 P0 B0
Send: N1511 G1 X149.58 Y146.633 E28.51041*89
Recv: ok N1505 P0 B0
Send: N1512 G1 X150.4 Y146.629 E28.53632*96
Recv: ok N1506 P0 B0
Send: N1513 G1 X151.194 Y146.821 E28.56213*108
Recv: ok N1507 P0 B0
Send: N1514 G1 X151.92 Y147.201 E28.58803*80
Recv: ok T:210.00 /210.00 B:70.00 /70.00 @:67 B@:30
Send: N1515 M204 S2000*100
Recv: ok N1509 P0 B0
Send: N1516 M205 X20 Y20*22
Recv: ok N1510 P0 B0
Send: N1517 G0 F7200 X152.146 Y146.867*118
Recv: ok N1511 P0 B0
Send: N1518 M204 S1500*111
Recv: ok N1512 P0 B0
Send: N1519 M205 X10 Y10*25
Recv: ok N1513 P0 B0
Send: N1520 G1 F600 X152.834 Y147.473 E28.617*53
Recv: ok N1514 P0 B0
Send: N1521 G1 X153.356 Y148.223 E28.64587*110
Recv: ok N1515 P0 B0
Send: N1522 G1 X153.684 Y149.082 E28.67492*105
Recv: ok N1516 P0 B0
Send: N1523 G1 X153.797 Y149.986 E28.70371*106
Recv: ok N1517 P0 B0
Send: N1524 G1 X153.688 Y150.91 E28.73311*80
Recv: ok N1518 P0 B0
Send: N1525 G1 X153.364 Y151.763 E28.76194*103
Recv: ok N1519 P0 B1
Send: N1526 G1 X152.848 Y152.511 E28.79065*100
Recv: ok N1520 P0 B0
Send: N1527 G1 X152.169 Y153.116 E28.81938*107
Recv: ok N1521 P0 B0
Send: N1528 G1 X151.344 Y153.552 E28.84887*110
Recv: ok N1522 P0 B0
Send: N1529 G1 X150.458 Y153.771 E28.87771*98
Recv: ok N1523 P0 B0
Send: N1530 G1 X149.552 Y153.771 E28.90633*104
Recv: ok N1524 P0 B0
Send: N1531 G1 X148.654 Y153.552 E28.93554*111
Recv: ok N1525 P0 B0
Send: N1532 G1 X147.844 Y153.127 E28.96444*111
Recv: ok N1526 P0 B0
Send: N1533 G1 X147.165 Y152.525 E28.99311*107
Recv: ok N1527 P0 B0
Send: N1534 G1 X146.644 Y151.776 E29.02194*98
Recv: ok N1528 P0 B0
[... 541 lines cut ...]
Send: N1797 G1 X147.345 Y148.622 E34.33922*111
Recv: ok N1795 P0 B1
Send: N1798 G1 X147.752 Y148.027 E34.362*111
Recv: ok N1796 P0 B1
Send: N1799 G1 X148.289 Y147.548 E34.38473*109
Recv: ok N1797 P0 B1
Send: N1800 G1 X148.928 Y147.21 E34.40758*93
Recv: ok N1798 P0 B1
Send: N1801 G1 X149.628 Y147.034 E34.43038*100
Recv: ok N1799 P0 B1
Send: N1802 G1 X150.348 Y147.029 E34.45313*108
Recv: ok N1800 P0 B1
Send: N1803 G1 X151.051 Y147.199 E34.47599*107
Recv: ok N1801 P0 B1
Send: N1804 G1 X151.685 Y147.528 E34.49855*110
Recv: ok N1802 P0 B1
Send: N1805 M204 S2000*104
Recv: ok N1803 P0 B1
Send: N1806 M205 X20 Y20*26
Recv: ok N1804 P0 B1
Send: N1807 G0 F7200 X151.912 Y147.195*117
Recv: ok N1805 P0 B2
Send: N1808 M204 S1500*99
Recv: ok N1806 P0 B3
Send: N1809 M205 X10 Y10*21

Caballete007, 2019-08-10:
I sent you another log file. 9/10

Caballete007, 2019-08-11:
Hi

The instructions that I follow for my Octolapse setup including a few things from your video and comments were from this video: https://youtu.be/CjOIxKxb3h8.
Maybe you saw it before. 
There is something that I added following his instructions that I have not seen in other videos and comments and is that I added in the script of my printer this: 
**"Change the snapshot command to ‘G4 P1’ in the printer profile. Add this to the layer change script in Simplify3D or the After layer change gcode in Slic3r. For Cura, it is more complicated:
A- Extensions - Post Processing - Modify G-code
B- Add a script - Search and replace
C- Then enter:
(;LAYER:).{1,}
G4 P1
Tick ‘Úse Regular Expressions’"**

Under the video he wrote step by step the instructions and settings that we should use.

FormerLurker, 2019-08-11:
I just sat down to look at the log you uploaded yesterday (plugin_octolapse.log), but noticed that logging seemed to be turned off :(

I hate to ask, but can you try it again, but this time make sure you use the full diagnostic live print debug profile?  Something very small would suffice (a calibration cube or something like that).

Regarding the instructions, in the log I reviewed you were using the layer trigger, so the above instructions wouldn't apply.  However, I do see the G4 P1 command.  On my printer this would be ok (it would pause for 1MS.  Not good.. not terrible :)

Maybe remove that gcode from your layer change script too, just in case.  I'm not sure how your firmware is configured (and don't really know that much about it to be honest...)

If we can't figure anything out, I have one more idea:  We could manually add M400 and M114 commands to your layer change script, run the gcode without Octolapse, and see if your printer has the same problems.  I'll wait to see your logfile first, but if neither of these work I suppose we could try upgrading to the Alpha version of Octolapse (v0.4) and see if that helps.  However, I'm reasonably sure it has something to do with the firmware, but I could be wrong.

Caballete007, 2019-08-11:
I just sat down to look at the log you uploaded yesterday (plugin_octolapse.log), but noticed that logging seemed to be turned off :(

How I turn this on?

FormerLurker, 2019-08-11:
Set your debug profile in Octolapse to 'Live Print - Full Diagnostic' (or whatever is the closest, I don't have an installed version of 0.3.4 in front of me).  It was definitely turned on during the first log you uploaded.

Caballete007, 2019-08-11:
2019-08-11 18:52:27,739 Print Start Detected.  Command: M110 N0, Tags:set(['trigger:comm.start_print', 'source:job', 'trigger:comm.reset_line_numbers'])
2019-08-11 18:52:27,740 Print start detected, attempting to start timelapse.
2019-08-11 18:52:27,759 Printer event received:PrinterStateChanged.
2019-08-11 18:52:27,819 Printer event received:PrintStarted.
2019-08-11 18:52:28,073 Camera settings applied without error.
2019-08-11 18:52:28,075 Camera settings applied without error.
2019-08-11 18:52:28,092 Extruder Triggers - OnExtrudingStart:True, OnExtruding:True, OnPrimed:True, OnRetractingStart:None OnRetracting:None, OnPartiallyRetracted:False, OnRetracted:True, ONDetractingStart:False, OnDetracting:False, OnDetracted:False
2019-08-11 18:52:28,101 Creating Layer Trigger - TriggerHeight:0.4 (none = layer change), RequiresZHop:False
2019-08-11 18:52:28,102 Print Started - Timelapse Started.
2019-08-11 18:52:28,112 Queuing Command: Command Type:None, gcode:M110, cmd: M110 N0, tags: set(['trigger:comm.start_print', 'source:job', 'trigger:comm.reset_line_numbers'])
2019-08-11 18:52:28,113 Queuing Command: Command Type:None, gcode:M140, cmd: M140 S70, tags: set(['fileline:1', 'source:file', 'filepos:222'])
2019-08-11 18:52:28,116 Extruder Changed: E:0.0, Retraction:0 IsExtruding:False-False, IsExtrudingStart:False-False, IsPrimed:False-True, IsRetractingStart:False-False, IsRetracting:False-False, IsPartiallyRetracted:False-False, IsRetracted:False-False, IsDetractingStart:False-False, IsDetracting:False-False, IsDetracted:False-False
2019-08-11 18:52:28,132 Printer event received:PrinterStateChanged.
2019-08-11 18:52:28,145 Sent to printer: Command Type:None, gcode:M110, cmd: M110 N0, tags: set(['trigger:comm.start_print', 'source:job', 'trigger:comm.reset_line_numbers'])
2019-08-11 18:52:28,150 Sent to printer: Command Type:None, gcode:M140, cmd: M140 S70, tags: set(['fileline:1', 'source:file', 'filepos:222'])
2019-08-11 18:52:28,154 Received from printer: line:ok N0 P15 B3

2019-08-11 18:52:28,160 Queuing Command: Command Type:None, gcode:M105, cmd: M105, tags: set(['source:file', 'filepos:228', 'fileline:2'])
2019-08-11 18:52:28,168 Received from printer: line:ok N1 P15 B3

2019-08-11 18:52:28,170 Queuing Command: Command Type:None, gcode:M190, cmd: M190 S70, tags: set(['source:file', 'fileline:3', 'filepos:238'])
2019-08-11 18:52:28,176 Sent to printer: Command Type:None, gcode:M105, cmd: M105, tags: set(['source:file', 'filepos:228', 'fileline:2'])
2019-08-11 18:52:28,190 Sent to printer: Command Type:None, gcode:M190, cmd: M190 S70, tags: set(['source:file', 'fileline:3', 'filepos:238'])
2019-08-11 18:52:28,193 Received from printer: line:ok T:215.83 /215.00 B:64.35 /70.00 @:63 B@:127

2019-08-11 18:52:28,197 Queuing Command: Command Type:None, gcode:M104, cmd: M104 S215, tags: set(['source:file', 'fileline:4', 'filepos:249'])
2019-08-11 18:52:28,208 Sent to printer: Command Type:None, gcode:M104, cmd: M104 S215, tags: set(['source:file', 'fileline:4', 'filepos:249'])
2019-08-11 18:52:28,211 Received from printer: line: T:215.83 /215.00 B:64.35 /70.00 @:63 B@:127 W:?

2019-08-11 18:52:29,215 Received from printer: line: T:215.00 /215.00 B:64.52 /70.00 @:75 B@:127 W:?

2019-08-11 18:52:30,211 Received from printer: line:echo:busy: processing

2019-08-11 18:52:30,234 Received from printer: line: T:215.00 /215.00 B:64.68 /70.00 @:72 B@:127 W:?

2019-08-11 18:52:31,206 Received from printer: line: T:214.53 /215.00 B:64.84 /70.00 @:78 B@:127 W:?

2019-08-11 18:52:32,197 Received from printer: line:echo:busy: processing

2019-08-11 18:52:32,207 Received from printer: line: T:214.38 /215.00 B:65.00 /70.00 @:79 B@:127 W:?

2019-08-11 18:52:33,204 Received from printer: line: T:214.38 /215.00 B:65.16 /70.00 @:77 B@:127 W:?

2019-08-11 18:52:34,190 Received from printer: line:echo:busy: processing

2019-08-11 18:52:34,210 Received from printer: line: T:214.45 /215.00 B:65.32 /70.00 @:75 B@:127 W:?

2019-08-11 18:52:35,200 Received from printer: line: T:214.73 /215.00 B:65.48 /70.00 @:69 B@:127 W:?

2019-08-11 18:52:36,194 Received from printer: line:echo:busy: processing

2019-08-11 18:52:36,207 Received from printer: line: T:215.00 /215.00 B:65.64 /70.00 @:64 B@:127 W:?

2019-08-11 18:52:37,219 Received from printer: line: T:215.00 /215.00 B:65.80 /70.00 @:64 B@:127 W:?

2019-08-11 18:52:38,224 Received from printer: line:echo:busy: processing

2019-08-11 18:52:38,228 Received from printer: line: T:215.10 /215.00 B:65.93 /70.00 @:63 B@:127 W:?

2019-08-11 18:52:39,210 Received from printer: line: T:215.36 /215.00 B:66.07 /70.00 @:58 B@:127 W:?

2019-08-11 18:52:40,198 Received from printer: line:echo:busy: processing

2019-08-11 18:52:40,210 Received from printer: line: T:215.83 /215.00 B:66.19 /70.00 @:50 B@:127 W:?

2019-08-11 18:52:41,196 Received from printer: line: T:215.83 /215.00 B:66.33 /70.00 @:51 B@:127 W:?

2019-08-11 18:52:42,188 Received from printer: line:echo:busy: processing

2019-08-11 18:52:42,204 Received from printer: line: T:215.83 /215.00 B:66.46 /70.00 @:51 B@:127 W:?
[... 144 lines cut ...]
2019-08-11 18:53:18,377 Queuing Command: Command Type:None, gcode:M221, cmd: M221 S100, tags: set(['fileline:13', 'source:file', 'filepos:599'])
2019-08-11 18:53:18,390 Sent to printer: Command Type:None, gcode:M221, cmd: M221 S100, tags: set(['fileline:13', 'source:file', 'filepos:599'])
2019-08-11 18:53:18,392 Received from printer: line:ok N11 P15 B3

[... cut ...]
2019-08-11 18:53:18,492 A position altering command has been detected.  Fetching and updating position.  Position Command: M420 S1 Z2
2019-08-11 18:53:18,494 Octolapse is requesting a position.
2019-08-11 18:53:18,494 Printer event received:Home.
2019-08-11 18:53:18,498 homing to payload:None.
2019-08-11 18:53:18,506 Queuing Command: Command Type:None, gcode:M400, cmd: M400, tags: set(['trigger:printer.commands', 'plugin:octolapse', 'source:plugin', 'snapshot_gcode'])
2019-08-11 18:53:18,508 Received from printer: line:echo:Unknown command: "M900 K0"

FormerLurker, 2019-08-11:
Hmm.. your firmware doesn't understand M900.  Are you sure it's marlin, or did I misunderstand you :)  In any case, take that out.

Caballete007, 2019-08-11:
Sorry for asking but how i take that out? 
From Cura Start Gcode?

FormerLurker, 2019-08-11:
Yes, if M900 is in there, rip it out :)

Caballete007, 2019-08-11:
I will do and try again avfull diagnostic live print.

FormerLurker, 2019-08-11:
Thank you so much.  Sorry for all this printing/logging I'm putting you through!  What I really need is a CR-10S.  I'm helping 3 different folks (including you) with this printer, and there are 3 totally different and seemingly unrelated problems.  How?!  lol

Caballete007, 2019-08-11:
I appreciate all what you're doing for me thanks a lot.

Caballete007, 2019-08-11:
2019-08-11 18:58:22,509 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X148.795 Y153.174 E39.01659, tags: set(['fileline:1923', 'source:file', 'filepos:54138'])
2019-08-11 18:58:22,511 Received G1
2019-08-11 18:58:22,514 Position Change - G1 X148.795 Y153.174 E39.01659 - Absolute Move From(X:149.609,Y:153.371,Z:2.4,E:68.99013) - To(X:148.795,Y:153.174,Z:2.4,E:69.01659) 
2019-08-11 18:58:22,516 Triggered E:0.02646, Retraction:0 IsExtruding:True-True, IsExtrudingStart:False-None, IsPrimed:False-None, IsRetracting:False-None, IsRetracted:False-None, IsDetracting:False-None, IsTriggered:True
2019-08-11 18:58:22,520 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X148.795 Y153.174 E39.01659, tags: set(['fileline:1923', 'source:file', 'filepos:54138'])
2019-08-11 18:58:22,556 Received from printer: line:ok N2027 P0 B0

2019-08-11 18:58:22,561 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X148.072 Y152.795 E39.04239, tags: set(['source:file', 'filepos:54170', 'fileline:1924'])
2019-08-11 18:58:22,564 Received G1
2019-08-11 18:58:22,565 Position Change - G1 X148.072 Y152.795 E39.04239 - Absolute Move From(X:148.795,Y:153.174,Z:2.4,E:69.01659) - To(X:148.072,Y:152.795,Z:2.4,E:69.04239) 
2019-08-11 18:58:22,565 Triggered E:0.0258, Retraction:0 IsExtruding:True-True, IsExtrudingStart:False-None, IsPrimed:False-None, IsRetracting:False-None, IsRetracted:False-None, IsDetracting:False-None, IsTriggered:True
2019-08-11 18:58:22,566 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X148.072 Y152.795 E39.04239, tags: set(['source:file', 'filepos:54170', 'fileline:1924'])
2019-08-11 18:58:22,605 Received from printer: line:ok N2028 P0 B0

2019-08-11 18:58:22,606 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X147.476 Y152.266 E39.06757, tags: set(['source:file', 'fileline:1925', 'filepos:54202'])
2019-08-11 18:58:22,609 Received G1
2019-08-11 18:58:22,612 Position Change - G1 X147.476 Y152.266 E39.06757 - Absolute Move From(X:148.072,Y:152.795,Z:2.4,E:69.04239) - To(X:147.476,Y:152.266,Z:2.4,E:69.06757) 
2019-08-11 18:58:22,613 Triggered E:0.02518, Retraction:0 IsExtruding:True-True, IsExtrudingStart:False-None, IsPrimed:False-None, IsRetracting:False-None, IsRetracted:False-None, IsDetracting:False-None, IsTriggered:True
2019-08-11 18:58:22,614 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X147.476 Y152.266 E39.06757, tags: set(['source:file', 'fileline:1925', 'filepos:54202'])
2019-08-11 18:58:22,664 Received from printer: line:ok N2029 P0 B0

2019-08-11 18:58:22,669 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X147.002 Y151.59 E39.09365, tags: set(['source:file', 'fileline:1926', 'filepos:54233'])
2019-08-11 18:58:22,671 Received G1
2019-08-11 18:58:22,672 Position Change - G1 X147.002 Y151.59 E39.09365 - Absolute Move From(X:147.476,Y:152.266,Z:2.4,E:69.06757) - To(X:147.002,Y:151.59,Z:2.4,E:69.09365) 
2019-08-11 18:58:22,674 Triggered E:0.02608, Retraction:0 IsExtruding:True-True, IsExtrudingStart:False-None, IsPrimed:False-None, IsRetracting:False-None, IsRetracted:False-None, IsDetracting:False-None, IsTriggered:True
2019-08-11 18:58:22,678 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X147.002 Y151.59 E39.09365, tags: set(['source:file', 'fileline:1926', 'filepos:54233'])
2019-08-11 18:58:22,719 Received from printer: line:ok N2030 P0 B0

2019-08-11 18:58:22,722 Queuing Command: Command Type:None, gcode:G1, cmd: G1 X146.729 Y150.878 E39.11775, tags: set(['source:file', 'fileline:1927', 'filepos:54265'])
2019-08-11 18:58:22,724 Received G1
2019-08-11 18:58:22,724 Position Change - G1 X146.729 Y150.878 E39.11775 - Absolute Move From(X:147.002,Y:151.59,Z:2.4,E:69.09365) - To(X:146.729,Y:150.878,Z:2.4,E:69.11775) 
2019-08-11 18:58:22,725 Triggered E:0.0241, Retraction:0 IsExtruding:True-True, IsExtrudingStart:False-None, IsPrimed:False-None, IsRetracting:False-None, IsRetracted:False-None, IsDetracting:False-None, IsTriggered:True
2019-08-11 18:58:22,726 Sent to printer: Command Type:None, gcode:G1, cmd: G1 X146.729 Y150.878 E39.11775, tags: set(['source:file', 'fileline:1927', 'filepos:54265'])
2019-08-11 18:58:22,762 Received from printer: line:ok N2031 P0 B0

FormerLurker, 2019-08-11:
Can you upload the whole plugin_octolapse.log file to your github repo please?  It does look like logging is working now!

Caballete007, 2019-08-11:
Done

FormerLurker, 2019-08-11:
I'm seeing quite a few resend requests, but it looks like it recovers pretty quickly.  You might have some usb cable issues (not sure if that is what's causing your issue though).

If you have some time, can you take a video of what the printer is doing when it stalls?  I think that might help too.

FYI, I don't have a lot more time today to work on this, but I'll try to squeeze in as much as possible.  I know how popular the CR-10S is, so I really want to get it to work.  Maybe creality would send me one :)

Caballete007, 2019-08-11:
Thats a good idea Lol! I will upload a video

Caballete007, 2019-08-11:
Done.
https://github.com/Caballete007/In-test-mode-works-fine-but-not-when-printing-369

FormerLurker, 2019-08-11:
OH man, it happens AFTER the extruder has returned!  I re-read your initial issue, and you explained this, but I just didn't understand it.  I'll re-review the logs later (probably tomorrow) with this in mind.  That was extremely useful, thanks!

Caballete007, 2019-08-11:
Ok. Appreciate all you have done. Have a good night!

Caballete007, 2019-08-21:
Hi
I hope you doing good. Please when you have time let me know when we can fix this. Thank you. Jeff

FormerLurker, 2019-08-21:
Hey, I've reviewed your logs so many times I feel like I'm familiar with the print!  I was SO SO confused as to what was happening (the last time I took a look was this past weekned), but a fresh set of eyes has yielded the solution I think.

I see this gcode in your snapshot commands: ```G1 E5.00000 F4.800```  Look at the feedrate on that command.  It is incredibly low (4.8 millimeters per MINUTE).  In theory this should take 62.5 seconds to complete, but there is probably some minimum speed that your steppers can handle. (prob somewhere around 50mm/min), so it is taking around 10 seconds to complete.  From the logs I can't tell how long that command takes to run, but after many more gcodes are sent, I can see the printer eventually responds with a busy message because the gcode command queue is full.  

Take a look at your slicer settings and change your deretraction speed (set it to the retraction speed probably).  That should fix it.

The next version of octolapse sucks the slicer settings right out of your gcode file, hopefully elliminating this kind of issue.  I'm editing the release notes for V0.4rc1dev0 right now, which will be the first new release in over 9 months.  With any luck (and I doubt I'll have any), there will be few issues and the production release will be soon to follow!

Let me know if that works (I suspect it will)!

FormerLurker, 2019-08-23:
HA!  I get that way sometimes too, lol.  

So, your Cura settings are probably fine, but your Octolapse settings are not.  I suspect a typo.  Make sure you 'Retraction Prime Speed' within the Octolapse printer profile is set like so:

[image]

Let me know what was there originally please.  I will be here most of the day in case you have more problems!

Caballete007, 2019-08-23:
Hi
I just took a look at that setting it was 1.8 I don’t know where I got that number. I think I never found that setting in Cura or maybe Cura name it differently and I mess things up. I’m going to try with 45 and let you know. Thanks again.

FormerLurker, 2019-08-23:
Ohhhh  boy, I'm 99.9% sure changing that will solve the problem!  So glad.  I'm awaiting your response with great anticipation!

Caballete007, 2019-08-23:
Hi 
LOL. Now it won't print. Out of luck .

[8:23octoprint.log](https://github.com/FormerLurker/Octolapse/files/3536121/8.23octoprint.log)
[8:23plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/3536122/8.23plugin_octolapse.log)

FormerLurker, 2019-08-23:
While I look at the log, can you elaborate on what you mean when you say "it won't print?"  Was there an error message?  Is it just that nothing happens when you click print?  Something else?

FormerLurker, 2019-08-23:
I see your printer is still complaining about the ```M400 K0``` command.  Remove that since your firmware or printer does not support linear advance.  I don't think that caused your issue though.

However, it looks like something happened during your home command.  Your printer is homing (and Octolapse requests a position when this process completes).  It looks to have tried to home for over 30 seconds, then your printer threw this error:

'''
2019-08-23 21:34:37,580 Received from printer: line:Not SD printing
2019-08-23 21:34:37,837 Received from printer: line:  *******RTS_HandleData********
2019-08-23 21:34:37,844 Received from printer: line: 
2019-08-23 21:34:37,849 Received from printer: line:
2019-08-23 21:34:37,870 Received from printer: line:   recdat.data[0] ==241
2019-08-23 21:34:37,876 Received from printer: line:   recdat.addr ==4104== Checkkey==3
2019-08-23 21:34:37,879 Received from printer: line:   stopping ==
2019-08-23 21:34:37,884 Received from printer: line://action:cancel
```

I saw this same error in an earlier log you sent me.  Looks like some other problem is going on with your firmware/usb/serial communication.

For now try going to your printer profile and disabling **Position Auto-Detection** and see if that helps.

Caballete007, 2019-08-23:
Sorry I didn't explain my self. When gets the PLA temp the extruder moves to the center left of the bed and stops there. Position Auto-Detection is in Octolapse?
[image]

FormerLurker, 2019-08-23:
Yes, that looks correct!

Caballete007, 2019-08-24:
Good Morning. Is doing the same as yesterday. I found an error check the picture. Have a great day!

[image]
[8:24octoprint.log](https://github.com/FormerLurker/Octolapse/files/3537320/8.24octoprint.log)
[8:24plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/3537321/8.24plugin_octolapse.log)

FormerLurker, 2019-08-24:
Just ignore that.  There is an option within the snapshot profile settings called Feature Detection that works best when all slicer speeds are unique, but if you are not using it (layer change high quality) you will be fine.  That has been removed from the next version because people are always confused by those messages.

Caballete007, 2019-08-24:
Im sending a few pics of my configuration. The setting that you mention is disable do you want me to enable it?

[image]

[image]
[image]

Caballete007, 2019-08-24:
i cancel the printer in octolapse and i have to turn off the printer and disconnect the pi to make it work. It won't stop and go to the printer main menu.

FormerLurker, 2019-08-24:
@Caballete007, 

No, do not enable feature detection.

I am looking through your log and see that your printer is sending the following message over and over at a pretty high rate of speed:

RTSUpdate Waitway1

The majority of the log is filled with this message.  I have never seen it before and don't know what it means.  I searched through the marlin github repository and can't find this message even within the source code!  Also, you mention you are using Marlin B7 firmware.   I'm guessing that is some kind of Creality3d fork of Marlin?  I can't find any source code for this, and am unable to figure out what that message means.

It could be that M400 just isn't supported, but it was sort of working before!

I'm really confused.  Maybe if I can find the firmware source code I could get to the bottom of this.  Is is possible that some other plugin is creating a conflict?  It wouldn't hurt to disable all other plugins and see if the problem remains.

Caballete007, 2019-08-24:
All the plugins are disabled. If something else comes to your mind let me know. Im going to play with this a bit more, if something else comes to your mind let me know. Thanks a lot my friend

FormerLurker, 2019-08-24:
Of course I will!  I'd really like to get this working for you.  We could try the beta version if you want.  Im heading out for about 5 hours, but could help you get that working if you are interested.

Caballete007, 2019-08-24:
Ok. Just let me know. Have a great day. Im very thankful of all your help.

FormerLurker, 2019-08-24:
If you want to get a head start, take a look at this post:  https://www.patreon.com/posts/new-development-29363149

In step one, use this URL instead of the one listed there:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

By the way, I'm glad to help!

FormerLurker, 2019-10-07:
@Caballete007, any update about your issue?  I'm marking it 'Waiting for Response', and will consider it closed if I don't hear back from you.  Thank you for posting your issue, and enjoy Octolapse!

Caballete007, 2019-10-07:
HiI have not work with it yet. I had a surgery lately and still in recovery. Please don’t close this yet. I will get in touch with you as soon I feel better. Thank you. Jeffrey

FormerLurker, 2019-10-07:
@Caballete007, no problem, and I hope your recuperation is speedy and that your recovery is 100%!  I'm looking forward to hearing back from you!

FormerLurker, 2019-11-09:
Hey, how is it going?  I hope your recuperation is going well.

FormerLurker, 2020-02-26:
I believe this issue must be resolved, so I'm going to go ahead and close it.  Feel free to post a new one at any time, and thanks for your contributions!

closed 2020-02-26 by FormerLurker