FormerLurker/Octolapse #370: XYZ axis woes and G-code

nooooddy, 2019-08-16:
Hi, I'm so close to getting Octolapse working with my DSLR but I feel quite out of my depth modifying the gcode with regard to axis position. Any guidance on Cura code would be greatly appreciated.

Octolapse Version: 0.3.4

OctoPrint Version: 1.3.11

Diagnostic Logging was Enabled: YES

1. Octolapse is in Full Diagnostic Test Mode
2. Initiate printing a test print
3. Octolapse unable to access XYZ axis position 

#### What should have happened?
Octolapse triggers an external DSLR on completion of each layer.

#### What happened instead?
Errors occurred with regard to XYZ axis positioning.

OS Name: OctoPi
Os Version: 0.16.0

Printer Model: Creality CR-10S
Printer Firmware Version: TH3D U1.R2.14

Browser: Safari 12.1
Browser OS: MacOS Mojave 10.14.4

Link to Gcode File: https://pastebin.com/RFQMkBfg

Link to plugin_octolapse.log:  https://pastebin.com/cn4Y66rH

Link to octoprint.log:  https://pastebin.com/iL9g3np4

FormerLurker, 2019-08-16:
Hi!  Thanks for creating this issue!

Please edit your printer profile and change the 'XYZ Axis Mode' to 'Default to Absolute'.  My quick review of your gcode suggests this should fix that issue.  You might be interested to know that this problem has been solved in the alpha version, so it won't cause problems for too much longer.  However, you may have other issues.  In that case, please just reply here and we'll work through them one step at a time.

Begin Rant... I have to admit, I've gotten LOTS of issues created by CR-10S users in particular recently.  I asked Creality3d for an Ender 3 for testing purposes, but they didn't get back to me (after sending me a message saying they would get back with me, lol!).  I ended up purchasing one for my father since it was the causing more issues with Octolapse than any other printer, and I was tired of not having access to a machine for testing.  The Ender 3 issues have, as a result, mostly been solved, but the CR-10S is probably now the printer I hear the most issues from (perhaps in part due to it's popularity).  It's also baffling to me that many Cr-10S users have no problems at all.  I will reach out to Creality to see if I can get a machine for testing purposes, but I'm not confident I can do this.  Hopefully you will find yourself in the group that has no problems beyond this simple configuration problem I have hopefully solved for you :)....  End rant :)

nooooddy, 2019-08-16:
Thank you SO much! All appears to be working well.

https://imgur.com/a/XjWTpxf 

From looking at the gcode I could see it was injecting an absolute value but I wasn't confident enough to alter the setting in Octolapse, so thank you for the assistance. Next time I'll trust my judgment and go for it.

I think I have a particularly awkward setup as I'm using the CR-10s and a Canon 800D which gphoto2 doesn't seem to like too much but I think I've ironed out the kinks, largely thanks to your code here: https://github.com/FormerLurker/Octolapse/wiki/DSLR-Image-Capture-With-No-Download

I'll carry out a real test print tomorrow morning (GMT) and post the results. Thank you again, I'll be signing up on Patreon!

FormerLurker, 2019-08-16:
Yes, that is looking good!  Don't worry about posting the issue, I'm always here to help!  I've heard about this to often (XYZ and E axis modes) that I was inspired to fix this permanently by creating a profile library and an easy way to export/import/download/auto-update profiles.  However, the basic problem of knowing what axis mode the printer is in still exists.  I just decided to take the most permissive approach.  Any problems caused by this are due to incomplete starting gcode, which is often missing a G90 or M82 (not usually M83, which is rarely a problem since everyone assumes the extruder starts out in absolute mode).  In reality, the printer could be in any axis mode at the start of a print, so these codes should always be included.  In fact, it wouldn't hurt to add a G90 as the very first line in your start gcode in your slicer and an M82 as the second line.  99.9 percent of the time it will be fine as is, but that 0.1% can be bad.

Regarding Patreon, thank you so much!  That's pretty much how I measure how well I'm doing.  I REALLY appreciate your potential patronage!

FormerLurker, 2019-10-07:
Since it looks like your initial issue was solved, I'm going to go ahead and close this thread.  If you have any more problems, feel free to create a new issue, and THANK YOU for posting!

closed 2019-10-07 by FormerLurker