FormerLurker/Octolapse #372: I have problem with installing octolapse

TheTomasVideos, 2019-08-19:
#### If this is a feature request describe it here

---

#### Version of Octolapse
 
Octolapse Version: Newest

#### Version of OctoPrint

OctoPrint Version: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

I tried to install octolapse through Putty, because I couldn't do it through PluginManager

#### What should have happened?
The plugin should be installed

#### What happened instead?
Plugin not installed = Error

#### Operating System running OctoPrint and Octolapse

OS Name: Windows
Os Version: ---

#### Printer model & used firmware incl. version

Printer Model: Flsun I3
Printer Firmware Version: ---

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: ---

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: I didn't print

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  ---

#### Link to octoprint.log

Link to octoprint.log:  ---

#### Link to contents of Javascript console in the browser

Link to javascript console output: ---

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://ctrlv.sk/c7ST

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-08-19:
From the snapshot it doesn't look like you activated the OctoPrint virtual environment, so PIP didn't have access to any of the libraries installed there (like the setuptools, as is shown in the error).  Before you install anything for OctoPrint via pip, you need to activate the virtual environment like so:

```
source /home/pi/OctoPrint/venv/bin/activate
```
The result should look something like this:

[image]

Then you can use pip to install into the venv as you tried to do earlier.  Use the following command once you are finished installing to exit the virtual environment:

```
deactivate
```

However, it would be very helpful if you could provide the output from the plugin manager within OctoPrint that shows the installation error messages.  If I had to guess, I'd say it is having trouble installing the Pillow library (used for jpg editing, text overlays, and a few other things).  Typically this indicates a problem with either your power supply (this has been verified multiple times by several users) or your USB cable.  If this IS the issue (and it likely is) you may suffer from similar unexpected issues while printing or while running OctoPrint or certain plugins.  Pay close attention to Octoprint to see if you are suffering from either thermal throttling or undervoltage indicators.  Take a look at this page for more information about this:  https://community.octoprint.org/t/pi-undervoltage-power-issues-i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256

If you want to try installing via the plugin manager before using the venv solution I posted above, please link to a copy of the output and I can verify the exact error.

Thanks, and good luck!

TheTomasVideos, 2019-08-20:
**this is my log when i try to install octolapse through pluginmanager, i also have the same log i have in putty:** 
Installing plugin from https://github.com/FormerLurker/Octolapse...
/home/pi/OctoPrint/venv/bin/python -m pip --disable-pip-version-check install https://github.com/FormerLurker/Octolapse --no-cache-dir
Collecting https://github.com/FormerLurker/Octolapse
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0xb5e40e50>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /FormerLurker/Octolapse
Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0xb5e40a90>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /FormerLurker/Octolapse
Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0xb5e40bb0>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /FormerLurker/Octolapse
Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0xb5e40f50>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /FormerLurker/Octolapse
Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0xb5e2bf90>: Failed to establish a new connection: [Errno -2] Name or service not known',)': /FormerLurker/Octolapse
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /FormerLurker/Octolapse (Caused by NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0xb5e40870>: Failed to establish a new connection: [Errno -2] Name or service not known',))
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

FormerLurker, 2019-08-20:
It looks like a network issue.  Do you have this problem installing any other plugins?  Also, do you have some kind of firewall or proxy installed?  Still, try installing from the plugin manager within OctoPrint, click 'Get More' and paste this into the 'from url...':  https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip

See if that works.  If not, let me know if the error message changed.

FormerLurker, 2019-08-20:
Hey, where are you trying to install from?  I was reading about a very similar issue that was due to some countries blocking IPs wholesale (github in this case).  This could be your issue.

FormerLurker, 2019-08-20:
I just thought of a way to work around this I think.  Try downloading the zip archive at this address:  https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip

Next, open the plugin manager and select 'Get More'.

Finally, click browse next to the '... from an uploaded archive' box, select the zip archive you just downloaded, and click install.  As long as you can download the zip, I think it should work!

TheTomasVideos, 2019-08-21:
**So I found out that the antenna was not connected and I couldn't install any plugins. after I fixed this problem, another one came up. When I installed the plugin via Putty, it did not install despite this log:**
pi @ octopio: ~ $ source / home / pi / OctoPrint / venv / bin / activate
(venv) pi @ octopio: $ $ install "https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip"
Collecting https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
  Downloading https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
     - 6.8MB 1.2MB / sec
Collecting pillow (from Octolapse == v0.3.3)
  Downloading https://files.pythonhosted.org/packages/51/fe/18125dc680720e4c3086dd3f5f95d80057c41ab98326877fc7d3ff6d0ee5/Pillow-6.1.0.tar.gz (33.8MB)
    99% | 33.8MB 3.3MB / sec eta 0: 00: 01Killed
**Next I tried to install plugin via pluginmanager = also unsuccessful ... log:**
2019-08-21 08: 53: 27,209 / home / pi / OctoPrint / venv / bin / python2 -m pip -disable-pip-version-check install .3.4.zip --no-cache-dir
2019-08-21 08: 53: 30,673> Collecting https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
2019-08-21 08: 53: 32,177> Downloading http://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
2019-08-21 08: 53: 39,189> Collecting pillow (from Octolaps == v0.3.4)
2019-08-21 08: 53: 48,328> Downloading http://files.pythonhosted.org/packages/51/fe/18125dc680720e4c3086dd3f5f95d80057c41ab98326877fc7d3ff6d0ee5/Pillow-6.1.0.tar.gz (33.8MB)
**So I don't know where the bug might be .. maybe I already have full memory and I can't install the plugin there ... But that would be weird because I don't have any other plugin installed there ... Btw: I'm using Orange Pi Zero 256mb + 8gb . And I'm from Slovakia :)**

FormerLurker, 2019-08-21:
There doesn't appear to be an error there.  Did it just lock up, or did it exit?  If it locked up, you probably have problems with your power supply.  That sounds odd, but I've seen it many time with exactly this module (pillow, which is the last package that OctoPrint tried to install).  Check your PSU definitely, because that will cause you other mysterious problems down the road, but you might also be able to try this:

1.  Connect to your pi and run the following commands:

sudo apt-get update
sudo apt-get dist-upgrade
sudo reboot

2.  Let your pi reboot, then re-connect and run these commands:

source oprint/bin/activate
pip install --upgrade pip
pip install pillow --no-cache-dir

If pillow installs correctly, log into OctoPrint and try installing Octolapse again.  It should complete this time.

Let me know how it goes!

TheTomasVideos, 2019-08-23:
Hi, octolapse works for me but not quite ...
The problem is that my printer is set up so that I can adjust the head height and print before printing. So I don't start gcode. start gcode is empty. And that's probably the problem. When I put an object to print with octolapse, it shows me the position error and that the printer is off the print pad. Do you have any solution for this problem?

TheTomasVideos, 2019-08-23:
and I don't have an inductive sensor for auto bed leveling because it doesn't suit me ...

FormerLurker, 2019-08-23:
@TheTomasVideos, this is unfortunately a problem.  Octolapse requires a home command in order to start detecting position changes.  The new version is somewhat more lenient here.  Let me see if i can get it to work without a home, and I'll let you know what happens.

Are you willing to install a beta?  I just released the devel/rc yesterday, so it's been minimally tested (about 6-7 users so far, most of whom have it working).  let me know since that will affect how I can resolve this for you.

TheTomasVideos, 2019-08-23:
I am also willing to install the beta version

FormerLurker, 2019-08-23:
So, I determined that the new Smart Layer Trigger worked just fine without the home command, which is only available in the beta.  However, the classic layer trigger (the one you are using) in the beta version did not work, but I believe I have fixed this too!  As long as Octolapse knows the current position via some other method (like seeing absolute coordinates for X, Y and Z axis), it will now take snapshots.  I also updated the printer state info panel to reflect the new state I added to deal with this (called has_definite_position behind the scenes and **Known Position** in the info panel).

I just pushed the changes, and have updated the version number to V0.4.0rc1.dev1, so please install from the plugin manager using this url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Quite a bit has changed in this version, but you should be able to follow [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse#after-installation) starting with the **After Installation** heading.  Please post any feedback specific to that version [here](https://github.com/FormerLurker/Octolapse/issues/373).  I will close this issue after you let me know if it has installed correctly, and we can continue on the other thread.  I will probably create new tickets for any new issues that are related to V0.4.0rc1.dev1.

Thanks!

FormerLurker, 2019-08-23:
OH, one more thing, the changes have not been tested on a live printer yet fyi.  Make sure you watch your printer closely, and stop the print immediately if you have any issues.

Also, please test using the 'Classic Layer Trigger' first (I also recommend you use test mode), just to make sure the problem has been solved, then feel free to play around :)

FormerLurker, 2019-10-07:
@TheTomasVideos, any update on your issue?  I'm marking this thread as Waiting for Response.  If I don't hear back I'll consider it closed.  Thanks for posting your issue!!!

FormerLurker, 2019-11-09:
So, it's been over 30 days, so I'm going to go ahead and close this issue.  Let me know if you want it reopened.  Thanks!

closed 2019-11-09 by FormerLurker