FormerLurker/Octolapse #373: V0.4.0rc1.dev0 Feedback

FormerLurker, 2019-08-22:
This issue is to be used for reporting your experience using V0.4.0rc1.dev0.  If you are NOT using this version, please create a separate issue.

All feedback related to the newest release is welcome here!  Feel free to report problems, successes,  likes, dislikes, suggestions, or just let me know that you tried it out.  ANY feedback is helpful, and I appreciate your time and effort.

If any issues are found (and I'm sure they will be), I will either ask you to create a new issue, or I will create one and send you a link.  This will keep things clean and on-topic.

THANK YOU SO MUCH for helping me out!

FormerLurker, 2019-08-29:
@b-morgan, FYI, I think I have improved the quality situation.  I'll be running lots of tests over the next couple of days, and if they all turn out right I'll be dropping the next build.  There were a lot of misc bug fixes, so huge performance increases for pre-processing (down by over 60%), and one new setting when using the snap to print feature to deal with an issue I had.  I've also removed some smart layer trigger settings that were just getting in the way and didn't work as well as I wanted them to.

So, nothing major except the quality stuff.  For best results when using Slic3r you need to turn on verbose g-code output (print settings->output options->Verbose G-code).  I'll include all of this info when I drop the new version.

tomrav, 2019-09-01:
I've been using `rc1` for a couple of days (with a Prusa MK3S), only tried a handful of prints - but I'm already getting better results than I used to. The auto-config-detection feature is awesome.

The dedicated camera settings panel is also a really nice addition and makes optimizing the whole thing way easier.

Thank you for all the hard work. 👍

FormerLurker, 2019-09-01:
@tomrav, thank's for trying out 0.4.0rc1dev0!  Can I ask what trigger profiles you used (smart layer, snap to print, classic, etc)?  Also, were you able to find your printer in the make/model drop down?

So glad you got the camera settings to work too!

tomrav, 2019-09-01:
So I've done a couple of prints. I'm using a Prusa MK3S which showed up in the menu, and a logitech c920 with the new dynamic panel to set it up.

The [first print](https://i.imgur.com/StSgqMw.gifv) was my first attempt with the new version, I believe the settings I was using were stabilize center, and smart - high quality. Focus is a bit off here sadly.

For the [second print](https://i.imgur.com/fzgLZ0J.gifv) I used the stabilize extruder feature to better focus and line up my camera. I used the printing animation with the snap to print settings (also, this was a vase mode print).

I don't think either timelapse degraded the resulting print.

Overall I know these can be a lot better (mostly positioning, lighting and focusing), but these are the best results I've gotten yet, so I'm pretty pleased.

FormerLurker, 2019-09-02:
Holy crap, those look excellent!  Was the second one made with snap to print sliced in vase mode?  No quality impact at all?

FormerLurker, 2019-09-03:
The next version has been released, so I'm closing this issue.  Please check [this](https://github.com/FormerLurker/Octolapse/issues/378) issue for details.

Thanks everyone!  Closed!

closed 2019-09-03 by FormerLurker