FormerLurker/Octolapse #375: Stops extruding after some layers

KaminoKGY, 2019-08-29:
#### Version of Octolapse
Octolapse Version: Octolapse: 0.3.4

#### Version of OctoPrint
OctoPrint Version: OctoPrint: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES
I can reproduce the issue.

#### What were you doing when the problem occurred
I print on an Ender 3 via Octoprint + Octolapse. I take the pictures using an external camera (Canon 70D) and the script provided on the Octolapse documentation.
I just have to launch a print using the external camera from the extruder to stop after some layers.
If I use only the webcam to take pictures it seems to work (on small prints at least).
I tried different configuration to be almost sure that it comes from octolapse. When I print via Octoprint and disable Octolapse it works.

The issue seems to have started when I replaced the Creality motherboard from 1.1.3 to 1.1.5.

#### What should have happened?
It should finish the print and continue extruding.

#### What happened instead?
It stops the exxtrusion after some layers. It stops quite early after less that 10 layers for big base prints and more like 50 for smaller base prints.
The head continues to move but the extrusion stops.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version
Printer Model: Creality Ender 3
Printer Firmware Version: 1.1.8
Motherboard 1.1.5

#### Browser and version of browser, operating system running browser
Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://gist.github.com/KaminoKGY/19c7f09ab3457ee4b95320a895903df9

#### Link to settings.json
Link to settings.json with all passwords removed: https://gist.github.com/KaminoKGY/cd286db97b0378a0e3175147dc8e4f98

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://1fichier.com/?t3xd1ai44v5rfgfrsdch

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/KaminoKGY/48a417056c124e574fd3371ed26198a1

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: https://youtu.be/3kidDGgn88E

FormerLurker, 2019-08-29:
Ok, so I looked at your gcode and your octolapse settings, and I see several problems that can be fixed pretty easily!  Some will take some work, however, but I'm confident it can be done.

I can't tell what your cura settings are, but I don't see any retractions in your gcode.  Why are you not using retraction?  Sometimes there are good reasons for this, but when using Octolapse you want to enable retraction and calibrate it via some kind of stringing test BEFORE you use Octolapse.  Currently Octolapse uses the retraction length both to detect retraction AND to perform a retract while moving (there is no independent retract length yet, but I may add one in the future).  Combine this with the fact that your retraction length is set to 20mm in octolapse, and you're most likely getting filament jams!  Either set retraction length to 0 in Octolapse, or up your retraction in cura.

Now, having said that, go through your cura settings and match them EXACTLY with your slicer settings in Octolapse.  Just copy each setting in until they match.  It is EXTREMELY important that you do this for every setting.  Ignore any errors about non-unique speeds in your Octolapse printer profile, btw.  FYI, the next version of Octolapse can suck your settings right from the gcode file, which would have prevented this mistake that I often make myself too (hence the new feature).

Second, I see very high camera delays of one second.  That will cause some serious oozing issues, especially if retraction is disabled.  I recommend turning these down to something more reasonable like 50-150 milliseconds and only adjust them up if you absolutely need to.  It also looks like you are using a DSLR.  If so, turn the delay to 0 for your DSLR.  Delay for a DSLR is almost never necessary unless your printer is shaking and wobbling after the travel (which is also a problem for quality).

Once those are solved, give Octolapse another go, but use a stringing test (something small) to test your retraction.  These are usually worst-case sorts of tests, so you'll probably see some stringing.  Upping your retraction length only helps to a point.  Too high and you'll get jams (as you've seen).  Same for retraction speed.  Too high and you'll separate the liquid and solid filament, causing more oozing and other problems.  You want to use the minimum retraction length possible and the highest speed possible in my experience.  Also your temps play a huge role.  Too low and you'll get jams and stringing.  Too high and you'll get stringing and more oozing.

Also, I see from your video that your DLSR must be taking forever to acquire an image.  It shouldn't take more than a couple seconds.  If it does, either your image resolution/quality is too high, your usb transfer is slow, or your camera has too many automatic settings (auto focus, long exposure, etc all add up).  There are ways of reducing this (deferred download is the best IMHO, where images are kept on the camera until rendering, then are all downloaded at once), and I can point you in the right direction.  However, for now make sure it works while using a webcam ONLY with 60MS delay and let me know how that works.  From there it can only get better!

THANKS FOR POSTING YOUR ISSUE!  I'm glad to help!

KaminoKGY, 2019-08-29:
Wow this answer and so kickly ! :D 

You are right the retraction was disabled on Cura. It's a mistake from me, after lots of tests I guess that I lost track -_-" Sorry for that
Indeed the 20mm retraction is very high. It was to try to avoid the stringing when Octolapse was moving the print to take a picture.

Oh I am waiting for this Octolapse update. Extracting the parameters from gcode is a good one !

For the DSLR settings it takes time because of the long exposure.

Thanks a lot for your answer ! It is very clear.
I will keep you informed ;)

FormerLurker, 2019-08-29:
> Indeed the 20mm retraction is very high. It was to try to avoid the stringing when Octolapse was moving the print to take a picture.

5-7MM seems to be the sweet spot for the Ender, and somewhere around 40-70mm/sec retraction speed.  Just make sure whatever you choose you enter into both cura and octolapse :)  FYI, I had a bowden for a while too, and getting the retraction speed and length tuned was a constant headache, but one that was well worth it!

> For the DSLR settings it takes time because of the long exposure.

Increase your lighting then to reduce exposure.  Bright lights are the #1 overlooked element for increasing your timelapse quality and reducing your exposure time.  The delay in the Octolapse camera profile isn't necessary no matter what the exposure is since the snapshot script will only return after the image has been acquired.  In short, turn the delay for the DSLR down to 0 :)

>Thanks a lot for your answer ! It is very clear.

Thanks, I try :)

FormerLurker, 2019-09-24:
@KaminoKGY,

So, it could still be that your main problem is settings related.  Octolapse stabilizations take longer than the average retraction.  Maybe the extra time while retracted is leading to jams?  I JUST had another issue like this over the weekend, and the user managed to get much better extrusion after reducing retraction.  Also, the slicer settings he was using didn't align with the Octolapse printer profile settings (which they must).

Maybe for fun try reducing your retraction to 2MM and increase your temps by about 10 degrees Celsius?  If that fails, reduce your temps by 10C and try again.  You could also try slowing down the retraction speed somewhat too.  The goal here would not be to increase print quality, but to narrow the problem down.

After you've tried that, maybe you would consider upgrading to the latest alpha (soon to be beta)?  We could then eliminate Octolapse slicer settings as a possible issue.

I can't think of any other reason than jamming to explain your issue, and that's usually related to retraction/temps.  I would be very hesitant to ditch my brand new silent motherboard!

FormerLurker, 2019-10-07:
@KaminoKGY, any update on your issue?  I'm marking it as waiting for response, and will consider it closed if I don't hear back.  Thanks for your post!!!

KaminoKGY, 2019-10-07:
@FormerLurker Not really
I am still in contact with Creality but it's slow and they were closed last week -_-
Still trying to make it work ^^

FormerLurker, 2019-10-07:
No problem, I removed the waiting for response label.  Thanks for getting back to me!

KaminoKGY, 2019-10-31:
@FormerLurker It's been quite some time and much more investigations ^^
It's not easy to deal with Creality. I am not sure that they know from where it comes from. I will probably just get a spare part that does the exact same thing. But for you there is nothing to do. It's a known issue from the 1.1.5 motherboard. I found other people talking about it. The motherboard doesn't seem to handle a lot of retraction well. It is the reason why the result with retractions tests were the worst.
Sorry for the opening the issue :/

closed 2019-10-31 by KaminoKGY

FormerLurker, 2019-10-31:
Don't be sorry!  It was worth a shot.  If you need anything else let me know!