FormerLurker/Octolapse #376: Feature detection not working with small fast tower printd

seltix5, 2019-09-01:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

When using feature detection (infill, inner wall, top/bottom and initial layer) to print something like this https://www.thingiverse.com/thing:1015641 using trigger type layer/height the snapshot does not trigger. 

#### What should have happened?
There should be a snapshot during the infill.

#### What happened instead?
No snapshot was taken.

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: 

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 7

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://www.thingiverse.com/thing:1015641

#### Link to settings.json

Link to settings.json with all passwords removed: how do I get the settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-09-01:
Hi!  Thanks for posting this issue!

So, feature detection is based on the feedrate, which gets adjusted arbitrarily by slicers if the estimated layer print time is below a certain threshold.  This means that all features can end up printing at the same speed, breaking the feature detection.  I'm certain this is what you're experiencing.

So, there is good news and bad news.  The bad news is I've removed feature detection from the next version of Octolapse because of this exact issue (and for some other reasons, like it's a pain to configure, bloats the printer settings, and confuses users who don't intend to use it).  However, I've also added a similar, less error prone version of feature detection that requires no settings at all to the new version.  It's based on gcode comments, and only works when using a new feature called the 'smart layer trigger'.   It works out of the box for CURA and Simplify3d, but requires you to turn on 'Verbose GCode' commenting in Slic3r and PrusaSlicer (slic3rPE too).

For now I recommend disabling feature detection on V0.3.4.  If you want to test out the new beta (I'm working on v0.4.0rc2.dev2 right now), let me know.  I'm running test prints all weekend and fixing all of the issues that have been reported or that I've found myself.  I have NOT tested this in simplify 3d yet, so if you are using that you might want to hold off.  Anyway, let me know!

Thanks!

seltix5, 2019-09-01:
Hello,
I'm using Cura so I think it may be interesting for me, i do want to try it =D For a workaround for this situation I found I could enable "travel" for the feature detection to work but the print quality decrease because some blobs are showing compared to the print without octolapse.
How do I try the new version?

FormerLurker, 2019-09-01:
Oh yeah, enabling the travel feature will cause issues.  Thanks for your offer to help test the development version!  It's been a HUGE project and there aren't enough hours in the day to test everything :)  My printer is constantly making benchys, small vase mode rockets, and 50mm test towers :)

So, [here is a guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) with basic instructions for installing the development version.  When the instructions have you choose which link to install from, use the second one (this link https://github.com/FormerLurker/Octolapse/archive/devel.zip).  

Please keep in mind that this is still experimental.  So if you don't mind running and possibly ruining a few prints, I'd REALLY appreciate the help.  Also, I'm running a series of tests right now to verify some pretty big changes within the gcode parser and position calculation algorithm.  If you want to hold off until I've tested the major bits, that's fine.  I'll send you an update when I'm finished.  It will probably be a few hours (4-15 minutes per test, 11 tests total).  Let me know!

Thanks.

[Follow this link]()

seltix5, 2019-09-01:
happy to help. i'm a php developer myself, so I understand the work and dedication needed for these projects :D
I will install now and do some tests!

FormerLurker, 2019-09-01:
So, apparently I broke the smart layer trigger somehow (lol), so you might want to start out with the classic layer trigger. I'll let you know when I fix this and you can reinstall from the same link.

FormerLurker, 2019-09-01:
I fixed the issue (refactoring bug - parameter value was the same as a member, and my IDE didn't complain.. ug..) and have finished a few test prints.  So far so good (on test 4 of 12).  You might want to reinstall when you have a chance from this URL:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

seltix5, 2019-09-01:
installed. the new config are super begginer friendly, congrats!
2 bugs :
- printer > Cura V4.2 and Above : "Retraction Enabled" should be "Enable Retraction"
- cameras > Test : `Errors were detected - An unexpected error was raised while testing custom image preferences for the 'Default Webcam' camera profile.`

seltix5, 2019-09-01:
first question : for my print test cura is showing 84 layers, yet the Snapshot Plan says 75 plans. Is it not related or are some layers missing?

FormerLurker, 2019-09-01:
> printer > Cura V4.2 and Above : "Retraction Enabled" should be "Enable Retraction"

Fixed, thanks!

> cameras > Test

Are you using a raspberry pi cam?  If so, did you follow [this guide](https://github.com/FormerLurker/Octolapse/wiki/Configuring-a-Raspberry-Pi-Camera) for enabling custom image preferences?

If you are using a USB camera, did you follow [this guide]() for enabling access to control.htm?  If so, see if you can navigate to control.htm via an address similar to this:  http://{ip_of_pi}/webcam/control.htm

FormerLurker, 2019-09-01:
Can you export your settings.json file (Gear icon on octolapse tab->Export Settings button) and create a gist (gist.github.com)?  Also create a gist for your gcode file.  From that I'll be able to figure out what is going on.  There are several possibilities (bugs included), and your settings+gcode will let me figure it out pretty easily I think.

seltix5, 2019-09-01:
i'm using logitech webcam and I did not enable custom image preferences

FormerLurker, 2019-09-01:
> i'm using logitech webcam and I did not enable custom image preferences

Hmm..  that is mysterious.  Your settings.json file should clarify things, I hope!

seltix5, 2019-09-01:
https://gist.github.com/seltix5/d5ccb9687ff9b16ed5dc2fdbcbdceccb

FormerLurker, 2019-09-01:
Sorry, I should have clarified that I need your .gcode file, not the .stl.

seltix5, 2019-09-01:
sorry, it is the gcode, I just put the wrong filename.

I try to enable custom image preferences as per instruction here https://github.com/FormerLurker/Octolapse/wiki/Troubleshooting#why-cant-i-change-contrast-zoom-focus-etc but I got 404 on {IP}/webcam/control.htm any way :S

seltix5, 2019-09-01:
PS : the timelapse did work, its just showing that error when I click test anyway

seltix5, 2019-09-01:
using "Smart - High Quality" did not improve result with previous version :(
[image]
[image]

FormerLurker, 2019-09-01:
Yikes!  Hopefully it is some bug in the new feature tagging.  It seems to
work pretty well in slicer.  I will take a look at the gcode and will let
you know what I find.

FormerLurker, 2019-09-01:
FYI, I think I found and fixed the camera settings error message that popped up.  It was testing for connectivity to control.htm when it shouldn't have been trying.

seltix5, 2019-09-01:
thats why then. i'm trying to enable control but no success. searching online for possible solution...

FormerLurker, 2019-09-01:
> if its not working it's probably because its not compatible with my old logitech c250 right?

I think it should work with it.  Can you paste in your octopi.txt file (or create a gist)?  Also, if you are trying https you might have to switch to http when trying the control.htm address.  Any proxy running or anything like that?  

Also, you might try just going to http://{ip_of_pi}/webcam/ and see what happens (the ending / is important).  If a webpage loads, there should be a control link in a menu bar, which should take you to control.htm.

seltix5, 2019-09-01:
the "Smart - Snap To Print - High Quality" just takes snapshot on layer change without moving right? because the print quality is perfect normal (the same as with octolapse disabled). there is only one problem that is not appening with the "Smart - High Quality" profile, its taking a snapshot at the begining when is purgin the side line (you can see this on the timelapse videos)

FormerLurker, 2019-09-01:
> the "Smart - Snap To Print - High Quality" just takes snapshot on layer change without moving right? 

Pretty much!

> its taking a snapshot at the begining when is purgin the side line

This can be prevented actually.  There are a few ways, but I'm most interested in the first way since I can use this information to update the ender 3 profile and prevent this for everyone all at once.

1.  Restrict the snapshot area - It looks like the ender purges on the left edge of the bed, if I'm reading the gcode.  I'm going to be conservative and say the usable bed volume starts at absolute X >= 11, with no restrictions on the Y axis.  So, if all that is true, this should work - Open print settings, go to restrict snapshot area and set the values to the following:

[image]

2.  Go into rendering settings and set **Skip Snapshots At Beginning** to 1.  Octolapse will still take a snapshot, but the rendering engine will know to skip this frame, both for any pre-roll, and for your initial timelapse frame.

Let me know which one you choose, and if it works.  I'm still looking at your gcode regarding the 'Smart Layer - High Quality' profile.

Also, just for fun, you might want to turn your camera delay (in your camera profile) down to 0 and re-try the regular snap to print profile.  The delay has a HUGE impact here.  Your timelapse might be just a bit jittery (there are ways to fix that, like cranking your FPS up to 30-60), but the print quality impact will be much lower.

seltix5, 2019-09-01:
> I think it should work with it. Can you paste in your octopi.txt file (or create a gist)? Also, if you are trying https you might have to switch to http when trying the control.htm address. Any [...]

sory, it was my stupidity. I was editing octopi.txt with filezilla but the file wasnt updating duo file permission and I did not notice. After updating the file with SSH it worked! thanks!

seltix5, 2019-09-01:
> > the "Smart - Snap To Print - High Quality" just takes snapshot on layer change without moving right?
> 
> Pretty much!
> 
> > its taking a snapshot at the begining when is purgin the side line
> 
> This can be prevented actually. There are a few ways, but I'm most interested in the first way since I can use this information to update the ender 3 profile and prevent this for everyone all at once.
> 
> 1. Restrict the snapshot area - It looks like the ender purges on the left edge of the bed, if I'm reading the gcode.  I'm going to be conservative and say the usable bed volume starts at absolute X >= 11, with no restrictions on the Y axis.  So, if all that is true, this should work - Open print settings, go to restrict snapshot area and set the values to the following:
> 
> [image]
> 
> 1. Go into rendering settings and set **Skip Snapshots At Beginning** to 1.  Octolapse will still take a snapshot, but the rendering engine will know to skip this frame, both for any pre-roll, and for your initial timelapse frame.
> 
> Let me know which one you choose, and if it works. I'm still looking at your gcode regarding the 'Smart Layer - High Quality' profile.
> 
> Also, just for fun, you might want to turn your camera delay (in your camera profile) down to 0 and re-try the regular snap to print profile. The delay has a HUGE impact here. Your timelapse might be just a bit jittery (there are ways to fix that, like cranking your FPS up to 30-60), but the print quality impact will be much lower.

yes by default (at least from Cura 4.1 and up) and many other profiles I think I always saw that start gcode with the purge line at the left. I will set the 1st option because :
- is unlickely to have something printing between X 0 <> 10 only
- option 2 is a great option but for this case it will cause a skip frame in all other profiles too ( I want all the frames! :P )

the snap to print is perfect, no need to reduce the delay, there is nothing to improve (at least in this small test).

speaking of printer profile for Ender 3, I already updated my profile here because I know from the paper Ender 3 has a 220x220 bed but in reality is 235x235. Cura 4.2 is using 220x220 by default too but I updated booth to 235x235. I dont know if the best is to leave the 220x220 or the true bed size...

seltix5, 2019-09-01:
for the restrict area I dont need to set the max values right?

FormerLurker, 2019-09-01:
@seltix5, yeah, I could default them.  I'll think about changing that at some point.

FormerLurker, 2019-09-01:
So, I'm getting on the right track regarding the smart layer trigger.  For some reason when I run your file, it is super slow in the debugger.  I've never seen this before, so I suspect it's something with the new C++ parser/position processor.  It might be causing stalling (did you happen to notice this?  Will keep you posted.

seltix5, 2019-09-01:
Stalling where? What is the debugger? The snapshot plan? When I send the print order from cura to octoprint the snapshot plan modal show up just after and I just click continue, did not notice any waiting time other than for the eating

FormerLurker, 2019-09-02:
@seltix5, forget the stalling issue.  It was something messed up with the virtual printer.  A reboot fixed it (I spent way to much time figuring that one out, doah!).

I did find something that might have caused the quality issues, however.  Just by sheer unluck, the stabilization position aligned with the layer change position between two travels.  This caused a retract + lift + travel + snapshot + return + delift + deretract, and then all over again.  That means double retraction a lot of waiting.  

Also, I figured out the missing layers.  Your start gcode is a bit different than what I've seen before, and I've added some eager layer detection was causing a layer detection at height = 2 MM (line 35 here:  G1 Z2.0 F3000).  Your extruder is primed at that point.  I recommend you add a retract before that command, which would reduce stringing on the way to your first skirt point.

To fix this I made my layer detection a bit less eager.  It now will trigger a new layer if there has been either extrusion or a deretraction on a higher level.  I think that's better anyway.  The purpose of eager detection was to prevent double retractions for the classic triggers that were causing some ppl problems, but I think this way takes care of both issues.

There was also a problem getting a snapshot after (or during) the first layer because it was still detecting the priming lines as the highest layer found.  So I also excluded that region from the most_recent_extrusion_height calculation, and it seems to work ok now.  Going to do a few tests, then will push those changes for you to try out with the _smart - high quality_ trigger.

FormerLurker, 2019-09-02:
My first test is worked just as expected!  I'm still not 100% sure what was causing the quality issues you were seeing (hopefully it was the two snapshot calls so close together).  Maybe reinstall from the devel branch using the same url (```https://github.com/FormerLurker/Octolapse/archive/devel.zip```) and try the _Smart - High Quality_ trigger again?  If it turns out bad, run it again, but change your debug profile to 'Live Print - Debug Issues', print again and send me your plugin_octolapse.log file (gist would be preferred again).  You can find it by clicking the OctoPrint Settings button (wrench/gear icon)->Logs, then finding plugin_octolapse.log.

seltix5, 2019-09-02:
That starting gcode is the new default for Ender 3 on Cura 4.2. I will check further on next prints but I don't think I noticed any stringing or oozing between the prime line and skirt.

About the snapshots count I did not have much time this morning to analyse it before leaving for work but it this they all there. 

For the quality it not there yet but I was watching the printer this morning and the retraction and z-hop was there in every snapshot, I think now I need to do a 2 tower stringing test because i think it will do the same artfacts. I will test a higher z-hop too because if the snapshop (ideally) is taken during infill, the higher z-hop should give a better result or at least confirm it is taking the snapshot at the right time

seltix5, 2019-09-02:
about another topic ;
in camearas settings > Custom Image Preferences :
- the "Apply All Settings Now" do nothing
- the "Check for Custom Page" shows popup error but it does update the streaming with the new settings
[image]

FormerLurker, 2019-09-02:
@seltix5,

> For the quality it not there yet

Doah!  Well, let me know how the stringing tests go.  I recommend you print two towers at opposite corners to simulate a travel from the center to the edge and back, making sure Octolapse is disabled.  Let me know what you figure out.  

Also, interesting idea about zhop.  I'm interested in every detail you find regarding the stringing test, and if you can tell that Octolapse is taking snapshots at the best time possible for each layer.

FYI, I pushed some updates this AM, mostly dealing with the gcode trigger (it was broken), some small parsing improvements, and logging enhancements (better output gcode logging).  Feel free to install if you want.

seltix5, 2019-09-02:
Sure, I will report my tests results. Happy to help. I will install last version to be sure to have the most recent version.

Is it possible to add custom gcode before each trigger? It wold be helpful to add a pause or other things for debugging. In this case to make sure the trigger happened on the right spot.

We could exange messenger or WhatsApp, it wold facilitate feedback

FormerLurker, 2019-09-02:
Discord?  Here is an invite to the Octolapse private discord server:  https://discord.gg/GCtbfh

Once I see that you've joined, I'll add a role so you can post.  If I don't see you right away I apologize :)

It is possible to add gcode, but only during the snapshot portion.  It's easier to see what's going on by turning on 'test mode - debug issues' debug profile, then looking at the log.  However, that would create a big log that's difficult to read if you don't know how things work exactly.  It might be easier to create a new debug profile, enable 'Additional Logging' then remove any modules that are logging (click all trash can icons) then add the following:

octolapse.gcode -> Verbose

I'm going to add the triggering command line number to the gcode output, then you can just look at the code, refer to your gcode file, and see if the gcode generated by octolapse makes sense.

Also, you can watch the log from a terminal by connecting to your pi, then enter the following commands:

```
cd /home/pi/.octoprint/logs
tail -f plugin_octolapse.log
```

That should make it easy to read along and check out what's going on.  I'll let you know when I add the file line to the gcode logger (this will only work when using one of the smart triggers, since I don't know the file line otherwise).

FormerLurker, 2019-09-02:
Hey, I added that logging.  You should now see something like this if you have logging for octolapse.gcode enabled at the Info, Debug or Verbose levels (in the Octolapse Debug profile under Additional Logging).  Let me know if that helps!

FormerLurker, 2019-09-02:
Ok, regarding the webcam error:  Can you send me input.json?  This file is generated by mjpg-streamer, and can be found at a similar url:  http://{ip_of_pi}/webcam/input.json

It should NOT be changing your settings if this check fails.  I will work on this next.

FormerLurker, 2019-09-02:
Hey, I saw you join and added a role for you.

FormerLurker, 2019-10-07:
@seltix5, doing a bit of issue cleanup, and since this feature was removed and replaced with a better one, I'm going to go ahead and close this now.  Thanks for your post!

closed 2019-10-07 by FormerLurker