FormerLurker/Octolapse #377: [Request] Option to upload timelapse video to YouTube with meta data

iandouglas, 2019-09-01:
#### If this is a feature request describe it here

Once a print is finished, upload video to YouTube as a secret video so it can be edited before publishing. The filename would be the video title, the description would be populated with meta data like the print duration, amount of filament used, etc., and the last timelapse photo taken as the thumbnail of the print.

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-09-01:
Interesting idea!  Isn't there already a YouTube uploader plugin?  If so, maybe i could add a hook and convince the dev to snag octolapses as well?

iandouglas, 2019-09-01:
There might have been one back in the day but a search through octoprint doesn't locate it so it might be deprecated and/or too old.

FormerLurker, 2019-09-02:
Ok, in any case i will add the hook.  This sounds like a great feature for another plugin.  That way it could also move Octoprint style timelapses too!

gege2b, 2019-09-02:
Personnally I use https://github.com/tokland/youtube-upload/ 
It works fine, it should be triggered with a MovieDone event ( https://docs.octoprint.org/en/master/events/#timelapses )

iandouglas, 2019-09-03:
@gege2b that looks great, how do you hook that into your OctoPrint environment to run?

FormerLurker, 2019-09-03:
Use the plugin manager, click 'get more' then use this link to install:  https://github.com/tokland/youtube-upload/archive/master.zip

Just paste it in the 'from url...' box and click install.

I'll need to send the MovieDone event.  This is only done in some situations on v0.3.4, and not at all in v0.4 (pre-release), though I plan to remedy that.

gege2b, 2019-09-03:
@iandouglas take a look at https://github.com/foosel/OctoPrint/wiki/Cookbook:-Useful-Eventhooks

I don't know if it's up to date, but as far as I can remember, that's what I did (I don't have access to my instance right now)

gege2b, 2019-09-18:
Damn, I didn't read your last line @FormerLurker :D 
I just notice my timelapses wasn't uploaded to YT anymore (since end of july, but I didn't print anything until this monday) and was wondering why

FormerLurker, 2019-09-18:
Booo.  I'm not sure what the solution is exactly at this point.  If I send the 'MovieDone' event, it creates another pop up, and some side effects that are undesirable.  I may have to request some changes in OctoPrint to get this working.

gege2b, 2019-09-19:
OK I got it, I can't help...
Just by curiosity, which kind of side effects?

FormerLurker, 2019-09-19:
Hmm..  Actually it looks like there are no other side effects.  However, in looking at the event data, I can't specify a path, only a file name.  When Octolapse gets its own file browser (which it will soon), this will cause additional problems.  I think the solution is to send my own movie done event and figure out how to adjust youtube upload to work for it.  I can't tell from the cookbook you posted how exactly it knows where the movies are located.  Do you know how it does this?

FormerLurker, 2019-09-19:
So, I added a custom PLUGIN_OCTOLAPSE_MOVIE_DONE event which supports the same payload info that the MOVIE_DONE event supports:  gcode, movie, and movie_basename.  This can be installed from the devel branch (if you are not using the latest RC, the upgrade might be quite a shock, fyi... it's way different than the master).

I believe you should be able to follow the recipe posted above, substituting the MOVIE_DONE event with the PLUGIN_OCTOLAPSE_MOVIE_DONE event.

If you are feeling brave, you can install it from this URL:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Please note that that is the devel branch, which is constantly changing.  I will not be making any additional changes until tomorrow afternoon at the earliest.  I will be pushing a new devel/rc version over the next couple days, which will hopefully turn into an actual release candidate.  However, I've been saying this for a while and something always comes up, lol!

FormerLurker, 2019-09-19:
Ok, I was wrong about so many things :)  The event should work, but I DO see where the file name/path comes from and how it's used in the event subscription.  Also, the MOVIE_DONE event DOES hold the full path!  I still think the custom message is the best way to go.  There won't be any extra popup boxes, I will be able to add more metadata that can be used, and I won't have to bother Gina with any feature requests.

gege2b, 2019-09-23:
Oops I'm sorry I missed the notifications :)
indeed the movieDone event carries the full path as well as just the file name
I will test this tomorow if I get some time but it's not guaranteed  ^_^

FormerLurker, 2019-10-16:
Has anyone had a chance to try this out yet?  Let me know if you need any assistance, though I'm not very familiar with the youtube upload plugin.

gege2b, 2019-10-21:
last weeks was insane at work, I don't even had the time to print a single object since my last post
I'll post back here as soon as I tested your suggestion

FormerLurker, 2019-10-21:
Thanks, ive been experiencing a similar situation, and understand.

gege2b, 2019-11-08:
Hi there
I finally got some time to test this, but it seems not to work
I mean, the timelapse is done fine, but either the event isn't triggered, or I messed up something in the event declaration in config.yaml

Here is what I typed : 
```
  - command: '/home/pi/youtube-upload/bin/youtube-upload blah blah blah
    event: PLUGIN_OCTOLAPSE_MOVIE_DONE
    type: system
```
Please can you tell me if this is OK ?

FormerLurker, 2019-11-08:
@gege2b, maybe it is case sensitive?  You could try this for the event:  plugin_octolapse_movie_done

FYI, I'm setting up a subscription now to see if I can detect the event.

FormerLurker, 2019-11-09:
@gege2b, I've verified that the event is being sent.  Now to configure the youtube uploader (not looking forward to that, lol!)

gege2b, 2019-11-13:
Actually I "lowercased" the event handler, but I forgot to restart octoprint, and I was wondering why it still doesn't work... :rofl: 
A print was still in progress when I left my office, so I'll test this tomorow for sure

FormerLurker, 2019-11-13:
Excellent, thank you!  I've been bogged down trying to get DSLR stuff working, and haven't had much time this week.  However, lots of progress on enhanced script logging and fault tolerance, so that's good!  I eagerly await your results!

gege2b, 2019-11-15:
OK, it works with the lowercased event ! thanks !

FormerLurker, 2019-11-16:
Woot!!!  Thanks for testing this!!

FormerLurker, 2020-02-26:
I neglected to close this issue after it was resolved.   Whoops.  Closing now, and thanks!

closed 2020-02-26 by FormerLurker

bote795, 2022-02-25:
sorry  but is there any documentation on how to set this up :D