FormerLurker/Octolapse #379: [Request] Reset Camera Settings

ViolentCrumble, 2019-09-04:
#### If this is a feature request describe it here

Hello can you please add a way to reset camera settings back to the camera default. I have noticed that the default settings on my camera were awesome but i stuffed them up on 2 of my printers by clicking apply. Spent a lot of time tweaking and tuning to try and get a half good image.

Please add a way to reset them back to the camera defaults.. is there anyway for me to do this now?

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-09-04:
Oh man, have I got the fastest feature request resolution ever for you.  Take a look at [this](https://www.youtube.com/watch?v=7Xr2ED5Utak&t=855s).

This version is in beta now.

ViolentCrumble, 2019-09-04:
oh man i think i love you, when can i use it? how do i install the beta?

FormerLurker, 2019-09-05:
Check this out:  https://www.patreon.com/posts/v0-4-rc1-dev2-29678095

FormerLurker, 2019-09-05:
check this out:  https://www.patreon.com/posts/v0-4-rc1-dev2-29678095

ViolentCrumble, 2019-09-05:
> 
> 
> Check this out: https://www.patreon.com/posts/v0-4-rc1-dev2-29678095

great thank you! i assume nothing will break? I use Octolapse now on 5 printers and just wanna make sure this won't clear any of my existing working setup? I just want to reset all the cameras back to their default.

FormerLurker, 2019-09-05:
It may break.  It is beta after all.  Use the octoprint backup, then download  your settings.json for octolapse (octoprint settings->octolapse->download settings button) and all bases should be covered.  Octolapse has never broken octoprint for me (knock on wood), but better safe than sorry.  If octolapse breaks and you have the settings.json file, it will be quite easy to roll back.  Just uninstall octolapse, and reinstall the master from the plugin manager as usual, then copy settings.json back to octolapse directory (home/pi/.octoprint/data/octolapse/) and it will be back to normal.

closed 2019-10-07 by FormerLurker