FormerLurker/Octolapse #380: Fresh Octoprint install, 'Could not parse output from pip' error.

escknx, 2019-09-07:
Freshly flashed Octoprint 1.3.11
Octolapse v0.3.4
Pi 3B

Nothing else is installed, no settings changed, not connected to printer, web camera works, ethernet connectivity

No diagnostic log

Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
Downloading https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip
Collecting pillow (from Octolapse==0.3.4)
Downloading https://files.pythonhosted.org/packages/51/fe/18125dc680720e4c3086dd3f5f95d80057c41ab98326877fc7d3ff6d0ee5/Pillow-6.1.0.tar.gz (33.8MB)
Requirement already satisfied: sarge in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (0.1.5.post0)
Requirement already satisfied: six in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (1.12.0)
Requirement already satisfied: OctoPrint>1.3.8 in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (1.3.11)
Requirement already satisfied: psutil in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.4) (5.6.3)
Requirement already satisfied: flask<0.11,>=0.10.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.1)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.8.1)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (4.5.3)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.2.11)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2019.1.24)
Requirement already satisfied: Flask-Principal<0.5,>=0.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.4.0)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.12.2)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.12)
Requirement already satisfied: werkzeug<0.16,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.15.6)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.1)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (5.1.2)
Requirement already satisfied: markdown<3.1,>=3.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.0.1)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.4)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.7.19)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.9.0)
Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.9)
Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.2.0)
Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (4.0)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.5.0.1)
Requirement already satisfied: requests<3,>=2.21.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.21.0)
Requirement already satisfied: semantic_version<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.6.0)
Requirement already satisfied: Click<8,>=7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (7.0)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.6.5)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (5.2.1)
Requirement already satisfied: chainmap<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.3)
Requirement already satisfied: future<0.18,>=0.17.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.17.1)
Requirement already satisfied: scandir<1.11,>=1.10 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.10.0)
Requirement already satisfied: websocket-client<0.57,>=0.56 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.56.0)
Requirement already satisfied: wrapt<1.12,>=1.11.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.11.2)
Requirement already satisfied: futures<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.2.0)
Requirement already satisfied: emoji<0.6,>=0.5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.5.3)
Requirement already satisfied: monotonic<1.6,>=1.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.5)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.2)
Requirement already satisfied: sentry-sdk==0.7.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.7.7)
Requirement already satisfied: typing<4,>=3.6.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.7.4.1)
Requirement already satisfied: itsdangerous>=0.21 in /home/pi/oprint/lib/python2.7/site-packages (from flask<0.11,>=0.10.1->OctoPrint>1.3.8->Octolapse==0.3.4) (1.1.0)
Requirement already satisfied: MarkupSafe in /home/pi/oprint/lib/python2.7/site-packages (from Jinja2<2.9,>=2.8.1->OctoPrint>1.3.8->Octolapse==0.3.4) (1.1.0)
Requirement already satisfied: singledispatch in /home/pi/oprint/lib/python2.7/site-packages (from tornado==4.5.3->OctoPrint>1.3.8->Octolapse==0.3.4) (3.4.0.3)
Requirement already satisfied: certifi in /home/pi/oprint/lib/python2.7/site-packages (from tornado==4.5.3->OctoPrint>1.3.8->Octolapse==0.3.4) (2018.11.29)
Requirement already satisfied: backports-abc>=0.4 in /home/pi/oprint/lib/python2.7/site-packages (from tornado==4.5.3->OctoPrint>1.3.8->Octolapse==0.3.4) (0.5)
Requirement already satisfied: blinker in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Principal<0.5,>=0.4->OctoPrint>1.3.8->Octolapse==0.3.4) (1.4)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Babel<0.13,>=0.12->OctoPrint>1.3.8->Octolapse==0.3.4) (2.6.0)
Requirement already satisfied: webassets>=0.11.1 in /home/pi/oprint/lib/python2.7/site-packages (from Flask-Assets<0.13,>=0.12->OctoPrint>1.3.8->Octolapse==0.3.4) (0.12.1)
Requirement already satisfied: argh>=0.24.1 in /home/pi/oprint/lib/python2.7/site-packages (from watchdog<0.10,>=0.9.0->OctoPrint>1.3.8->Octolapse==0.3.4) (0.26.2)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from watchdog<0.10,>=0.9.0->OctoPrint>1.3.8->Octolapse==0.3.4) (0.1.2)
Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python2.7/site-packages (from rsa<5,>=4.0->OctoPrint>1.3.8->Octolapse==0.3.4) (0.4.5)
Requirement already satisfied: urllib3<1.25,>=1.21.1 in /home/pi/oprint/lib/python2.7/site-packages (from requests<3,>=2.21.0->OctoPrint>1.3.8->Octolapse==0.3.4) (1.24.1)
[... cut ...]
Command "/home/pi/oprint/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-bnt_vb/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-6_PA7Y/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/pillow" failed with error code 1 in /tmp/pip-install-bnt_vb/pillow/
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

FormerLurker, 2019-09-26:
OMG, I totally missed this issue, so sorry!

Looks like pillow installation problems strike again.  99% of the time this is either an issue with PIP being too old, or undervoltage (it's usually this) due to some power supply problem.

[See this thread for more info]()

You might try installing pillow manually either by connecting to your pi remotely, or by connecting a keyboard and mouse.  Here are the two manual install commands:

Step 1: update pip (the first line is VERY important)
```
source ~/oprint/bin/activate
pip install --upgrade pip
```
Step two (you MUST do step 1 first) - install pillow
```
pip install pillow --no-cache-dir
```

Then try to reinstall Octolapse.  Check for any under voltage issues reported by Octoprint if you have problems.

Again SO SORRY I somehow missed this.

FormerLurker, 2019-10-07:
@escknx, any update about your installation issue?  I'm marking this 'waiting for response' and will assume the issue is resolved if I don't hear back.

Thanks!

escknx, 2019-10-07:
Oh I'm sorry I haven't seen first reply. 
Yes I've fixed it but I can't really recall exactly how. That wasn't voltage issue, I have installed and uninstalled pillow, multiple re-flashes and then planets aligned and installation went thru without any reasonable explanation. It just did.

closed 2019-10-07 by escknx