FormerLurker/Octolapse #381: Unable to find time lapse after print.

ZeekEteek, 2019-09-12:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

Print was finished and octolapse was generating the timelapse. The power went out while rendering.

#### What should have happened?
Timelapse should be showing up under "Finished Timelapses"

#### What happened instead?
It isn't there, and I can't find it anywhere.

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 10

#### Browser and version of browser, operating system running browser

Browser: 76.0.3809.132
Browser OS: Google Chrome

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/fCpPk07r

ZeekEteek, 2019-09-13:
Ok so i was able to get the files over, and now i have all the snapshots. How would I turn them into a timelapse? Thanks for all the help so far.

FormerLurker, 2019-09-16:
No problem!  I'm glad you found your images!  

You can use ffmpeg.  [Here is a guide I found](http://mahugh.com/2015/04/29/creating-time-lapse-videos/) that explains some of this.  There are lots of other programs you can use.  I use sony vegas pro myself (just because I happen to have a copy for other reasons), and I hear others use adobe premier.

FormerLurker, 2019-10-07:
@ZeekEteek, since there are already issues open regarding retrying failed renderings and downloading snapshots, I'm going to close this issue.  Let me know if you want it reopened, and I will do so.  Thanks!

closed 2019-10-07 by FormerLurker