FormerLurker/Octolapse #382: [Request] Post-print script

dudeofawesome, 2019-09-16:
#### If this is a feature request describe it here

I'd really like to be able to execute a script after the print has finished, like the pre-print script.

#### Version of Octolapse
 
Octolapse Version: `0.3.4`

#### Version of OctoPrint

OctoPrint Version: `1.3.11`

FormerLurker, 2019-09-16:
Hi, thanks for posting!  I will take this into consideration.  In the interim, you could install the [Gcode System Commands](https://plugins.octoprint.org/plugins/gcodesystemcommands/) plugin, add a system command gcode to your end gcode or the OctoPrint after print script and run your batch/sh script from that.

FormerLurker, 2019-10-13:
An 'on-print-end' script has been added to the devel branch.  I also fixed an error that was preventing 'on print start' camera scripts from being executed for webcams and gcode cameras.

The on-print-end script will run if the print fails or completes, so I might add an on-print-fail and on-print-complete at some point too.  Let me know if you are able to test this.

dudeofawesome, 2019-10-17:
Sounds great! I'm not currently set up to take timelapses anymore, but once I'm set up again, I'll definitely give it a shot. Thanks!

FormerLurker, 2019-12-16:
I'm going to go ahead and close this now.  If you try the new script and have issues, just add a new issue.  Thanks!

closed 2019-12-16 by FormerLurker