FormerLurker/Octolapse #383: Extruder stops mid print but printing continues

robertwyeoman, 2019-09-22:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. file prints well from SD and from Octoprint and even with octoprint standard timelapse
2. extruder stops after 30-50 layers when octolapse is enabled. 
3. I am not running test mode but did run my last print with logging. 

#### What should have happened?
I should have a completed print and an awesome timelapse

#### What happened instead?
I have the first 45 layers of a print instead

#### Operating System running OctoPrint and Octolapse

OS Name: Octoprint
Os Version:  1.3.10 

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3
Printer Firmware Version: 1.1.5 silent board running 

#### Browser and version of browser, operating system running browser

Browser: Opera
Browser OS: No sure. Most recent version

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-09-22:
My suspicion is that you are getting a jam probably caused by retraction settings.  Does your retraction length in your Octolapse printer profile match your slicer setting EXACTLY?  Every slicer setting in Octolapse should match the slicer settings you used to create your gcode file exactly.

Let me know!

robertwyeoman, 2019-09-22:
I have tried different files and different retraction settings and yes I make sure the retraction setting are exactly the same between the slicer and octolapse.  The extruder just stops turning and just freewheels. I have a knob on my extruder gear and normally I cant turn it mid print because the stepper is active but when this bug happens it has no tension in it.

FormerLurker, 2019-09-22:
Have you tried disabling all other plugins?  Octolapse does not turn off the stepper gears.  Test mode can strip extrusion parameters, but if your extruder isn't engaged at all then something else is going on.

I suspect something other than Octolapse (though surely Octolapse is involved somehow) because the Ender 3 is a very common printer, and I've never heard of this issue.  If you have time to disable all plugins and give it another go, that would be appreciative.

Also, if you see any errors in either octoprint.log or plugin_octolapse.log, those would be VERY useful for debugging since I currently have nothing to go on for tracking this down.

Thanks!

FormerLurker, 2019-09-22:
Oh, welcome to github by the way!  I noticed it's your first issue, which is great!

robertwyeoman, 2019-09-22:
Hey, Thanks! kinda of a long time listener/first time caller situation. The only other pluging I have is the EPROM loader and I haven't even used that yet. I just upgraded my main board to the 1.1.5 silent board and this problem jumped up at me. I am reviewing the logs but I dont see any indication of an error.

FormerLurker, 2019-09-22:
Were you having this problem before you upgraded, or did you not try octolapse before this?

Please attach settings.json (can get from OctoPrint Settings->Octolapse->Download settings button), and plugin_octolapse.log.  I prefer gist.github.com for both of these files please, though I'll take whatever I can get :)

I'm about out of time for the week, fyi, so it might be a bit until I can dig in.  I have been working on the next version for a long while, though, so maybe some problem DOES exist, but it's been fixed in the alpha?  Anyway, if upgrading is an option that might be another thing to test later.

Thanks!

robertwyeoman, 2019-09-22:
I was not having this problem before I upgraded the mainboard. 

https://gist.github.com/robertwyeoman/6e7506cd8c5e52556645089702ae6e6d

FormerLurker, 2019-09-22:
Can you also post the gcode file you were trying to print?

robertwyeoman, 2019-09-22:
Ok I updated the Gist

FormerLurker, 2019-09-22:
So, I found several things wrong with your slicer settings.  

Here is what I extracted from your settings file (it might look different than your version of Ocotlapse, I'm using an alpha)

[image]

I have highlighted settings issues.  Notice that your retraction speed is 60mm/min, which is way too slow.  That certainly could cause you printing issues.  You probably got the units mixed up (mm/s vs mm/min).  Your gcode indicates that your slicer settings for that file are as follows:

```
retraction length = 6.0 mm
Movement Speed (travel) = 9000mm/min
Retraction Speed = 1500mm/min
Deretraction Speed = 1500mm/min
z-movement speed = 600mm/min
```

For now just update the values in your slicer to the values listed above.  After you run a test (but not before), you might want to switch the slicer type to Cura, which would make it easier to copy the settings.  Don't do that before trying the above, though, so we can verify the issue.

Let me know how it goes!

robertwyeoman, 2019-09-23:
I wanted to thank you for all your help so far! It did better but only made it through about 200 layers before it stopped extruding. That is still a huge improvement over all my previous prints by a factor of about 4. I am going to take a bit to copy exactly my Cura Settings for my next print over. Is there anything else I should look for with my settings?

FormerLurker, 2019-09-23:
I'm happy to help!

If I were you I might try reducing the retraction length somewhat.  You might consider doing a few stringing tests to home in on it.  I've seen numbers from 2mm-5mm on the ender 3 depending on the filament and printer upgrades.  Some people have better luck with slower retraction speeds as well, and don't forget temperature!  Once you get that all dialed in, use those settings for Octolapse.

I think I can confidently say that retraction was/is causing a jam, which is why your printer stopped printing.  Maybe go to some ender 3 form or subreddit and see if you can get some advice on your temp/retraction length and speed.  Let me know what you end up doing!

KaminoKGY, 2019-09-24:
@FormerLurker @robertwyeoman 
Oh I am having the same issue on my side ! Happy (or not) to see that I am not the only one. So I would say that it is coming from the Creality 1.1.5 motherboard.

@robertwyeoman have you tried contacting Creality ? I could get an answer from them...

FormerLurker, 2019-09-24:
@KaminoKGY and @robertwyeoman, I have asked creality for some development units several times and I have so far been ignored.  If you contact them, maybe bring this up.  It would be so much easier to debug this if I had a machine to test on.  I don't get a lot of issues for Prusa's printers, but I get a TON from Ender3 and CR-10 users.

KaminoKGY, 2019-09-24:
Ah ah I would gladly trade my Ender 3 for a Prusa but I am not sure that they would agree knowing the price difference ^^
I will try to get a feedback from them and see from there if I can get you some ;) I hope that they won't think that I am a scam or something :P

robertwyeoman, 2019-09-24:
So after adjusting my slicer to Cura and mirroring all my slicer settings exactly I am getting a full print and full timelapse. My print is a little rough with some light stringing and zits (more rough than the sd or regular octoprint prints) but it did turn out. I am going to run some other, longer files to test and see if this problem pops its head up again or not.

FormerLurker, 2019-09-24:
@robertwyeoman, obviously getting a print to complete is the first and most important step, so congratulations!  Hopefully it holds out on the longer prints.  We can work on your print quality once you verify that a longer print works as expected.

You'll be pleased to know that Octolapse v0.4 will eliminate all issues related to copying settings from your slicer due to a new 'Automatic Slicer Configuration' option!  Cura requires a small script to be added to the start gcode to get it working, but it's pretty simple, and I have detailed guides.  It works out of the box with Slic3r, Slic3rPE, PrusaSlicer and Simplify 3d.  If I can ever deal with my regular workload I may get around to releasing it eventually, lol!

FormerLurker, 2019-10-07:
@robertwyeoman, I'm going to go ahead and close this since your initial issue has apparently been solved.  Let me know if you want me to reopen it.  Thanks!

closed 2019-10-07 by FormerLurker