FormerLurker/Octolapse #384: how to change full HD resolution？

bloodywinggithub, 2019-09-24:
can anyone tell me how to change the resolution?

raspberry pi 3b+ with octopi 0.16 and pi cam v2 noIR.

the time lapse file is 640X480, how to change to 1920X1080?

THX

bloodywinggithub, 2019-09-25:
> 
> 
> Octolapse will render the video in whatever resolution your camera is currently set for. [Here is a youtube video](https://www.youtube.com/watch?v=K1F7n5JqURM) I found detailing one way of doing this. I hope that helps!

 It's a little strange.  in the octopi.txt 
camera_raspi_options=""
what's the  correct ptions？  “-fps 15 -x 1280 -y 720”  and “-r 1280x720 -f 15”  can get different results

FormerLurker, 2019-09-25:
-r and -fps don't work on the raspberry pi camera.  Try this:

-x 1280 -y 720 -fps 20

Also, you may want to post at discourse.octoprint.org for help configuring octoprint or octopi.  Even though I wrote a plug-in for octoprint, I'm nowhere near as knowledge as dome of the folks there.

That being said, let me know if those parameters work for you.

FormerLurker, 2019-09-25:
Hey, i made a typo in my last message (since fixed).  -fps controls the frame rate for the raspberry pi cam

bloodywinggithub, 2019-09-26:
It's woking now with -x -y -fps. But I don't know if -quailty option is useful. thx

FormerLurker, 2019-09-26:
@bloodywinggithub, not sure what that does exactly.  I'm not very familiar with the options.  I recommend looking [here](https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/README.md) for info.

FormerLurker, 2019-10-07:
@bloodywinggithub, since this isn't an Octolapse issue, and is related more to OctoPi and mjpgstreamer settings, I'm going to go ahead and close the issue.  Let me kow if you want me to reopen it.  Thanks!

30wolf03, 2020-12-30:
Hello, I have set the camera settings in the octopi.txt to 1920x1080 but the Octolapse footage still looks grainy.

FormerLurker, 2021-01-05:
Have you looked into the custom image preferences?  See [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera), then check this setting:

[image]

That makes a HUGE difference.

30wolf03, 2021-01-06:
Thanks, but never mind. If I have to disable The Spaghetti Detective for that, I'd rather live with grainy time lapses...

FormerLurker, 2021-01-06:
@30wolf03, you do NOT need to disable TSD for that, just check the options and change the the streaming server to compatibility mode.

FormerLurker, 2021-01-06:
Oh, also, I think you can also create a bash script to make the same adjustment.

30wolf03, 2021-01-06:
Where can I find that option?

FormerLurker, 2021-01-06:
I'm not sure exactly since I don't use TSD, but the creator and I have had several discussions (very nice person), and I know this option was added to TSD.  He implemented a custom streaming server to compress the stream, but as a side effect it doesn't work with any mjpgstreamer specific functionality, and it increases snapshot image acquisition time (substantially), both are bad for Octolapse.  The 'compatibility mode' (not actually sure that's what it is called) resolves the issue but loses the benefits of a more compressed stream (useful especially for OctoprintAnywhere, another of his plugins).

Anyway, there can't be too many options in TSD are there (lol, famous last words)?  It should have some documentation explaining what it does.  Take a screen cap of the settings if you can't find it and I'll see if I can identify it.

30wolf03, 2021-01-06:
Ah thanks, I found it. I wasn't sure whether it was an Octolapse or TSD option. And since I had TSD disabled the options for that didn't show up in OctoPrint, so I was searching in the TSD web app. I didn't find it there and decided to google TSD compatibility mode where I got the hint to enable the plugin and go to the advanced settings for TSD.
[image]

FormerLurker, 2021-01-06:
Cool!  Let me know if you have any issues with the image preferences screen.  I find them to be very useful personally.  Also, fyi, I'm working on a way to work around mjpgstreamer and change the settings of the camera directly.  this would simplify a lot of things for Octolapse, but it will take a while.

FormerLurker, 2021-01-06:
Hey, thanks for sponsoring me!  That was very thoughtful, highly appreciated, and keeps me motivated to continue making more software  :)

I will leave this open until your image is high quality.

30wolf03, 2021-01-06:
I found it very nice of you to take the time to explain to me how to make it work and to give me the hint that I don't have to give up on TSD so take the Sponsoring as a “Thank You”. 
I am printing something random now to find out if it works.

30wolf03, 2021-01-07:
So, I can't find the codec controls to change the bitrate, but I tried to increase the jpeg quality and that didn't really make a noticeable difference...

FormerLurker, 2021-01-07:
Jpg quality seems to have no effect.  Can you screencap all the available controls please?

30wolf03, 2021-01-08:
[image]

FormerLurker, 2021-01-08:
Hey, I thought this was a pi cam v2 noIR!?  I think I've figured out what was going on initially, lol!  Do you have 2 cameras?  If so, you might have been changing the settings on the wrong camera (usually 2nd camera is on port 8081, but it can be anything).

Anyway, regardless, the 920 does not have any video bitrate control, so we're sort of back to square 1.  I think you will appreciate the focus, white balance, and exposure controls.  I recommend setting all those to manual per [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls#step-7---dialing-in-your-camera), as you will get MUCH better results.

Now, for your resolution (though it looks like 1080p to me, but just in case), that needs to be set in octopi.txt.  If you need help getting to that, let me know.  Turns out, we have the same cameras, so this should be easy :)  Here is my octopi.txt file:

[image]

If your timelapse videos are grainy but your stream is not, edit your rendering profile within Octolapse and bump up the bitrate a bit:

[image]

If your camera stream quality is not as good as you'd like, I recommend trying to increase the brightness.  I use a 65W photography LED array, and it makes a HUGE HUGE difference.  Some LEDs use something called pulse width modulation (PWM) to control the brightness, or to reduce power consumption, and it basically turns the LED on and off rapidly.  If the PWM frequency is low, you will get a wavy looking timelapse (even if the stream does not look wavy, though it could).  Good PWM operates in the 10,000+ HZ range, and bad PWM runs at power line frequency (50-60HZ).  Somewhere in-between is fine too.  When in doubt, try using incandescent or halogen lighting as a test, and turn your LEDs off if you notice flickery wavy timelapse videos.  This is assuming you are using manual white balance and exposure (auto settings can cause issues too).

If none of this is on-target, let me know exactly what the image quality issue is, and we can go from there.

Thanks!

30wolf03, 2021-01-08:
OK, thank you very much for your help now my image just seems to be out of focus. I'm sure I'll find a way around this issue.

FormerLurker, 2021-01-08:
Yeah, I believe I mention somewhere in the guide how I focus.  It is a bit tricky, but once you figure it out it's def better than auto-focus. Let me know if I can help at all.