FormerLurker/Octolapse #386: Timelapse doesn't start to render

XilefTech, 2019-09-26:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.11 

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. starting a print with octolapse
2. letting the print finish
3. tried to let octolapse render the timelapse

#### What should have happened?
The timelapse should have started to render immediately after finishing the print

#### What happened instead?
Octolapse was still in '"running" mode and after i pressed the stop button it just said that it waits for the print to finish for rendering the timelapse, even though the print already finished about an hour or so ago. After a reboot of the Pi, Octolapse was back in idle mode, but the timelapse still wasn't rendered. Luckily I was able to save the images from the pi to make the Timelapse with an external program.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version
Printer Model: CR10s
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser
Browser: Chrome
Browser OS: Windows

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://drive.google.com/open?id=1tuWeNgJ-sk8v12_f62F4ZQdBei4ZP8lq

#### Link to settings.json
Link to settings.json with all passwords removed: https://drive.google.com/open?id=1Skv6RNZ4Qj9k2-k7huRpmcQe-QycHcjd

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://drive.google.com/open?id=1Js6zx9kRxEz3u0w83ZqyBGjNZZ6S3u1-

#### Link to octoprint.log
Link to octoprint.log:  https://drive.google.com/open?id=1fFVgWqlOY7RcadSXqu1wXpYyrKwyoQQ-

#### Link to contents of Javascript console in the browser
Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-09-26:
Do you mean you clicked the 'Stop Timelapse' button?  Unfortunately it looks like diagnostic logging was not enabled, so I can't tell much from your logs.  If you change your 'Debug' profile to 'Full Diagnostic' mode and try again, I should have the info I need.

However, are you sure snapshots were captured (at least 2) before you stopped the timelapse?  Also, did you check the 'Timelapse' tab to make sure the timelapse actually got rendered?  There are a few UI issues in Octolapse 0.3.4 that might have prevented the timelapse state from being updated.  I'm pretty sure all these issues are fixed in the alpha version of 0.4, which features a much enhanced rendering process.

XilefTech, 2019-09-28:
Hi

Thaks for the quick reply.

Yes, I checked the timelapse tab and also searched on the internet for a
solution.
As i described I was able to rescue all images via FileZilla so I'm sure
that there were over 100 pics.
Next time I'll enable logging and reply with the log.

Have a nice Day
Felix

FormerLurker, 2019-09-28:
Ug.. i obviously did not read your issue properly, sorry about that.  I've been rushing things recently due to an extreme workload.  I expect work to start slowing down the week after next (thankfully), so I will be able to give your issue (and others) the time they deserve.

XilefTech, 2019-10-11:
Hi again!

I recently found time to print something again and this time with
diagnostic mode enabled. I got the same Problem.
I hope you can use the log to fix the problem or tell me what I was doing
wrong.

Have a nice Day!
Felix

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
11.10.19,
21:08:34

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virenfrei.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

FormerLurker, 2019-10-13:
So, I think this has been fixed in v0.4.  Please try installing it from this link:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Then follow [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) which explains how to install the new version (but be sure to use the link above to install and not the link found within the guide).  If you have any problems installing or configuring the new version, let me know!

Once that's finished, run a small print (maybe a calibration cube) using the 'Test Mode - Debug Issues' debug profile.  Your printer will not warm up, nor will it extrude any plastic when using this mode, but a full log will be created.  If you still have problems rendering, upload plugin_octolapse.log to gist.github.com and add a link here so that I can take a look.

Thanks, and let me know how it goes!

XilefTech, 2019-10-13:
Hi!

Thanks for that tip. It solved the problem right away! Now I'll be able to
have even more fun with these Timelapses!
Because I'm just a student with not very much money, I sadly can't donate
some money to you for this project but I want to say thank You for offering
this Plugin for free!
It's my Top 1 Plugin for Octoprint and it will forever be.
And thanks again for the great support.

Felix

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
13.10.19,
22:15:04

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virenfrei.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

FormerLurker, 2019-10-13:
@XilefTech, I'm glad the new version is working for you!  Let me know what you think of all of the new features, or if anything needs to be improved.  I pushed a few more updates today (added popup help for common print start issues), so there is still incremental improvement!

> Because I'm just a student with not very much money, I sadly can't donate
some money to you for this project

That's OK!  I was in your boat for a very long time, so I understand completely.  Besides, you are helping me test the next version, and that is invaluable.  Just be sure to report any bugs you find and provide any feedback you may have.  I'm going to drop v0.4rc1.dev3 soon, and will create a feedback thread for that.  

If you want to be notified via Octoprint when the new devel/rc version drops, open up the Software Update plugin (OctoPrint Settings->Software Update), edit the settings (wrench/spanner icon), and then change the OctoPrint version tracking to **Release* and then select the **Devel RCs** OctoPrint Release Channel.

> but I want to say thank You for offering this Plugin for free!

My pleasure, I'm glad you like it.

> It's my Top 1 Plugin for Octoprint and it will forever be.

You are too kind :)

FormerLurker, 2019-10-16:
So, it looks like this issue is closed!  It was a pleasure :)  Feel free to create another issue if you find any more bugs.

Closed!!!

closed 2019-10-16 by FormerLurker