FormerLurker/Octolapse #389: snapshot thumbnail create error

kanook450, 2019-10-12:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

OctoPrint 1.3.11 running on OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

yes
#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

I'm having this issue. I'm very new to the pi environment and followed your guide. the camera is recognized and snapping pictures at the correct point but I keep getting this error. any help would be appreciated. 
snapshot-thumbnail-create-error: Create Thumbnail - An unexpected exception occurred while creating a snapshot thumbnail. Check the log file (plugin_octolapse.log) for details. - Inner Exception: cannot identify image file '/home/pi/.octoprint/data/octolapse/snapshots/latest_ff25889b-2a3a-4eeb-a78a-9a865f1b6a9e.jpeg'
#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE
OctoPi 0.16.0
macOS
catalina 10.15

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE
ender 3
stock firmware 1.1.6
[octoprint-2.log](https://github.com/FormerLurker/Octolapse/files/3720447/octoprint-2.log)

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-10-12:
So, I don't have the Octolapse log, so I can't be 100% sure, but your PI is reporting either undervoltage or overheating errors (probably undervoltage).  This has been known to affect Pillow, which is what generates the thumbnails.  If I were you I'd first try to fix that problem (probably by replacing your PSU) and then try Octolapse again.  In the meanwhile if you could set your Octolapse Debug Profile to 'Full Diagnostic', then run about 3-4 layers of a print (you can cancel after that).  Attach the log and I should be able to see the full error report.

Thanks for posting!

kanook450, 2019-10-12:
Thanks. I did try to drag and drop my log file but I guess that didn’t work. I’ll try your suggestions and give it another go. Ill try to provide the log info too. Thanks so much for the support you provide. 
Schylar

kanook450, 2019-10-13:
Ok here is hopefully some data from a test print I just tried. Hopefully this comes to you. 
Thanks

kanook450, 2019-10-13:
2019-10-11 20:23:44,679 - octoprint.startup - INFO - ******************************************************************************
2019-10-11 20:23:44,681 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-10-11 20:23:44,682 - octoprint.startup - INFO - ******************************************************************************
2019-10-11 20:23:45,836 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-10-11 20:23:48,192 - octoprint.plugin.core - INFO - Found 15 plugin(s) providing 15 mixin implementations, 17 hook handlers
2019-10-11 20:23:48,320 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-10-11 20:23:48,397 - octoprint.server - INFO - Intermediary server started
2019-10-11 20:23:48,399 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-10-11 20:23:48,918 - octoprint.plugin.core - INFO - Found 15 plugin(s) providing 15 mixin implementations, 17 hook handlers
2019-10-11 20:23:48,921 - octoprint.printer.profile - ERROR - Profile _default does not exist, creating _default again and setting it as default
2019-10-11 20:23:49,049 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-10-11 20:23:49,050 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-10-11 20:23:50,903 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2019-10-11 20:23:52,309 - octoprint.util.pip - INFO - Version of pip is 19.0.1
2019-10-11 20:23:52,310 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-10-11 20:23:52,311 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-10-11 20:23:52,316 - octoprint.plugin.core - INFO - Initialized 15 plugin implementation(s)
2019-10-11 20:23:52,417 - octoprint.plugin.core - INFO - 15 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2019-10-11 20:23:52,438 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 918188032
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.16.0
|  python:
|    pip: 19.0.1
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
2019-10-11 20:23:52,446 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2019-10-11 20:23:52,447 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2019-10-11 20:23:52,801 - octoprint.server - INFO - Shutting down intermediary server...
2019-10-11 20:23:52,923 - octoprint.server - INFO - Intermediary server shut down
2019-10-11 20:23:52,926 - octoprint.events - INFO - Processing startup event, this is our first event
2019-10-11 20:23:52,926 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2019-10-11 20:23:52,929 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2019-10-11 20:23:52,934 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2019-10-11 20:23:52,949 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on octopi' for _http._tcp
2019-10-11 20:23:52,960 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on octopi' for _octoprint._tcp
[... cut ...]
2019-10-13 01:23:49,343 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: swatch_band_adapter.gcode, owner: Schylar, user: Schylar
2019-10-13 01:23:49,358 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-10-13 01:23:49,383 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2019-10-13 01:23:50,112 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': 'local', 'file': 'e900fb2ec342fb603cd28cc584dafb40788943db'}

FormerLurker, 2019-10-13:
Wow, that's a whole lot of info to paste into an issue!  Next time please post this kind of info as an attachment.  I'm going to try to move your previous message to gist.github.com, which is a good place to put these kinds of things.

FormerLurker, 2019-10-13:
So, the log you posted doesn't contain diagnostic information.  Make sure you change your Octolapse debug profile to 'full diagnostic' and try the test again.  Then download plugin_octolapse.log and upload it to gist.github.com and post the link here please.

Thanks!

kanook450, 2019-10-13:
its clear i don't know what I'm doing. Not my first 3d printer but my first time working with Raspberry Pi. hopefully I did this correctly now.
Schylar
<script src="https://gist.github.com/kanook450/e352602730d1d82352a750486e206196.js"></script>

FormerLurker, 2019-10-13:
LOL, no worries.  It looks like you have 2 cameras going, one DSLR and one webcam.  Looks like the webcam is working, so the problem must be with your /home/pi/scripts/take-snapshot.sh script.  It looks like Octolapse successfully executed the snapshot script, and that an image file was downloaded.  However, pillow (the image library used for manipulating image files) was not able to open the file.

My guess is that your DSLR camera is returning a .raw file.  See if you can switch your camera to only return .jpg images.  My DSLR (D5200) camera can be set to create .raw files, .jpg files, or both, so maybe yours is similar.  Octolapse only supports .jpg files at the moment.  I should probably update the wiki to make this clear.

DSLRs can be a bit tricky.  Every one is a bit different, and not all cameras are supported by gphoto2.  Keep trying through, because most cameras will eventually work :)

Let me know how it goes!

kanook450, 2019-10-13:
Damn that’s probably it. I normally shoot raw. Ill change those settings on my camera.
Thanks so much

FormerLurker, 2019-10-16:
@kanook450 , have you tried switching to jpg yet?  I'm curious if this was your issue.  I'm going to mark this waiting for response and possible resolution so others can see that this is being handled.

Thanks!

kanook450, 2019-10-16:
Yes sorry I got busy with work but that does appear to have been my issue. Thanks again for all you have done. Can you point me to a guide to saving the images to the camera? Also do you have a tip jar anywhere or just your Patreon?

FormerLurker, 2019-10-16:
>> Can you point me to a guide to saving the images to the camera?
There are two options:

1.  [Only save the images on the SD card.](https://github.com/FormerLurker/Octolapse/wiki/DSLR-Image-Capture-With-No-Download)
2.  [Defer download until after the print is completed, then render on the pi.](https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download)

For option 1, make sure you disable rendering else you'll get a rendering error at the end of each print.  The error is annoying, but it is not an actual problem.

Option 2 is a bit harder to get working, but is my favorite.  I like things to be automated :)

Both of those options will MASSIVELY reduce the image acquisition time.  Note that not all cameras support option 1, and fewer cameras support option 2.  Be prepared for some trial and error, and be sure to test everything in gphoto2 before attempting anything inside Octolapse.

>> Also do you have a tip jar anywhere or just your Patreon?

You are too kind :)  Unfortunately I only have Patreon ATM.  I really need to add some paypal links don't I?  I will try to make some time for that.  I really appreciate the though in any case!

FormerLurker, 2019-10-16:
I'm going to go ahead and mark this closed.  If you encounter bugs with the no download/deferred download scripts go ahead and create a new issue please.  Keep in mind I'm not an the different cameras and how compatible they are with gphoto2, but I'll help where I can.

Closed!!!

closed 2019-10-16 by FormerLurker