FormerLurker/Octolapse #391: Profile Update Error:INTERNAL SERVER ERROR

EJtechandDiy, 2019-10-16:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.0rc1.dev2

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.11

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. First installed Octolapse with the link that was provided by Lurker 
2. Reboot Octoprint
3. A message started popping out after reboot asking me update profiles

#### What should have happened?
It should update the profiles when clicking update

#### What happened instead?
I got a error saying : Profile Update Error Octolapse was unable to update your profiles to the most recent version the current settings. Status: error. Error: INTERNAL SERVER ERROR

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Modded creality CR-10, 
Printer Firmware Version: marlin 1.9

#### Browser and version of browser, operating system running browser

Browser: Brave Version 0.69.135 
Browser OS: chromium based

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: 
[Octolapse-simplify3d.zip](https://github.com/FormerLurker/Octolapse/files/3736522/Octolapse-simplify3d.zip)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:
[octoprint v.zip](https://github.com/FormerLurker/Octolapse/files/3736538/octoprint.v.zip)

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links:
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

EJtechandDiy, 2019-10-16:
Also with the link that you provided the previews problem octolapse not starting has been solved.

FormerLurker, 2019-10-16:
Did you try restoring the default settings?  If not, first export your current settings then restore the defaults.  There were some issues in one of the previous RCs related to bad guids.

EJtechandDiy, 2019-10-16:
Thank you, that fixed the problem.
I have one last question for you and sorry for asking so much. I upgraded to a Bondtech Direct drive and now I cant get rid of this artifact 
[image]
 I try to get the retraction ass far as possible(3mm E3D v6) but they still there. should I change something else other that retraction to try to fix this problem

Im using a Nikon D5300 and saving the pictures on the SD card here is the camera file:
[Nikon octolapse.zip](https://github.com/FormerLurker/Octolapse/files/3736677/Nikon.octolapse.zip)

Again thank you for your support.

EJtechandDiy, 2019-10-16:
The problem 
[image]
Still there, the problem is with the DSLR profile when I try to use one a pre-configured it asked me to update the profile and then I get the error.

FormerLurker, 2019-10-16:
Hmm.  I will look into the profile update thing.  Can you export and post your current settings?

EJtechandDiy, 2019-10-16:
Camera settings:
[Nikon Octolapse.zip](https://github.com/FormerLurker/Octolapse/files/3736801/Nikon.Octolapse.zip)

Octolapse settings: 
[Settings Octolapse0.4.json.zip](https://github.com/FormerLurker/Octolapse/files/3736805/Settings.Octolapse0.4.json.zip)

Also about the artifacts I was looking at the nozzle while printing and I cold see that when the nozzle moves **center left** the extruder start ozing a little bit of filament, when the camera finish taking the picture and it goes back to print, there is where the plastic sticks back on the print wall. also I tested a stringing test without Octolapse and it was perfect.

FormerLurker, 2019-10-16:
How long does your Dslr take to snap a pic?  Maybe try with a webcam as a test?

For your stringing test, slow down travel so that the movement between the pillars takes the same amount of time as an octolapse snapshot with your dslr and let me know if there is still no stringing.  I will try to take a look at your settings asap.

EJtechandDiy, 2019-10-17:
I will say as soon as the nozzle hits the target on the left corner, the camera takes a picture i will say 1 second, also its there a way to make it take the snapshot from the infill of the part and not the outer wall?

FormerLurker, 2019-10-17:
Yes there is, that's why i wanted your settings :) in the meantime, take a look at [this](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) guide.  Id appreciate any feedback relating to the guide if you have any!

FormerLurker, 2019-10-17:
Hey, can you also upload the sliced gcode file from that print that is stringing?  I'm looking at your settings now, and so far so good.

FormerLurker, 2019-10-17:
So, the first thing I notice is that zhop is not enabled.  That's probably why the oozed filament is sticking right on the edge of the print like that.  Add like 5mm of zlift and see what that does.

I will verify this once I've seen your gcode file (it might not be today since I'm about to retire for the evening), but octolapse should already be traveling to take a snapshot while printing infill.  Again, the zhop thing might make it look like it is not.

Also, that is some incredible oozing.  I was looking on mobile earlier and didn't see how long each piece is.  Not sure what the answer is, but when I see that the first thing I think is 'CR-10/Ender3'.  Not sure why this is such a common thing with those printers.  At some point I'll have to sit down with one and spend a few days running tests.  I have tried to contact Creality to get a machine for testing, but they responded saying they would look into it and have since completely ignored my emails.

EJtechandDiy, 2019-10-17:
I will check that in the morning its 2:30 AM here and I need to go to work in like 4 hours(why I do this to my self? lol).

here is the file: 
[20mm_calibration_cubev8.zip](https://github.com/FormerLurker/Octolapse/files/3736918/20mm_calibration_cubev8.zip)

Also oozing is not that long, I watched the hole print happening, and what is doing is every time the nozzle comes back to print and the little bit of oozing stick to the tip sticks to the one that was there before, that is how it gets so long.

Thanks for everything.

FormerLurker, 2019-10-17:
Wow, get some sleep!  However, I do understand obsession, which is basically what Octolapse started as.  Now I'm a bit more balanced :)

FormerLurker, 2019-10-17:
So, I ran your gcode through the debugger and see that it is indeed taking snapshots while printing infill on all but the first layer.  Here is the generated snapshot plan:
```
Plan# 1, Layer:1, Line:207, StartX:0.1, StartY:200, StartZ:0.3, Speed:1500, Travel Distance:100.168, Type:fastest_extrusion, Gcode:G1 X0.1 Y200.0 Z0.3 F1500.0 E15, Comment: draw 1st line
Plan# 2, Layer:2, Line:478, StartX:141.113, StartY:150.062, StartZ:0.41, Speed:2400, Travel Distance:276.226, Type:feature-solid_infill_feature, Gcode:G1 X141.113 Y150.062 E23.0328
Plan# 3, Layer:3, Line:563, StartX:141.113, StartY:149.938, StartZ:0.61, Speed:2400, Travel Distance:276.226, Type:feature-solid_infill_feature, Gcode:G1 X141.113 Y149.938 E3.3292
Plan# 4, Layer:4, Line:677, StartX:143.212, StartY:150, StartZ:0.81, Speed:1368, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 5, Layer:5, Line:735, StartX:141.113, StartY:147.901, StartZ:1.01, Speed:1368, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 6, Layer:6, Line:785, StartX:143.212, StartY:150, StartZ:1.21, Speed:1368, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 7, Layer:7, Line:843, StartX:141.113, StartY:147.901, StartZ:1.41, Speed:1368, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 8, Layer:8, Line:893, StartX:143.212, StartY:150, StartZ:1.61, Speed:1368, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 9, Layer:9, Line:951, StartX:141.113, StartY:147.901, StartZ:1.81, Speed:1368, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 10, Layer:10, Line:1001, StartX:143.212, StartY:150, StartZ:2.01, Speed:1368, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 11, Layer:11, Line:1059, StartX:141.113, StartY:147.901, StartZ:2.21, Speed:1368, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 12, Layer:12, Line:1109, StartX:143.212, StartY:150, StartZ:2.41, Speed:1368, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 13, Layer:13, Line:1167, StartX:141.113, StartY:147.901, StartZ:2.61, Speed:1368, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 14, Layer:14, Line:1217, StartX:143.212, StartY:150, StartZ:2.81, Speed:1368, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 15, Layer:15, Line:1275, StartX:141.113, StartY:147.901, StartZ:3.01, Speed:1368, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 16, Layer:16, Line:1325, StartX:143.212, StartY:150, StartZ:3.21, Speed:1368, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 17, Layer:17, Line:1383, StartX:141.113, StartY:147.901, StartZ:3.41, Speed:1368, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 18, Layer:18, Line:1478, StartX:143.212, StartY:150, StartZ:3.61, Speed:1467, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 19, Layer:19, Line:1588, StartX:141.113, StartY:147.901, StartZ:3.81, Speed:1469, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 20, Layer:20, Line:1683, StartX:143.212, StartY:150, StartZ:4.01, Speed:1474, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 21, Layer:21, Line:1761, StartX:141.113, StartY:147.901, StartZ:4.21, Speed:1415, Travel Distance:276.258, Type:feature-infill_feature, Gcode:G1 X141.113 Y147.901 E1.2359
Plan# 22, Layer:22, Line:1827, StartX:143.212, StartY:150, StartZ:4.41, Speed:1418, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 23, Layer:23, Line:1928, StartX:141.113, StartY:152.824, StartZ:4.61, Speed:1458, Travel Distance:276.284, Type:feature-infill_feature, Gcode:G1 X141.113 Y152.824 E1.0454
Plan# 24, Layer:24, Line:2019, StartX:143.212, StartY:150, StartZ:4.81, Speed:1467, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 25, Layer:25, Line:2120, StartX:141.113, StartY:152.824, StartZ:5.01, Speed:1458, Travel Distance:276.284, Type:feature-infill_feature, Gcode:G1 X141.113 Y152.824 E1.0454
Plan# 26, Layer:26, Line:2200, StartX:143.212, StartY:150, StartZ:5.21, Speed:1445, Travel Distance:280.424, Type:feature-infill_feature, Gcode:G1 X143.212 Y150.000 E0.4866
Plan# 27, Layer:27, Line:2286, StartX:141.113, StartY:152.824, StartZ:5.41, Speed:1436, Travel Distance:276.284, Type:feature-infill_feature, Gcode:G1 X141.113 Y152.824 E1.0454
Plan# 28, Layer:28, Line:2482, StartX:141.113, StartY:146.668, StartZ:5.61, Speed:2400, Travel Distance:276.306, Type:feature-solid_infill_feature, Gcode:G1 X141.113 Y146.668 E5.3242
Plan# 29, Layer:29, Line:2576, StartX:141.113, StartY:152.824, StartZ:5.81, Speed:2400, Travel Distance:276.284, Type:feature-solid_infill_feature, Gcode:G1 X141.113 Y152.824 E1.6292
Plan# 30, Layer:30, Line:2834, StartX:141.113, StartY:146.668, StartZ:6.01, Speed:2400, Travel Distance:276.306, Type:feature-solid_infill_feature, Gcode:G1 X141.113 Y146.668 E5.3017
```
[... cut ...] deretraction override setting that allows Octolapse to use different settings to generate snapshot gcode than the current slicer settings.  I've been thinking about this, but have been hesitant to add more settings that further complicate the setup.  However, situations like this (zlift disabled in slicer, required by octolapse to prevent artifacts) suggest I should rethink this.

I'm REALLY curious to know what happens when you turn on zlift.  If it doesn't help, maybe I won't need to add this.

EJtechandDiy, 2019-10-17:
Well this are all the test I runned:
[image]

After many test with Zhop, from 0.05 to 0.50 It was helping but when I got around 50 It started to affect the quality so I when down to 0.20 and reduced the temp from 210 to 200 and and this is why I got:
[image]

Oww Im also using smart high-quality.

EJtechandDiy, 2019-10-17:
Also my retraction is set now to 3mm, I cant go higher than that with a E3D v6.

And those little artifacts are from when the nozzle gets back to printing after taking the snapshot, the oozing get stock to the print.

You can latterly see how the nozzle starts oozing on almost every snapshot.

FormerLurker, 2019-10-19:
@EJtechandDiy, that does look much better.  I'm not sure if this is an octolapse problem, or a slicer settings/printer problem.  Somehow we need to distinguish between the two.  It's definitely possible that I could be doing something different in Octolapse that would reduce these issues, but I don't know what those things are.  If you have some ideas, and some way to test them, I'd be totally willing to experiment.

Please try the following and let me know the results:

1.  Try stabilizing in the center of your bed.
2.  Try switching to a webcam and see if the reduction in delay helps.
3.  You could try switching to a [no-download](https://github.com/FormerLurker/Octolapse/wiki/DSLR-Image-Capture-With-No-Download) or [deferred-download](https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download) snapshot script if you haven't already.  
4.  Not sure what your feedrate is, but you could try turning it down to 90% as a test.
5.  You could try further reducing your temps to 195C
6.  You could try increasing and decreasing your retraction speed.  Too fast and too slow are bad.

Note:  For items 4,5, and 6, only do these one at a time.  If things improve, keep that setting, then try the next.  If things don't improve, revert to the original setting.

EJtechandDiy, 2019-10-19:
1. I really dont like it for Youtube 
2. Webcam quality sucks for YouTube.
3. I download it because I need to process the RAW pictures on my PC with LR timelapse and Light room
4. Feedrate is ad 94 I can try 90
5. I gonna try 195 and see what happend 
6. E3D v6 cant go higher then 3mm if  you do there is a big chance of failure

Im gonna Use the timelapse for my channel thats why im not using a webcam.

FormerLurker, 2019-10-19:
So, regarding the DSLR camera profile type, I'm probably going to remove those.  The error is being thrown because those profiles don't exist on the server.  I will add a better message in case this error happens in the future.

The problem with default DSLR profiles is that the scripts necessary to get the snapshots working won't exist, and gphoto2 will not be installed.  There is no good way for me to add the scripts, add permissions, and install gphoto2.  Gina doesn't allow plugins to run certain commands in setup.py (for very good reason).

Also, I see your response now, and will respond to that.

1.  That's OK, I just want you to do it as a test.  My theory is that your printer is just oozing so much per second, and this will give me a sense of what the oozing looks like without the travel delay.
2.  That's OK, but I want to see the difference in oozing from a DSLR delay (large) and a webcam delay (vey low).
3.  If you are processing RAW images, you don't need to use the default download script at all!  Just grab the photos from the camera's SD.  It will save a TON of time.

6.  I'm talking about retraction speed not length.  Keep it at 3mm for now.

FormerLurker, 2019-10-19:
Oh, I have something to add to item number 3.

If you want to download them for convenience, you could use the deferred download script I linked to above.  That guide shows how to get octolapse to download the snapshots after the print is complete.  It saves as much time as the no-download option.  Be sure to read the guide carefully because that is the most difficult one to get working, and it won't work on some cameras.

EJtechandDiy, 2019-10-19:
About number 3
The pictures are only saved on the camera SD card, they are not been downloaded.

Yea I have played with retraction is now ad 60mm

what do you mean with the default DSLR profiles?

FormerLurker, 2019-10-19:
Ok, thanks for the info about your camera script.  That makes sense.  I know you said it only took about a second to grab an image, so that makes sense.

About the DSLR profile, I was talking about the issue title, "Profile Update Error:INTERNAL SERVER ERROR".  You were trying to download some DSLR profile via the 'camera profile edit screen', and you got that internal error.  I just wanted you to know I figured out that issue, and it's because those profiles don't exist in the profile repository.  I'm going to remove them completely from that drop down box, because I don't think those profiles will ever be viable since a whole guide needs to be followed before they will work.  Does that make sense?

keawepoo, 2019-10-29:
@FormerLurker heres video of the server error... at least the one i got.
[https://drive.google.com/open?id=19iXPI5DSDC6HjrDHLvsrGGREPpKwtj3w](https://drive.google.com/open?id=19iXPI5DSDC6HjrDHLvsrGGREPpKwtj3w)

keawepoo, 2019-10-29:
> 3. If you are processing RAW images, you don't need to use the default download script at all!  Just grab the photos from the camera's SD.  It will save a TON of time.

Also just saying if its a Canon, It will not save the images to your CF/SD card... It'll save it to the pi. Its something to do with Canons firmware. Im a CPS, and a Photographer who use to use tethering alot.

EJtechandDiy, 2019-10-30:
> > 1. If you are processing RAW images, you don't need to use the default download script at all!  Just grab the photos from the camera's SD.  It will save a TON of time.
> 
> Also just saying if its a Canon, It will not save the images to your CF/SD card... It'll save it to the pi. Its something to do with Canons firmware.
> 
> Also might wanna look at useing Large JPG, RAW is at least 30mb per image... 30*1000=30GBs... (at least for me its around 30mb for a 1DX Full Frame RAW)

What version of octolapse are you using? And i have a nikon d5300. so you are telling me that if I remove the script it will take the pictures faster and safe time in that way? Becuase now the pictures are saved  on the SD card with the script.
What is going to be the defense without the script? 
Thanks in advance

FormerLurker, 2019-11-08:
@EJtechandDiy, if you want to try upgrading to the latest devel version, it's possible that some of your quality issues have been dealt with.  I'm not 100% sure, but I did find a few issues in and around the snapshot gcode section that have been fixed.  Here is a link:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Let me know if/when you install this.  As always, the devel branch is subject to change, but I've run several prints with it now, and it seems to be working pretty well (at least for me).

EJtechandDiy, 2019-11-08:
> @EJtechandDiy, if you want to try upgrading to the latest devel version, it's possible that some of your quality issues have been dealt with. I'm not 100% sure, but I did find a few issues in and [...]

I will do that after I finish upgrading the printer to Bigtreetech SKR 1.3  Bltouch and another couple of upgrades. Thanks for letting me now

FormerLurker, 2020-02-26:
This is a pretty old thread, and I think that due to all of the updates in octolapse since this issue was opened, that it's possible your issue has since been solved.  Please feel free to open another issue if you have any problems, and be sure to try the latest RC (or install from the devel branch).  Thanks again!

closed 2020-02-26 by FormerLurker