FormerLurker/Octolapse #392: DSLR issues

bloodywinggithub, 2019-10-17:
octoprint version : 1.3.11
octolapse version : Octolapse 0.4.0rc1.dev2
camera : canon eos 5d mark II

with the tutorial :  https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera

I can pass step 6, but can't pass the test print part.

show me this:
Camera Error
An unexpected error occurred while taking snapshots. See plugin_octolapse.log for details.

bloodywinggithub, 2019-10-17:
**### plugin_octolapse log:**

2019-10-17 02:26:51,629 - octolapse.timelapse - INFO - Print Start Detected.  Command: M110 N0, Tags:set(['trigger:comm.start_print', 'source:job', 'trigger:comm.reset_line_numbers'])
2019-10-17 02:26:51,638 - octolapse.__init__ - INFO - Print start detected, attempting to start timelapse.
2019-10-17 02:26:51,748 - octolapse.snapshot_plan - INFO - Running smart layer stabilization preprocessing.
2019-10-17 02:26:51,751 - octolapse.gcode_position - INFO - Parsing Position Args.
2019-10-17 02:26:51,751 - octolapse.gcode_parser - INFO - Rectangular bed set
2019-10-17 02:26:51,753 - octolapse.gcode_parser - INFO - No snapshot restrictions set.
2019-10-17 02:26:51,754 - octolapse.snapshot_plan - INFO - Parsing Stabilization Args.
2019-10-17 02:26:51,755 - octolapse.snapshot_plan - INFO - Parsing Stabilization Args.
2019-10-17 02:26:51,756 - octolapse.snapshot_plan - INFO - Processing File.
2019-10-17 02:26:51,757 - octolapse.gcode_parser - WARNING - No gcode command was found: ;FLAVOR:Marlin

2019-10-17 02:26:51,758 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TIME:1047

2019-10-17 02:26:51,758 - octolapse.gcode_parser - WARNING - No gcode command was found: ;Filament used: 1.14912m

2019-10-17 02:26:51,759 - octolapse.gcode_parser - WARNING - No gcode command was found: ;Layer height: 0.2

2019-10-17 02:26:51,759 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MINX:89

2019-10-17 02:26:51,760 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MINY:89

2019-10-17 02:26:51,761 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MINZ:0.2

2019-10-17 02:26:51,761 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MAXX:121

2019-10-17 02:26:51,762 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MAXY:121

2019-10-17 02:26:51,763 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MAXZ:20

2019-10-17 02:26:51,763 - octolapse.gcode_parser - WARNING - No gcode command was found: ;Generated with Cura_SteamEngine 4.2.1

2019-10-17 02:26:51,764 - octolapse.gcode_parser - WARNING - No gcode command was found: ;LAYER_COUNT:100

2019-10-17 02:26:51,765 - octolapse.gcode_parser - WARNING - No gcode command was found: ;LAYER:0

2019-10-17 02:26:51,766 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:SKIRT

2019-10-17 02:26:51,767 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MESH:F.stl

2019-10-17 02:26:51,769 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:WALL-INNER

2019-10-17 02:26:51,770 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:WALL-OUTER

2019-10-17 02:26:51,770 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:SKIN

2019-10-17 02:26:51,771 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MESH:NONMESH

2019-10-17 02:26:51,771 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TIME_ELAPSED:26.915345

2019-10-17 02:26:51,772 - octolapse.gcode_parser - WARNING - No gcode command was found: ;LAYER:1

2019-10-17 02:26:51,773 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:WALL-INNER

2019-10-17 02:26:51,773 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MESH:F.stl

2019-10-17 02:26:51,773 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:WALL-OUTER

2019-10-17 02:26:51,774 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:SKIN

2019-10-17 02:26:51,774 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MESH:NONMESH

2019-10-17 02:26:51,775 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TIME_ELAPSED:38.239617

2019-10-17 02:26:51,775 - octolapse.gcode_parser - WARNING - No gcode command was found: ;LAYER:2

2019-10-17 02:26:51,775 - octolapse.gcode_parser - WARNING - No gcode command was found: ;TYPE:WALL-INNER

2019-10-17 02:26:51,776 - octolapse.gcode_parser - WARNING - No gcode command was found: ;MESH:F.stl
[... 1845 lines cut ...]
2019-10-17 02:27:50,834 - octolapse.timelapse - DEBUG - Sent: G1 Y96.4 X96.4
2019-10-17 02:27:51,344 - octolapse.timelapse - DEBUG - Sent: G1 Y112.8 X96.4
2019-10-17 02:27:51,361 - octolapse.timelapse - DEBUG - Sent: G0 F2295 X96.4 Y113.6
2019-10-17 02:27:51,375 - octolapse.timelapse - DEBUG - Sent: M204 S3000
2019-10-17 02:27:51,385 - octolapse.timelapse - DEBUG - Sent: M205 X10 Y10
2019-10-17 02:27:51,394 - octolapse.timelapse - DEBUG - Sent: G0 F2550 X96.33 Y113.67
2019-10-17 02:27:51,439 - octolapse.timelapse - DEBUG - Sent: G0 X95.67 Y114.33
2019-10-17 02:27:51,752 - octolapse.__init__ - INFO - Print cancelling.
2019-10-17 02:27:51,843 - octolapse.__init__ - INFO - Print cancelled.
2019-10-17 02:27:51,844 - octolapse.timelapse - INFO - Snapshot jobs queue has completed, starting to render.
2019-10-17 02:27:51,901 - octolapse.timelapse - INFO - Started prerendering the timelapse. JobId: 79b239a8-7e58-4fa4-8dd3-facbbb1c96f4
2019-10-17 02:27:51,904 - octolapse.render - INFO - Found 0 images via a manual search.
2019-10-17 02:27:51,906 - octolapse.__init__ - INFO - Print failed.
2019-10-17 02:27:51,907 - octolapse.render - ERROR - Rendering Error
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/render.py", line 725, in _render
    self._pre_render()
[... cut ...]
RenderError: insufficient-images: No snapshots were available for the 'DSLR' camera profile.
2019-10-17 02:27:51,909 - octolapse.render - INFO - Sending render fail message
2019-10-17 02:27:51,910 - octolapse.timelapse - INFO - Completed rendering. JobId: 79b239a8-7e58-4fa4-8dd3-facbbb1c96f4
2019-10-17 02:27:51,913 - octolapse.render - INFO - Sending render end message
2019-10-17 02:27:51,914 - octolapse.render - INFO - Sending render end message
2019-10-17 02:27:51,914 - octolapse.timelapse - INFO - Completed all pending rendering jobs.

FormerLurker, 2019-10-17:
Can you post long log messages to gist.github.com please.  This is way too long to deal with in an issue thread.  Thanks!

bloodywinggithub, 2019-10-17:
> 
> 
> Can you post long log messages to gist.github.com please. This is way too long to deal with in an issue thread. Thanks!

sorry, for some reason I can't login gist.github.com now  :(

FormerLurker, 2019-10-17:
Ok, i will look at it when im at my PC tomorrow.  It takes like a min to scroll to the bottom 😂

bloodywinggithub, 2019-10-17:
> 
> 
> Ok, i will look at it when im at my PC tomorrow. It takes like a min to scroll to the bottom on mobile 😂

I'm tring to use vpn now

bloodywinggithub, 2019-10-17:
> 
> 
> Ok, i will look at it when im at my PC tomorrow. It takes like a min to scroll to the bottom on mobile 😂

I just uploaded log to gist.github.com, but I'm not sure if it correctlly, 'cause i'm a rookie  :P

FormerLurker, 2019-10-17:
Can you paste in the link to the gist please?

bloodywinggithub, 2019-10-17:
> 
> 
> Can you paste in the link to the gist please?

https://gist.github.com/bloodywinggithub/87ab6d8269a8c6c7028f8051501afa5d

FormerLurker, 2019-10-17:
That did it!  I will let you know when I figure something out, probably tomorrow.

bloodywinggithub, 2019-10-17:
> 
> 
> That did it! I will let you know when I figure something out, probably tomorrow.

THX

bloodywinggithub, 2019-10-17:
> 
> 
> That did it! I will let you know when I figure something out, probably tomorrow.

I found in Octolapse 0.4.0rc1.dev2, includ dslr option  in camere profile like profile import DSLR-download image/ gcode ... can I that, or do some seting?

bloodywinggithub, 2019-10-17:
> 
> 
> That did it! I will let you know when I figure something out, probably tomorrow.

when I do the step 5 : 
Finally, we can test our script by entering in the following command (it's long, so make sure you copy everything):
/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg

show me this :
/bin/sh: 0: Can't open Put the arguments sent by Octolapse into variables for easy use

bloodywinggithub, 2019-10-17:
> 
> 
> That did it! I will let you know when I figure something out, probably tomorrow.

it's woking now!
I  checked the take-snapshot.sh script, I don't know why when i copy the letter,the format has change,
when i fixed  it, now it's work

thanks again!!!!!!

FormerLurker, 2019-10-17:
I didn't really do much, but I'm glad you figured it out!  I did see from your log that the error wasn't being reported.  I'm going to look into that next.  I'll create a separate issue for that.

It looks like this issue is closed!

closed 2019-10-17 by FormerLurker