FormerLurker/Octolapse #394: Octolapse Rendering Failed

JustDerek1, 2019-10-20:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.10

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. The print was finishing and rendering

#### What should have happened?
The mpg should have rendered through FFMPEG

#### What happened instead?
I received the following error:

[image]

Octolapse Rendering Failed
rendering-exception: ffmpeg failed during rendering of movie. Please check plugin_octolapse.log for details.. Inner Exception: local variable 'p' referenced before assignment

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: HEVO
Printer Firmware Version: Latest Duet Wifi

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-10-20:
So, this is failing right here:

```
                try:
                    p = sarge.run(
                        command_str, stdout=sarge.Capture(), stderr=sarge.Capture())
                    os.rename(temp_filepath, self._output_filepath)
                except Exception as e:
                    raise RenderError('rendering-exception', "ffmpeg failed during rendering of movie. "
                                                             "Please check plugin_octolapse.log for details.",
```
However, the error message doesn't make any sense.  It says that p was accessed before assignment, but that's where it's being assigned.  I suspect something is going on with ffmpeg.   I recommend running a quick timelapse with Octoprint's built in timelapse and see if that works.

Another  thing you could do is use the 'Full Diagnostic - Test Mode' debug profile and run a short print.  In that mode your printer will not warm up, and no plastic will extrude.  After 3/4 snapshots are taken, cancel the print.  Rendering should fail.  After it has failed, download plugin_octolapse.log (Octoprint settings->logs) and upload it to gist.github.com and paste a link in here please.

Thanks!

JustDerek1, 2019-10-21:
Ok, this is weird.  I was able to render time lapses for the last several months.  I turned off Octolapse and turned on the regular time-lapse functionality in Octoprint.  It worked as of yesterday.  Now, I get this message:

[image]

I have no idea what is going on.  Order of events:

*  2 days ago, I was using the built in time-lapse functionality
*  Yesterday and this morning, I started playing around with Octolapse and I did get one to render - it was when I had canceled a print.  
*  I thought I had all of the setting correct.  So, I printed a calibration cube and got the original message that I posted above.

Also, I've rebooted a couple times - hoping that the simple act of rebooting would solve this.

JustDerek1, 2019-10-21:
Here is what it says in the Octoprint.log file:

> 2019-10-21 02:57:48,452 - octoprint.timelapse - ERROR - Could not render movie due to unknown error
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 842, in _render
>     p = sarge.run(command_str, stdout=sarge.Capture(), stderr=sarge.Capture())
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
>     return f(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
>     p.run(input=input, async_=async_)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
>     return f(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
>     self.run_node(node, input=input, async_=False)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
>     return f(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
>     result = getattr(self, method)(node, input, async_)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
>     return f(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
>     node.cmd.run(input=input, async_=async_)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
>     return f(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 662, in run
>     self.stdin = p.stdin
> UnboundLocalError: local variable 'p' referenced before assignment

FormerLurker, 2019-10-21:
Very strange.  Octolapse doesn't upgrade or change ffmpeg/avconv in any way, it just calls it.

Im not sure exactly how to go about debugging this, but I will think about it and will get back to you.

JustDerek1, 2019-10-21:
tldr:  Do a full system shutdown.  Update Octoprint.

So, another interesting point - I rebooted and then tried upgrading from 1.3.10 to 1.3.11 and it failed.  I disabled Octolapse and tried again (a long shot since this seems like an Octoprint thing).  It still didn't work.

Maybe Octoprint it corrupt in some way?

Then, I completely shutdown the pi.  I've been rebooting the system, but I thought that maybe shutting down the whole thing would help.  I did that and it then updated Octoprint.  It worked!  I'm not sure why.

I then did a couple layers on a calibration cube and the built-in time-lapse worked.  So, I've re-enabled Octolapse and am trying another full calibration cube print.

Honestly, it just seems like something was weird with the system.  I have no idea why fully turning off the pi would be any better than rebooting the pi.  Makes no sense.

Ok, I just completed a full cube and it rendered it beautifully.  Consider this closed.

closed 2019-10-21 by JustDerek1

FormerLurker, 2019-10-21:
Maybe.  Or your psu is dying maybe?  Any power warnings in octoprint?  Lots of mysterious issues are caused by power problems.

FormerLurker, 2019-10-21:
Um not sure if octoprint 1.3.10 has those warnings actually.  I will check tomorrow.

JustDerek1, 2019-10-21:
Maybe - It seems to be well powered.  I bought a power supply specifically for the pi with the appropriate amps.

reddai75, 2020-04-28:
stupid question: is there a way in this case to force octolapse to render again ?

FormerLurker, 2020-04-28:
In octolapse v0.4, yes

reddai75, 2020-04-29:
great! 
however as you mentioned in the release notes, most of the times rendering issues are related with memory available, at least with RPI 3 (my setup includes a reflex camera). I'm wondering how would it go with a RPI 4...

FormerLurker, 2020-04-29:
It would probably succeed more often when using h264/5, but i haven't tried it.  I need to get ne one one of those!