FormerLurker/Octolapse #395: Step 6 - Configure Octolapse Issue

homescienceideas, 2019-10-22:
Hey FormerLurker:) Thanks for this awesome plugin! 

 My Octoprint settings are not the same as your in the following video because I have the newest version. My version: (OctoPrint 1.3.12 running on OctoPi 0.15.1) https://youtu.be/J5SskYwP5JQ?t=10m47s

Also my Octolapse version is 0.4.0rc1.dev0. 

I am running into an issue at this step "Step 6 - Configure Octolapse

We're almost done! Open OctoPrint and open your Octolapse camera profile. Change the drop down box at the top of the page from 'Webcam' to 'External Camera - Script'

Next enter the following for 'Snapshot Acquire Script'"
I do not have "Snapshot Acquire Script" as an option to add this line. "/home/pi/scripts/take-snapshot.sh".

Any help would be great!

FormerLurker, 2019-10-22:
You are using the alpha version of octolapse, so things are different.  Open your camera settings, make sure the 'customize profile' checkbox is checked, and choose 'script camera' (i think that is what it is called..  might not be exact.  Am not in front of my pc.) Then you should see that setting.

homescienceideas, 2019-10-22:
Wow!! That was fast bro!! Thanks! If this helps I have a Canon Rebel T2i.

homescienceideas, 2019-10-22:
Should I downgrade to 3.4?

FormerLurker, 2019-10-22:
If it works for you, keep using it.  There have been a few updates though.  Keep an eye out for 0.4.rc3.dev0

homescienceideas, 2019-10-22:
I can’t find those settings anywhere on my camera. Please be patient with me. I just got the camera yesterday and I’ve never worked with cameras of this caliber😂

homescienceideas, 2019-10-23:
I don’t see anything that says “Camera Settings”. Are you talking about the camera profile? If so I went under that and I do not see anywhere that says “customize profile”, nor do I see a checkbox for it😞

FormerLurker, 2019-10-23:
So, reinstall (no need to uninstall first) Octolapse via the plugin manager.  From the plugin manager, click 'Get More' and use this url to install:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Let me know when you are finished, and don't use that URL again.  That is the development branch, and changes pretty frequently.  It might contain untested things.  I'm printing with this version right now, so it's probably fine.

The missing setting was due to a bug I fixed between 0.4.0rc1.dev0 and 0.4.0rc2.dev0.  You are not crazy, it was broken ;)  Anyway, I forgot about that bug. Once you upgrade the setting will be obvious!

homescienceideas, 2019-10-23:
Okay, I got it fixed but when I go to print I see this now "Object doesn't fit print volume

Object in My_Wii_Bar_Mount_1.gcode exceeds the print volume of the currently selected printer profile, be careful when printing this.
More
Object exceeds print volume in width.
Object exceeds print volume in depth.
Object's bounding box: (128.39, 170.08, 25.20) × (1579683.79, 990287.04, 280.00)
Print volume: (0.00, 0.00, 0.00) × (310.00, 315.00, 400.00)

You can disable this check via Settings > Features > "Enable model size detection "

homescienceideas, 2019-10-23:
And when I hit print anyways I get this in red: "Octolapse Startup Failed
Manual Slicer Settings Missing
Your printer profile is missing some required slicer settings. Either enter the settings in your printer profile, or switch to 'automatic' slicer settings. Missing Settings: Retract Before Move,Lift When Retracted"

homescienceideas, 2019-10-23:
It also will not print the test print. Basically the print head will not move.

FormerLurker, 2019-10-23:
The the first error 

> Object doesn't fit print volume

Is from OctoPrint, not Octolapse.  It looks like it thinks your part is super huge.  Not sure why that is, and I'm not an expert in that error. Just ignore that for now.

Regarding the second issue, there should be a help button for that error, no?  I just added that feature, so why not try it out and see if it helps?  That is, unless it didn't show up for some reason?  The error should look similar (not not exact, I triggered a different error) to this:

[image]

homescienceideas, 2019-10-23:
Got it all fixed and was able to start the test print. Got 2 errors. 

1. "Camera Error
An unexpected error occurred while taking snapshots. See plugin_octolapse.log for details."

2. "Stabilization Error" - No other info on this one.

FormerLurker, 2019-10-23:
Ok.  So you dont have 2 problems at once, I recommend using a webcam before getting your dslr working.

Next, can you post your gcode file at gist.github.com?  Also, export your octolapse settings and post those too.  That way I can get to the bottom of the stabilization error.  I should have some time tomorrow.

FormerLurker, 2019-10-23:
Oh, i forgot to ask, did that help button assist you?  You are probably the first person to see those, so Im interested in any feedback you may have.

homescienceideas, 2019-10-23:
Only 1 issue now. 

I already have a webcam and have used it. It did not look very good so I want to try my dslr. 

I think I did this right bro. https://gist.github.com/homescienceideas/f8fb308431debe4106530014f4ec8b1d#file-octolapse-settings

FormerLurker, 2019-10-23:
Yes, looks great!  Thanks, I will run it through the debugger tomorrow.

homescienceideas, 2019-10-23:
Ok bro! Thanks for all the help:)

homescienceideas, 2019-10-23:
It took 1 shot. I was thinking when doing the script, does the shots taken auto-increment? Lets say it takes a shot and names that 1st shot shot1.jpeg. Does it then on the 2nd shot name the photo shot2.jpeg?

homescienceideas, 2019-10-23:
Sorry, I was wrong. I just looked and my webcam was running and that is the 1 item.

FormerLurker, 2019-10-23:
You could try logging the camera and snapshot modules.  Look in the debug profile to set that up.  The log will be in the octoprint settings under Logs: plugin_octolapse.log.  post it to gist when you have it

homescienceideas, 2019-10-23:
Not sure what you mean "You could try logging the camera and snapshot modules.". Here is the log:) https://gist.github.com/homescienceideas/3f6c0489b89ca0e53e497f1a06b46559

homescienceideas, 2019-10-23:
> 
> 
> You could try logging the camera and snapshot modules. Look in the debug profile to set that up. The log will be in the octoprint settings under Logs: plugin_octolapse.log. post it to gist when you have it

I was so close today! Ha ha ha! See you tomorrow bro and thanks so much:)

FormerLurker, 2019-10-23:
Hey, I haven't looked at your gcode yet, but I did discover big rendering issue (I just made a lot of changes there).  Please reinstall via this URL before continuing:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

homescienceideas, 2019-10-23:
>  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Seems the link is not working...

homescienceideas, 2019-10-23:
That was odd. It's working now:)

homescienceideas, 2019-10-23:
OK bro, thank you for your work already but I’m running into another issue now. It printed the skirt then it started to print the part and it moved to the back corner of the machine and move the part forward and took a snapshot. Then it re-started on the next layer and printed the whole layer then right when it was going to move to the back it froze. I’m leaving it on to see if any errors pop up. So far no movement but no errors popping up on the screen.

homescienceideas, 2019-10-23:
"Stabilization Error" popped up and restarted the print then froze again.

homescienceideas, 2019-10-23:
It unfrozen again and now it is working perfectly. The only thing is I cannot find where it’s storing the photos. I’m assuming it’s storing them on the camera only and I’m trying to figure out how to show if they are on the my pi.

FormerLurker, 2019-10-23:
Very strange.  Any idea why it froze up?  I ran your code through the debugger and didn't have any issues.  I saw from your logs that there was a timeout in WAY less time than the configured timeout, so I suspect SOMETHING strange is going on there, but I haven't figured it out yet.  I'm adding some additional logging now, so I may ask you to upgrade and re-run a print with specific logging enabled if you can reproduce the 'freeze'.

Regarding your images.  The timelapse will automatically be rendered and then the timelapse images will have been deleted from the PI.  The script you used downloads the images to the pi, and never saves them to SD memory on your camera.  There are a few options for keeping the images on the SD or on the Pi:

1.  turn off **Clean Snapshots After Rendering** in the rendering profile.  The snapshot will then be located at /home/pi/.octoprint/data/octolapse/snapshots.  Each folder within snapshots is a single print job.  Each folder inside of the print job folder is an individual camera.  SFTP is a good way to download these files.  You will still get an automatically rendered timelapse using this method.
2.  Use one of the other DSLR tutorials for '[No Download](https://github.com/FormerLurker/Octolapse/wiki/DSLR-Image-Capture-With-No-Download)' or '[Deferred Download](https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download)'.  For the 'No Download' method, you will have to render your timelapse manually.  For the deferred download method (the most difficult to get working), Octolapse will render a timelapse AND you will have images on your SD.  Neither method will give you snapshot previews since the images won't be on the pi, and are inaccessible.  For the deferred download method, all images are deleted from the SD card at print start (I've not found a workaround for this).

homescienceideas, 2019-10-24:
I ran another test print and this is what I got on top of the "Stabilization Error". 
"Camera Error
An unexpected error occurred while taking snapshots. See plugin_octolapse.log for details."

I am going to try and print a normal print without Octolapse to see if I get the freezing. Here is the last log. https://gist.github.com/homescienceideas/369ff1d2f6bcf9afbf0a74fdf17287c4

homescienceideas, 2019-10-24:
I have no idea why it is freezing. I put this video together and I will be pointing people to you when I get this all done:) This is a private video so only people on here will see it. https://www.youtube.com/watch?v=l3bFZNZuFfw

homescienceideas, 2019-10-24:
Okay, I printed without Octolapse and it printed good with no freezing. Same print.

homescienceideas, 2019-10-24:
Hey my friend:) Getting anywhere? I don’t know why it’s freezing on me but it does only when I run a print through Octolapse. I’m eager to get this done but I know you’re probably busy. Any update my friend!

FormerLurker, 2019-10-24:
So, I watched your youtube video, and it looks like the delay is caused by camera download time.  What resolution are you using and what quality setting?  I'm asking because all of that data has to be downloaded over USB2 (most likely, but you could test the xfer time via the command line using gphoto2)  The stabilization error is 99% caused by the camera timeout setting.  It's possible the script is hanging on the first run for some reason.

The script you are using that takes a photo, saves it to the camera's internal RAM (NOT the SD card), then downloads it over usb, which we know is very slow.  For that reason, I recommend you try [this method](https://github.com/FormerLurker/Octolapse/wiki/DSLR-Image-Capture-With-No-Download) that I recommended to you in a previous comment.  Keep in mind that your images will ONLY be saved to your camera, no snapshot preview will be provided within OctoPrint, no rendering will be performed (disable rendering to prevent an error here), and that not all cameras work properly with this method.  However, if it works, you should get MUCH better performance.

Regarding the 'missing' photos, did you get a rendered timelapse?  Check the timelapse tab to see if one rendered.

homescienceideas, 2019-10-27:
> So, I watched your youtube video, and it looks like the delay is caused by camera download time. What resolution are you using and what quality setting? I'm asking because all of that data has to be [...]

Not sure if you are talking about on Octolapse or my Camera? (What resolution are you using and what quality setting?). I had the Octolapse working on my C920 before but now it does not work ether. Maybe I should switch to 3.4? How can I do that? 

I tried moving back to Octolapse 3.4 and now I have no camera's it says. No profiles:( What can I do now?

[image]

homescienceideas, 2019-10-27:
This is the 3.4 that I installed. https://github.com/FormerLurker/Octolapse/archive/v0.3.4.zip

homescienceideas, 2019-10-27:
I Restored All Defaults and this issue in the image above went away.

homescienceideas, 2019-10-27:
Got Octolapse working on my C920 in 3.4... I will follow your video again to set up my DSLR and see if it works on 3.4:)

homescienceideas, 2019-10-27:
Got it working:) I was still having an issue but then I turned off flash and auto-focus and the TEST print went great! I will test a real print tomorrow. I am up way to late!

FormerLurker, 2019-10-27:
Oh my, lots going on here!  Glad you got it working, but I'd like to figure out what was wrong when you were using v0.4.  

> I had the Octolapse working on my C920 before but now it does not work ether.

Can you detail what you meant by 'does not work'?  Did the print complete?  Were there errors?  Did it say it was rendering and you just couldn't find the rendered file?  

If it said it was rendering, I might know what was wrong.  I changed some naming templates for the rendered file and there was a problem with the default configuration.  You might want to look in /home/pi/.octoprint and see if you have some mp4 files there.  If there are any, I've fixed that issue in the devel branch.

Anyway, if you have time let me know about the above.  Thanks, and sorry for the delay in responding.  I've been out this weekend for the most part.

homescienceideas, 2019-11-03:
It would freeze as well while I was using my C920. Same issue as I was having with my DSLR on 4.0. The print would complete if I waited a long time and wait through it freezing and un-freezing. Yes, same errors as I was getting with DSLR. It would render but it was really choppy.

FormerLurker, 2019-11-04:
@homescienceideas, I'm working on an update that should fix some of these rendering issues.  It turns out you were not alone.  I'll post again here once it's ready.  Sorry for the delays!

FormerLurker, 2019-11-08:
So, if you upgrade to the latest development release with this url via the plugin manager, I hope it will solve at least some of the issues:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

I've simplified the routine that sends gcode to the printer, and I'm hopeful that will at the very least make this easier to debug.

To eliminate any settings migration related issues, please restore the default settings for Octolapse after installation.  Let me know how it goes!

FormerLurker, 2019-11-11:
OK, I've found some logging problems when calling external scripts that timeout, and I think I've fixed them.  I've been simulating a timeout error and see that the default take-snapshot.sh script will fail in some situations if an image was downloading during a timelapse.  The next time the script is called it will respond with a message asking if you want to overwrite an existing file, which will always trigger a timeout unfortunately.  This wasn't logged because of the previously mentioned bug.  I'm working on a solution, fyi, but now the logging works at least!

FormerLurker, 2020-02-26:
So MANY bugs have been fixed and many rendering improvements have been made since this issue was last updated.  I've been doing a bit of issue maintenance today, so it's a good time to close this down.  If you have any other problems feel free to create a new issue!

closed 2020-02-26 by FormerLurker