FormerLurker/Octolapse #396: Webcam stream not loaded - After pi update/reboot

keawepoo, 2019-10-25:
#### Version of Octolapse
Octolapse Version: Octolapse 0.4.0rc1.dev2

#### Version of OctoPrint
OctoPrint Version: OctoPrint 1.3.12 running on OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred
sudo apt-get update && sudo apt-get full-upgrade
sudo reboot

#### What should have happened?
work?

#### What happened instead?
Webcam stream not loaded

It might not be configured correctly or require authentication. You can change the URL of the stream under "Settings" > "Webcam & Timelapse" > "Webcam" > "Stream URL". If you don't have a webcam you can also just disable webcam support there.

Currently configured stream URL: /webcam/?action=stream

#### Operating System running OctoPrint and Octolapse
OctoPrint 1.3.12 running on OctoPi 0.16.0

#### Printer model & used firmware incl. version
na

#### Browser and version of browser, operating system running browser
na

#### Link to the gcode file you were printing when the problem occurred
na

#### Link to settings.json
Link to settings.json with all passwords removed: https://gist.github.com/keawepoo/3534b2fd4c6aae08a0c586a78c4ec435

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log: https://gist.github.com/keawepoo/e85da66181f7f5e6708f40d689716dc1

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/keawepoo/35addc8643bd9f557ed4d0e515ec7850

#### Link to contents of Javascript console in the browser
Link to javascript console output: na

FormerLurker, 2019-10-25:
So, this happened after you enabled the raspicam controls, right?  Looks like there is some issue with that module.  Edit octopi.txt and change your camera type from usb to auto, then reboot.  That should fix it.

keawepoo, 2019-10-25:
I did that, it still didnt work... also I changed the http webroot back to the original path. something.
Heck I even changed the files back to the full original, and there still not working... I also tried on FireFox (normally use Chrome). So maybe its a OctoPi issue?
`sudo nano /boot/octopi.txt
  GNU nano 2.7.4                                  File: /boot/octopi.txt

# mjpg_streamer release do not support the -f parameter (for specifying
# the capturing framerate) and will just refuse to output an image if it
# is supplied.
#
# The webcam daemon will detect those devices by their USB Vendor and Product
# ID and remove the -f parameter from the options provided to mjpg_streamer.
#
# By default, this is done for the following devices:
#   Logitech C170 (046d:082b)
#   GEMBIRD (1908:2310)
#   Genius F100 (0458:708c)
#   Cubeternet GL-UPC822 UVC WebCam (1e4e:0102)
#
# Using the following option it is possible to add additional devices. If
# your webcam happens to show above symptoms, try determining your cam's
# vendor and product id via lsusb, activating the line below by removing # and
# adding it, e.g. for two broken cameras "aabb:ccdd" and "aabb:eeff"
#
#   additional_brokenfps_usb_devices=("aabb:ccdd" "aabb:eeff")
#
# If this fixes your problem, please report it back so we can include the device
# out of the box: https://github.com/guysoft/OctoPi/issues
#
#additional_brokenfps_usb_devices=()

### Additional options to supply to MJPG Streamer for the RasPi Cam
#
# See https://faq.octoprint.org/mjpg-streamer-config for available options
#
# Defaults to 10fps
#
#camera_raspi_options="-fps 10"

### Configuration of camera HTTP output
#
# Usually you should NOT need to change this at all! Only touch if you
# know what you are doing and what the parameters mean.
#
# Below settings are used in the mjpg-streamer call like this:
#
#   -o "output_http.so -w $camera_http_webroot $camera_http_options"
#
# Current working directory is the mjpg-streamer base directory.
#
camera_http_webroot="./www-octopi"
camera_http_options=""
`

FormerLurker, 2019-10-27:
Maybe remove the driver you added to the modules file and reboot?  

Also, your octopi.txt looks different than what I'd expect.  These two parameters look off:

```
camera_http_webroot="./www-octopi"
camera_http_options=""
```

try this instead:
```
#camera_http_webroot="./www-octopi"
#camera_http_options="-n"
```

The lines are commented out there, but that should still reset them to the defaults.

I'm going to try a clean install of OctoPrint and will try to see for myself what's going on here.  I first need to finish up some code changes.  Sorry about the delay in advance.

keawepoo, 2019-10-28:
Modules as in?

```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

i2c-dev
#bcm2835-v4l2
```

I already did that, and its still the same...

keawepoo, 2019-10-29:
@FormerLurker  So I did some googleing, and found this: https://raspberrypi.stackexchange.com/questions/85504/picamera-with-rpi-v3-b-gpioman-get-pin-num-pin-camera-led-not-defined?noredirect=1&lq=1

I did this
`vcgencmd get_camera`
And got this back
`supported=1 detected=1`

/etc/modprobe.d/raspi-blacklist.conf - was empty 
/etc/modules - Is listed above

free -l
https://gist.github.com/keawepoo/085b495e62e56b9811055f8ee9d08e89

df
https://gist.github.com/keawepoo/70b9e3fd419204c281b3e646c0d3ce96

sudo vcdbg log msg
https://gist.github.com/keawepoo/2688619862f7145e953d5444f7ab3c1f

sudo raspistill -v -o image.jpg
https://gist.github.com/keawepoo/29537899788682d4390150310ff8ba3c

Also increased the GPU memory from 128 to 256

keawepoo, 2019-10-30:
Fixed... Turned out the cable thats used for the touchscreen somehow conflicted with the codes.

closed 2019-10-30 by keawepoo