FormerLurker/Octolapse #398: Lost steps during stabilisation move

ochm, 2019-10-27:
Octolapse Version: 0.3.4
OctoPrint Version: 1.3.12
Diagnostic Logging was Enabled: NO
OS Name: OctoPi
Os Version: 1.3.12
Printer Model: Prusa i3 MK3
Printer Firmware Version: 3.8.2

Sometimes, not very often, but only wen the Octolapse is turned on,
the printer losts some steps on X or Y or both, and the print is damaged.
I tried to park extruder on different positions, but the lost steps ocours again.
Without Octolapse i haven't such problem.

I'm using the Gcode snapshot detection mechanism, 
the snap command is in the after layer change gcode.

IMHO the problem is in the too fast movement of extruder/bed to/from park position.
I'm trying to find such setting in Octolapse but without success. 
I don't think that is the same speed as travel_speed defined in slicer, 
because without Octolapse no lost steps in any prints, for example with more printed objects, where the travels ocours too (between objects).

So the question is, where is the travel speed for parking extruder in stabilisation defined, or how can be changed?

Thank you for your time.

FormerLurker, 2019-10-27:
Open your printer profile, select your slicer type from the drop down bo, then copy all of your settings from your slicer into Octolapse.  The travel speed is defined there.  If you still experience skipping, turn down your travel speeds in both octolapse and your slicer.  Let me know if that helps.

ochm, 2019-10-27:
Ah see, the Movement speed is used for movement and in feature detection mode. The travel in slicer and Movement speed in Octolapse is set to 180, and as i said, without Octolapse turned on the prints are OK, so i'll try to lower the number in Octolapse and do some prints. OK?
Is really it necessary to copy all settings from slicer to Printer profile? Because i'm changing often the settings in slicer, and to do the same in Octolapse is annoying, i thought that in the gcode detection mode doesn' mater about the settings in Octolapse printer profile, or not?

ochm, 2019-10-27:
Additionaly, the PrusaSlicer isn't in the list, there is only previous Slic3r Prusa edition. Can be used Slic3r Prusa edition option for PrusaSlicer https://github.com/prusa3d/PrusaSlicer/releases ?

FormerLurker, 2019-10-28:
PrusaSlicer should work fine.  Also, about entering slicer settings, the next version (still unreleased) extracts the settings automatically from the gcode file.  You shouldn't have to worry about this too much longer!

ochm, 2019-10-28:
I set all Octolapse printer profile speeds to the same values like in PrusaSlicer, and just for sure slower in both lower the travel speed from 180 to 160. Now I'm finishing second print and still without any issue. So I´m not sure if the lower speed or the same speed setting helped...

FormerLurker, 2019-10-28:
So, one more thing you could check.  Look in the main octolapse settings (OctoPrint settings->Octolapse->Main Tab->Edit Main Settings).  Then make sure **Display real snapshot time** is UNCHECKED.  That can occasionally cause skips on my MK2 (not sure about MK3).  Fyi, I have removed that setting entirely from the next release because of that issue.

ochm, 2019-10-29:
Yes, the option was turned on, so i turned it off, and will try some prints with higher travel speed.
Must be the travel speed value the same in slicer and in Octolapse? And if yes, can you tell me why?

FormerLurker, 2019-10-29:
No, it doesn't need to be unless you are using the **Feature Detection** option in the trigger profile.  In that case an incorrect speed may result in missed snapshots.

FYI, **Feature Detection** has been removed in the next version and replaced with a more reliable and easier to configure version that uses gcode comments to determine what feature is printing.

I have been thinking of allowing overrides of certain parameters for octolapse snapshot moves (hop, lift, retract, travel, etc...), but this will add quite a lot of complexity unless it is done right.  I'm not sure if/when that feature will be available.

ochm, 2019-10-30:
Perfect, everything is clear, IMHO the case can be closed. 

Sincerely 
​​-- 
Martin

FormerLurker, 2019-11-01:
Excellent!  I hope your timelapses are going well!

Case CLOSED!!

closed 2019-11-01 by FormerLurker