FormerLurker/Octolapse #399: Octolapse - Camera Error THEN Octolapse Rendering Failed

gdotp01, 2019-11-01:
Octolapse Version: Octolapse 0.3.4

OctoPrint Version: 1.3.12 running on OctoPi 0.16.0

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Simple Calibration prints
2. Initially I get the following error a few times 
Octolapse - Camera Error
snapshot-thumbnail-create-error: Create Thumbnail - An unexpected exception occurred while creating a snapshot thumbnail. Check the log file (plugin_octolapse.log) for details. - Inner Exception: cannot identify image file '/home/pi/.octoprint/data/octolapse/snapshots/latest_354def78-9eea-409a-ad23-ee966dfff4ba.jpeg'
3. The on completion of the print I get the following 
Octolapse Rendering Failed
render-error: Unknown render error. Please check plugin_octolapse.log for more details.. Inner Exception: [Errno 2] No such file or directory: u'/home/pi/.octoprint/data/octolapse/snapshots/01a107bc-fc86-44e9-945e-42a750e2d246/354def78-9eea-409a-ad23-ee966dfff4ba/20mm-hollow-box_0.15mm_PLA_MK3S_50m000158.jpg'
4. It looks like it collected the images as they show as an animation inn the Octolapse window 
5. If I SSH onto the pi it looks like the directory exists and it has jpg files

I've had  it working before
If I set Octolapse up in Test Mode it captures the entire test and does render it 

#### What should have happened?
Rendered the final file 

#### What happened instead?
The above errors and no render file 

#### Operating System running OctoPrint and Octolapse
OctoPi 0.16.0 running on a Pi Linux octopi 4.14.79-v7+ #1159

#### Printer model & used firmware incl. version
Printer Model: Prusa Mk3+ 
Printer Firmware Version: Latest 

#### Browser and version of browser, operating system running browser
N/A

#### Link to the gcode file you were printing when the problem occurred
https://www.dropbox.com/sh/i2nbpaxx2hcwq3q/AABPRfg1xu8O5AFgocTnNl3Ja?dl=0.

#### Link to settings.json
As above 

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  Not sure how to find this 

#### Link to octoprint.log
Link to octoprint.log:  Ditto

#### Link to contents of Javascript console in the browser
N?A

#### Screenshots and/or videos of the problem:
N/A

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-11-01:
So, can you do the following:

1.  Change your 'Debug' profile to 'full diagnostic - test mode'.
2.  Run a print using the gcode file you submitted.  Note that your printer will not warm up, and no plastic will be extruded for your convenience.
3.  Once the print is finished, download the log file from OctoPrint Settings (wrench/spanner icon)->Logs and download plugin_octolapse.log.
4.  Post the log file to gist.github.com and put a link to the gist in a comment.
5.  Download your octolapse settings by navigating to OctoPrint Settings (wrench/spanner icon)->Octolapse, then click 'download settings'.  Post that to gist.github.com just like the log file.

Fyi, I usually only see this issue if people are using a DSLR, but have their cameras set to RAW.  I'm really confused as to why the snapshot preview is working, unless you are using multiple cameras (DSLR + webcam for example).

Also, I will be pretty busy until Monday after 5:00PM CT, but will try to look at this ASAP, especially if you can post the required info very soon.

Thanks for posting!

gdotp01, 2019-11-01:
Thanks for getting back so promptly. I’ll run a full diagnostic test mode tomorrow and upload it.
I’m only running. A Logitech 920 web cam no others. The octolapse settings are posted already in the Dropbox link 

It seems to work when in test mode however so I might also do another small print. I’ve tried two different gcodes and both didn’t render.

Graham Powell
Grahampowell@email.com

gdotp01, 2019-11-02:
hi, I've run the print in Full diagnostic test mode and it did experience the same error messages. I've pasted a screen shot of these into the Dropbox. However there are no logs in the Octoprint area that I can see which looks like its /home/pi/.octoprint/logs and it tests as valid. I'd SSH'd onto the Pi and the log files area is empty.

gdotp01, 2019-11-02:
Ok tried again and it failed the test print but i did seem to get the two logs. Here they are 

https://gist.github.com/gdotp01/27635a25a33b4f013e751d87954674dd

gdotp01, 2019-11-02:
Settings JSON is here:

https://gist.github.com/gdotp01/55b843b0018a170c194f0e917e306ad5

FormerLurker, 2019-11-04:
Hey, I only see octoprint.log, not plugin_octolapse.log in your gist.  Maybe there was a problem uploading it (or maybe I'm crazy and just can't find it, lol!) I do see some very strange errors in octoprint.log, and I"m not sure where those are coming from.  Can you try to re-upload plugin_octolapse.log?  That would be very helpful.

Also, FYI, the next version of Octolapse (still in alpha) is WAY different than the current version, and whatever issue you have been experiencing might be fixed in the new version already.  If we can't get this working for you I might ask you to try the new version out if you don't mind.  First I'd like to look at your plugin_octolapse.log file.

gdotp01, 2019-11-05:
You are correct looks like i missed the plugin_octolapse.log file. I’ll upload it when I get home on Thursday. 

G.P

gdotp01, 2019-11-06:
Ok I think i've uploaded the Octoprint_plugin.log here 
https://gist.github.com/gdotp01/3a019c71a073c4dbc0b9246a557fa170

FormerLurker, 2019-11-06:
The log was very helpful, thanks!  Can you check the following directory on your PI and see if there are any images:

/home/pi/.octoprint/data/octolapse/snapshots/10d28216-6f76-4f08-8dec-c1bf01a7bf12/354def78-9eea-409a-ad23-ee966dfff4ba/

If there are can you zip them and post somewhere I can get to them?  I'm suspecting some kind of corruption since pillow couldn't generate thumbnails for some of the images either.  It seems to be irregular (3 of 161 images approx), so it might be difficult to get to the bottom of this 100%.

However, once I see the file numbering I should be able to tell if the next version of Octolapse (v0.4) is more tolerant.  I believe it should process in many cases were v0.3.4 fails.  I might even be able to check all of the snapshots ahead of time and remove any that are corrupt.

gdotp01, 2019-11-07:
Hi. In Dropbox here (Files.gtz), There are not however 161 images in the directory there are only 32 

https://www.dropbox.com/sh/i2nbpaxx2hcwq3q/AABPRfg1xu8O5AFgocTnNl3Ja?dl=0

FormerLurker, 2019-11-08:
OK, so I have gotten another report regarding this issue (see #401).  I've been trying to reproduce this on my end, and so far no luck.  However, the problem definitely has something to do with Pillow.  Can you try the following (also mentioned in the other issue) so that we can see what version of Pillow you are running?  Connect to your pi and run the following commands:
```
source oprint/bin/activate
python -c 'import PIL; print PIL.PILLOW_VERSION'
```

That should print out the version.  Once you have that let me know what it is.  Next, if you are not using pillow v6.2.1, you can upgrade with the following command:

```
pip install --upgrade pillow
```

Then try a print again and let me know if that helps.  I'm running some additional tests as we speak, but can't seem to reproduce the problem yet.

gdotp01, 2019-11-08:
The version reported is 6.1.0 so i've upgraded and will run the same print again now

gdotp01, 2019-11-08:
Ok that worked fin, no errors and it rendered fine. Looks like it was an issue with that version. I'll run a longer test tomorrow to make sure.

FormerLurker, 2019-11-08:
Excellent!  So, I believe I've figured out why it was failing in the previous version and have added some code that will hopefully fix it in V0.4.rc1.dev3.  I will leave this issue open until I hear back from you.  Thanks!

gdotp01, 2019-11-14:
Hi. I think this can be closed. I've done a couple of other prints and they have worked fine. Thanks for looking into it.

FormerLurker, 2019-11-14:
OH great!  Thanks for letting me know.  Let me know if you run into any more issues!

CLOSED!!!

gdotp01, 2019-11-15:
Is there a beta I can download, last night a print did error. Just one image looks like it didn't render but the video failed as a result.

FormerLurker, 2019-11-15:
@gdotp01, Yes there is.  The pillow error you received initially has been fixed by using context managers.  It is also possible that your webcam timed out during a file write, which is a new situation that I just debugged.  That too has been fixed, more logging has been added, additional bugs were found and fixed.  In general, all things related to related to snapshots and rendering have been generally improved.

Please install via the plugin manager from this link:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

DEFINITELY [read this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse), since a LOT has changed.  

Since you will be upgrading to a beta (alpha really), would you do me a favor to help me test the upgrade path?  Please BEFORE you upgrade, download a copy of your settings.json file (Octoprint Settings->Octolapse->Download Settings).  Then, after you upgrade, download your now updated settings.json file again (the button name will change from download settings to export settings), and post both versions here?  This will help me verify that the settings migration is working.

Next I recommend you restore the default settings in case there are still settings migration issues.  The new version is MUCH easier to setup (IMHO), so it shouldn't take too long to get things up and running.  Your printer is supported, so you should just have to pick the make and model.  Please try out the automatic slicer settings detection also, since that is a critical new feature that will benefit from additional testing.

Ask me if you have any questions or run into problems!  I'll be here all day, but will be traveling for the next few weeks, so I will be much slower to respond.  Thanks for helping me test, and good luck with the new version!

gdotp01, 2019-11-17:
I tried the old version once more and got the same error. So I’ve since installed the new , reset the settings and then reloaded my old jsonsettings and ran a longer print. It all worked well, the software took all the pictures and rendered the final video successfully. I have to say the new camera settings interface is nice , looks good. I’m using slicr so I set up the verbose mode.  I’ll keep you updated if I run into any issues. Thanks again

FormerLurker, 2019-12-16:
No problem!  I half/closed this right after you responded, but must have not hit the button.  Whoops :)

Thanks for your report!

closed 2019-12-16 by FormerLurker