FormerLurker/Octolapse #400: PkTriggerCord Support for Pentax Cameras issue

ligantx, 2019-11-04:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version:  1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1) I try to implement the Configuration for External Cameras with PKTRIGGERCORD instead of gphoto2, in order to support Pentax cameras. Until now i have done a good start, and i have managed to edit the take-snapshot.sh script in order to take photos in the correct folder. 

2) When i run the command `/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"
` there is a photo saved in `"/home/pi/scripts/test.jpg"` but unfortunately as test-0000.jpg and not as test.jpg (in windows the `pktriggercord-cli.exe --model=K-30 --file_format=JPEG -F SNAPSHOT_NUMBER `--output_file=test.jpg` will output the file as test.jpg)

3) everytime i run `/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"
` the photo will be replaced with the new one as /home/pi/scripts" "" "/home/pi/scripts/test-0000.jpg

4) the octolapse cannot render the final timelapse in test mode

#### What should have happened?
everything should work normally and i guess octolapse should create a new file with different filename after every picture

#### What happened instead?
i get the following errors in octolapse test mode:
"The snapshot was not found in the expected directory: '/home/pi/.octoprint/data/octolapse/snapshots/53f10f10-8fe7-42ef-bfb0-d77894d6e805/87849ef5-e7c1-4d67-98d4-e5a1373e5064/balloonbasket_fixed_0.2mm_PLA_MK3S_2m000000.jpg'."
and
"Octolapse Rendering Failed
insufficient-images: Only 1 frame was captured, cannot make a timelapse with a single frame"

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: MK3S
Printer Firmware Version: N/A

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

FormerLurker, 2019-11-08:
Can you post your modified script?  It sounds like you just need move a few parameters around.  The take-snapshot script is supplied with the following parameters (in order):

* Snapshot Number - an integer value that increments after each successful snapshot
* Delay Seconds - the camera delay in seconds
* Data Directory - the path to the Octolapse data folder
* Snapshot Directory - The path to the current camera's snapshot folder for the current print job.  This directory may not exist, so be sure to create it before using the path!
* Snapshot File Name - The expected file name of the snapshot after it has been taken
* Snapshot Full Path - the full path and filename of the expected snapshot.  If this file exists after the script has returned, Octolapse will treat it like any other snapshot, and will apply transposition, record metadata (for rendering overlays), and will generate a thumbnail that will be sent to the client.

So, the script can save the image with any file name you like as long as you copy or rename it via the ```Snapshot Full Path``` parameter.  The images MUST end with the following numbering in order for FFMPEG/AVCONV to be able to render the video:  000000, 000001, 000002, etc...

Anyway, if you post the script I will take a look.  Thanks for posting!

ligantx, 2019-11-21:
sorry for being so late...
so i recreated the issue..and here i have the two files [https://github.com/ligantx/pentax_octolapse](url)

i have added `cd ${SNAPSHOT_DIRECTORY}
rename "s/-0000.jpg/.jpg/" *-0000.jpg` so i can rename the default suffix pktriggercord-cli puts to every photo..

if i run:
`/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"
`
there will be a test.jpg under /home/pi/scripts/test.jpg folder..which i guess is what we want..

although it looks to work fine..when i run the octolapse with full test mode, it cannot work properly. (i included my script and the octolapse log shorted.

FormerLurker, 2019-12-16:
OMG, sorry for the delay.  Been working on so many things, but not much on an excuse.

Would you be willing to try this out on the development version?  I have added a ton of helpful logging to the snapshot scripts, and have already fixed many issues.  I just made some pretty big updates, and would like to run some DSLR tests, which I plan to do tomorrow.  I will take a closer look at your scrip too.  

If you're interested, let me know and I'll message you after I'm comfortable that the changes I made in devel are good

FormerLurker, 2019-12-18:
So, I put a TON of work into the camera scripts, and I think it's ready for testing.  I have added test buttons for ALL of the camera scripts, so it should be MUCH easier to test your scripts out within the interface.  I don't yet report the actual error messages via the UI (which may be resolved in a future update), but you should be able to ssh into your pi, navigate to /home/pi/.octoprint/logs and run ```tail -f plugin_octolapse.log``` to watch the console and error output (and a bunch of other things), which should make it MUCH easier to debug.

I also added a button in the debug profile you can use to clear the log file out, which might be useful if you want to capture a clean log to create an issue.  You can download the log file from the debug profile page now too.

Whew, that was a lot to change to debug this issue, but DSLR problems have been terrible due to improper logging and testing capabilities.  Hopefully this has been solved.  

I also took a closer look at your .sh file, and now I finally understand the issue you reported initially.  It looks like the following code is trying to strip off extra numbers from the snapshot file name:
```
#for f in *-0000.jpg; do mv "$f" "${f%-0000.jpg}"; done
#for f in *; do mv "$f" "$f.jpg"; done
#for file in *.jpg ; do mv $file ${file//-0000.jpg/.jpg} ; done
cd ${SNAPSHOT_DIRECTORY}
rename "s/-0000.jpg/.jpg/" *-0000.jpg
```
I don't think it's working as expected.  I think I would try to do the following:

1.  Download the image using a fixed name, so you know exactly what it will be.  Something like:
```sudo pktriggercord-cli --model=K-30 --file_format=JPEG --output_file="${SNAPSHOT_DIRECTORY}/tmp.jpg"```

2.  Then just move the file to the proper location.  If I'm correct, the image should have the following file name:  ```tmp-0000.jpg```.  If so, it should be simple to move.  Maybe something like this:
```
mv "${SNAPSHOT_DIRECTORY}/tmp-0000.jpg" "${SNAPSHOT_FULL_PATH}"
```

However, this error from your original post really confuses me:  
>Octolapse Rendering Failed insufficient-images: Only 1 frame was captured, cannot make a timelapse with a single frame"

How could 1 image be captured if the script isn't working right???  Anyway, I will try to test a modified script that could simulate your situation later today if I can get out of work early enough.  

Let me know if any of the above helps at all, or if I'm just misunderstanding you.  I really want to help.

Thanks!

ligantx, 2020-01-02:
sorry for being late...i wanted to put some time in this.. and thank you for putting real time into this..

So, i made the changes you suggested (steps 1 and 2)..
To make it clear: i understand that you suggest to always take a photo with the name `tmp.jpg`

That pktriggercord ALWAYS name as `tmp-0000.jpg` (and its true, because 
`sudo pktriggercord-cli --model=K-30 --file_format=JPEG --output_file="tmp.jpg"`
creates tmp-0000.jpg file.

And then mv (a.k.a. rename) tmp-0000.jpg to test.jpg with commands like:
`/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/**test.jpg**"`

So, my guess is that octolapse, change the name by itself and trigger the script with different:
`SNAPSHOT_FILENAME` and `SNAPSHOT_FULL_PATH`..and in the end i'll have a collection of:
`test1.jpg`, `test2.jpg`, `test3.jpg`, ...., `test%last_layer_number%.jpg` that will be used for rendering..
am i correct?

---------------------------
I just made a Full diagnostic - Test mode print, and it looks to work flawlessly..
I think the problem was that the naming from the pktriggercord wasnt consistent and had conflicted with the script..

I'll make a long real print and return with results..
If this works ill make a complete Pentax guide, for other poor guys that owns these cameras..

Btw a small question: can i use Layer change - high quality for my pentax? (i set it up and works fine for my Logitech web cam)..i guess its camera irrelevant

ligantx, 2020-01-09:
?

FormerLurker, 2020-01-09:
Sorry about my lack of a response, and thanks for the nudge.  Got sick over the holidays, and every IT project is always started on jan1 for some reason, so it has been hectic.

Regarding the error with the thumbnail, is your camera taking raw images?  If so, change to jpg only.

Regarding the script, yes it should rename and move the images.  The names will have five digits so that you have image00001.jpg, image00002.jpg, etc.

Yes you can use high quality mode.  There are some issues with it (a replacement feature exists but n the next version that is way better and easier to use).

Thanks for your offer to build a guide!  I will add it to the repo.

Did I miss anything?  On mobile, sorry.

FormerLurker, 2020-03-26:
Since it's been a couple months I'm going to go ahead and close this issue.  If you want me to reopen it, please let me know and I'll be happy to do so.  FYI, I have continued to enhance the script cameras, and have released Octolapse V0.4.0rc1, so you may want to update.

Thank you very much for your contribution!

closed 2020-03-26 by FormerLurker

ligantx, 2020-03-26:
Sorry, i had in mind that i want to see this thing again, but i put it low on my priorities..
ill try the new release, and ill keep you updated.
Thanks!