FormerLurker/Octolapse #401: Snapshot and Render Error

codemonkey812, 2019-11-08:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Just printing a test item, and get errors when the print starts (snapshot-thumbnail-create-error), and at the end (render-error: Unknown render error). Both error reference a directory error

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Klipper Print

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  <script src="https://gist.github.com/codemonkey812/c6e3c01eeefce93f95ba0d22015c88b7.js"></script>

#### Link to octoprint.log

Link to octoprint.log:  <script src="https://gist.github.com/codemonkey812/9d3cb7f794d4842873c036bd24a25ece.js"></script>

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-11-08:
So this is the second time I've heard of this recently.  I suspect something related to some recent update.  I am investigating another similar issue, and may combine this issue with another.  Please stand by for that.  I plan to spend the weekend tracking this down.

FormerLurker, 2019-11-08:
So, I checked my pillow version, and it was quite old.  I updated it to 6.2.1 (the latest version) and haven't had any similar issues yet.  My debugger is running on windows, so I will run a test on the raspberry pi next.

Can you see what version of pillow you are running?  You can do this on the PI by running the following commands:

```
source oprint/bin/activate
python -c 'import PIL; print PIL.PILLOW_VERSION'
```

That should print out the version.  Once you have that let me know what it is.  Next, if you are not using pillow v6.2.1, you can upgrade with the following command:

```
pip install --upgrade pillow
```

Then try a print again and let me know if that helps.

IIPoliII, 2019-11-08:
I just openend the same issues sorrym will check your fix :

I am using 6.2.0

So pillow is surely the issue

codemonkey812, 2019-11-09:
I was running 6.1.0 of pillow. I installed 6.2.0, but am still getting the error.
2019-11-09 00:22:39,829 Snapshot Delay - Waiting 0.125 second(s) after executing the snapshot script.
2019-11-09 00:22:39,957 Snapshot - downloading from http://127.0.0.1:8080/?action=snapshot to /home/pi/.octoprint/data/octolapse/snapshots/8c45fea4-1c5f-4acd-af5e-abc51f29d0d9/354def78-9eea-409a-ad23-ee966dfff4ba.
2019-11-09 00:22:40,360 Snapshot - Snapshot saved to disk at /home/pi/.octoprint/data/octolapse/snapshots/8c45fea4-1c5f-4acd-af5e-abc51f29d0d9/354def78-9eea-409a-ad23-ee966dfff4ba/clip175-v1000018.jpg
2019-11-09 00:22:40,370 Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/snapshot.py", line 535, in run
    self.create_thumbnail()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/snapshot.py", line 330, in create_thumbnail
    cause=e
SnapshotError: snapshot-thumbnail-create-error: Create Thumbnail - An unexpected exception occurred while creating a snapshot thumbnail.  Check the log file (plugin_octolapse.log) for details. - Inner Exception: cannot identify image file '/home/pi/.octoprint/data/octolapse/snapshots/latest_354def78-9eea-409a-ad23-ee966dfff4ba.jpeg'

2019-11-09 00:22:40,378 Snapshot Download Job completed, signaling task queue.

2019-11-09 00:24:07,855 render-error: Unknown render error. Please check plugin_octolapse.log for more details..  Inner Exception: [Errno 2] No such file or directory: u'/home/pi/.octoprint/data/octolapse/snapshots/8c45fea4-1c5f-4acd-af5e-abc51f29d0d9/354def78-9eea-409a-ad23-ee966dfff4ba/clip175-v1000021.jpg'
2019-11-09 00:24:07,856 Completed rendering. JobId: 8c45fea4-1c5f-4acd-af5e-abc51f29d0d9

IIPoliII, 2019-11-09:
Sadly didn't fixed it for me :C

FormerLurker, 2019-11-09:
OK, so I'm still not 100% sure what is causing this error, but I suspect it is because version 6.1 removed the ability to implicitly close the image’s underlying file.  I have changed these bits of my code to use a context manager, so if that is the problem, it is fixed in the devel branch.  Now the devel branch is WAY different than the current version.  I suppose since neither of you are able to get v0.3.4 working, that it is reasonable to try the devel version.

If you want to test the new version to see if this has been fixed, please try the following steps:

1.  Follow [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) regarding installing the latest version.  Be sure to use the devel (not the rc/devel) link to install from the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/devel.zip
2. I strongly recommend restoring the default settings immediately after installation/reboot by going to the octolapse tab, clicking the new gear icon in the upper right hand corner of the tab (also available via OctoPrint Settings->Octolapse), and clicking **Restore Defaults**.  At that point you will have to add a printer profile and configure your camera, but that is much easier in the current version.  Most likely your printer will be available in the make/model drop down, and everything will be configured automatically.

Give that a go, and let me know if you have any problems.  Obviously the version you are installing is an ALPHA version, and the odds are good that you will encounter some issues.  I strongly recommend starting off with a test mode print (set the debug profile = **Test Mode - Debug Issues**), which will prevent your printer from warming up or extruding plastic.

codemonkey812, 2019-11-09:
The beta version works (I also like the improvements). Not related to my original issue, but the link on the Automatic Slicer Setting page for Cura 4.2 or greater does not work( https://raw.githubusercontent.com/FormerLurker/Octolapse/devel/octoprint_octolapse/data/scripts/gcode/cura/settings_output_v4.2.gcode ).

Thank you.

FormerLurker, 2019-11-09:
@codemonkey812 - I fixed the link.  I changed the path and forgot to update it.  Try again!

Note that if you have multiple extruders, none of those scripts are correct.  I need to add instructions for those too.

FormerLurker, 2019-11-11:
@IIPoliII, were you able to try the new version from the devel link I sent earlier (https://github.com/FormerLurker/Octolapse/archive/devel.zip).  Check my previous post and see if that helps.  If it's more than a day or two from now maybe ask me if the link is still safe :)  It should be since I'm mostly fixing bugs and adding logging to help diagnose some issues.

IIPoliII, 2019-11-12:
Thanks a lot will check that out once i am able to remove the clogs of my printer ^^' will definitly try

FormerLurker, 2019-11-30:
@IIPoliII, I'm going to mark the thread 'waiting for response'.  If I don't hear from you after a while, I'll close the issue down since @codemonkey812's original issue has been solved.

Thanks!

FormerLurker, 2020-02-26:
I'm going to go ahead and close this issue down since it seems to be resolved.  Thanks everyone!

closed 2020-02-26 by FormerLurker