FormerLurker/Octolapse #404: Unable To Enable Custom Preferences Raspicam v2.1

Lewis-Hashbrown, 2019-11-11:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: 
OctoPrint 1.3.12 running on OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Tried to set up custom image preferences for my Raspicam v2.1 follwing [this guide](https://github.com/FormerLurker/Octolapse/wiki/Configuring-a-Raspberry-Pi-Camera)
2.  After completing the guide and pressing "Enable And Apply Preferences Before Print" I get the following error

Unable To Enable Custom Preferences
unknown-server-type: An unknown webcam server type 'Werkzeug/0.15.4 Python/2.7.13' is being used for the 'Pi v2.1 Test' camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences. 

#### What should have happened?
The image preferences should of come up

#### What happened instead?
I get the above error and cant change any of the settings

#### Operating System running OctoPrint and Octolapse

OS Name:  OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Anet E12
Printer Firmware Version: Marlin

#### Browser and version of browser, operating system running browser

Browser: Google Chrome
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/LvviSRfv

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-11-11:
You must have missed the warning about the Octolapse version in that guide you linked to.  You'll notice on the first line is the following text:

```
This page is about a feature that is not yet available in any current release. This feature will be available starting with Octolapse V0.4rc1dev0.
```

You are running V0.3.4, which isn't compatible with that guide unfortunately.  If you want to try V0.4 alpha, let me know.

Lewis-Hashbrown, 2019-11-11:
> You must have missed the warning about the Octolapse version in that guide you linked to. You'll notice on the first line is the following text:
>
> ```
> This page is about a feature that is not yet [...]

Ah my bad! 

I've just downloaded and setup 
Octolapse 0.4.0rc1.dev2

When I click test under the Snapshot Address Template I get the following error

Camera Test Failed
Errors were detected - An unknown webcam server type 'Werkzeug/0.15.4 Python/2.7.13' is being used for the 'Test' camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences.

Not sure if this helps but the snapshot in the main octoprint camera tab (Timelapse Recordings) works

FormerLurker, 2019-11-11:
FYI, there are some known issues in 0.4.0rc1.dev2 related to the cameras and camera logging.  There are also some settings bugs that could cause issues in the 'Trigger' profiles (guid mismatches since fixed..).  I will shoot you a link to the devel branch, which has most of these bugs fixed, as soon as I can push the latest commits.

Lewis-Hashbrown, 2019-11-11:
That would be great! Thanks for the help!!

FormerLurker, 2019-11-11:
OK, upgrade from this link:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

IMPORTANT NOTE:  I push development changes there a lot, so don't use that link again unless I ask you :)  It should be safe for at least half a day :)

I still expect you to receive that error, but would like to see a log.  Set the debug profile to 'Debug', test your camera, and download and post the log. Hopefully there is more in there now :)

Lewis-Hashbrown, 2019-11-12:
I've just downloaded that version. Will have to test it later when back from work.

Would you like me to run a test print when in test mode - debug issues?

Lewis-Hashbrown, 2019-11-12:
[Octoprint.log](https://pastebin.com/QZx1zg1p)

[Plugin_octolapse.log](https://pastebin.com/mtmJLLbU)

FormerLurker, 2019-11-13:
I suspect octoprint is returning a 404 or something.
Can you try navigating to this URL (making sure to substitute your PIs IP where appropriate):

http://IP_OF_PI:8080/control.htm

Let me know if you land on a page full of camera controls.  If it does not, there could be either a proxy issue or something wrong with the octopi.txt file.  If that fails, can you post octopi.txt and your modules file (the one you edited in the guide)?  

then try this one:
http://IP_OF_PI:8080/input.json  

if that works, post copy and post the json contents for me to look at please.

FormerLurker, 2019-11-13:
OMG..  Are you running Octoprint Anywhere by chance?  In any case, try disabling your other plugins, reboot, then try again.  Octoprint anywhere has it's own streaming server I believe, and that's not supported by Octolapse yet, at least as far as the image controls go.

FormerLurker, 2019-11-13:
I just looked again at octoprint.log and answered my own question.  Try disabling Octoprint anywhere if you would.  I need to install that and see what's going on there.

Lewis-Hashbrown, 2019-11-13:
> I just looked again at octoprint.log and answered my own question. Try disabling Octoprint anywhere if you would. I need to install that and see what's going on there.

Will try when my Internet comes back on! Cheers for the help

Lewis-Hashbrown, 2019-11-17:
Sorry my Internet is still down! Will try ASAP

Lewis-Hashbrown, 2019-11-21:
Got my internet back!

I've had a whole load of problems though. Up to the point of the camera not working at all on any of the pages.

What I've done
Complete new install of octoprint.
Installed octolapse (Used https://github.com/FormerLurker/Octolapse/archive/devel.zip again. should be ok to use?)
The first thing i setup was custom image preferences for the raspicam following the above guide.
It now works perfectly! 

I've now set up all my plugins except for octoprint anywhere and it is still working fine.

I decided to give octoprint anywhere a go and it is hasn't caused any problems. The custom image preferences still work and the preferences still stay after I reboot octoprint.

Thank you so much for the help!

FormerLurker, 2019-11-22:
Awesome to hear!  So glad things are working now.  If you have any feedback on tye new version, i would love to hear it.

Lewis-Hashbrown, 2019-11-22:
Overall I love the design, nice and easy to use and the info buttons for the image preferences are great to see what they do.

Only thing I'd maybe change is when adjusting the preferences to maybe have the video preview open in a popup tab so you can always see it (saves scrolling up and down to see what it looks like after changing a setting)

FormerLurker, 2019-11-22:
Have you tried the edit pen on the octolapse tab on the upper right corner of the latest snapshot?  It is easy to miss, but the scroll issue has been solved there.  I need to make that more obvious somehow, but not sure how to do that.

Lewis-Hashbrown, 2019-11-22:
Ah my bad! That's perfect.

Perhaps add some text below the custom image preferences in the edit camera menu saying you can also edit by pressing the pen icon while still seeing the video feed?

FormerLurker, 2019-11-22:
I will add something there after next week (am traveling), and will post here once it is done.  Thanks!

FormerLurker, 2019-11-30:
So, I changed the camera profile page so there is now a button that is visible only when custom image preferences are enabled that will show the same popup window that is available on the Octolapse tab (the one that fixes the webcam stream).  I have pushed this update to the devel branch, but it still needs to testing and tweaking.  There are a few minor issues.  Most notably the webcam stream attempts to load more than once when used from the webcam dialog.

ShagoY, 2019-12-28:
Hi can i try the new "Custom Preferences" on dev branch ? 

> 
> IMPORTANT NOTE: I push development changes there a lot, so don't use that link again unless I ask you :) It should be safe for at least half a day :)
>

FormerLurker, 2020-02-26:
@ShagoY, I somehow missed your response (was sick at the time)!  So sorry...

YES you can try it out!  Go ahead and install using this url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Before installation, read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse), and pay attention to the slicer settings bits for whatever slicer you use to ensure the automatic settings detection works.

If you are using a picam, [read this guide](https://github.com/FormerLurker/Octolapse/wiki/Configuring-a-Raspberry-Pi-Camera).  Even if you are not using a picam, make sure you make the suggested changes to octopi.txt to gain access to control.htm, else the custom image preferences will not work.  Be sure to let me know how it goes!

FormerLurker, 2020-03-26:
It has been about a month, so I'm going to go ahead and close the issue.  If you want me to reopen it, I'd be happy to do so!

Thank you for your post!

closed 2020-03-26 by FormerLurker