FormerLurker/Octolapse #405: issues with running dslr

irishnick22, 2019-11-13:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. I first received the error in my first test run
2. I reproduce the error to copy its description by pasting it into the octolapse camera option that says "Before Print Start Script" because it has the test feature
3. STEP_...

#### What should have happened?
i believe it should have created the directory and pulled the image taken into it

#### What happened instead?
Instead I receive an error
Octolapse - Camera Settings Error
There were 1 errors while applying custom camera settings/scripts:
camera_initialization_error: The custom camera initialization script failed with the following error message: mkdir: cannot create directory ‘’: No such file or directory
ERROR: Could not get image.
The snapshot was not found in the expected directory: ''.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 with TH3D ezboard
Printer Firmware Version: not sure however nearly certain it is unnecessary 

#### Browser and version of browser, operating system running browser

Browser: FireFox
Browser OS: Win 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-11-13:
The test button doesn't work for the snapshot acquisition script, because it doesn't inject the correct parameters.  Did you follow [this](https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera) guide?  If so, try connecting to your pi (directly or via SSH) and run this command to test the script:

```
/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"
```

If your camera snaps a pic, you see test.jpg in the /home/pi/scripts folder, the test is successful.  

Next run a print, but change your Octolapse debug profile to 'Test Mode - Full Diagnostic'.  Please note that 'Test Mode' will prevent your printer from warming up or extruding plastic, which is pretty convenient for testing.  Make sure you switch the profile back to 'Live Print' once you are finished.  If you encounter an error, navigate to Octoprint Settings->Logs and download plugin_octolapse.log and post it to gist.github.com, then leave a link here to the gist you created.

Also, have you been able to successfully use Octolapse without a DSLR?  Getting cameras to work via script can be tricky, and in order to simplify things I recommend you first get a webcam working.

Thanks!

irishnick22, 2019-11-13:
I'll start in on setting everything back up. I have run Octolapse to much success, with a Picam v2. I just wanted to see what my output could be with a T5i. 

I tried the whole set up a few days ago and got the same error. so i wiped my sd and started from fresh Octoprint/Octopi install. 

The setup link you've provided is exactly what I followed and all the command line tests work throughout the process and I don't hit a snag until testing with the complete system.

FormerLurker, 2019-11-13:
Ok, thanks for clarifying.  Make sure your camera is set to only take JPG images, not RAW images.  Also, MP4s might be unwatchable at resolutions higher than 4k (been working a lot on that this week, but it seems to be a limitation in all of my video players).  You might want to turn your rendering bitrate (in the octolapse rendering profile) up too, because the default is way too low for high res images.  Something like 20M would be a minimum.  If you get a video to render (it may take a long while to complete), and it wont play, you can repair it with a program like handbrake.  Let me know how it goes, ot if you have any more problems!

FormerLurker, 2019-11-30:
Hey @irishnick22, have you had a chance to look at my last response?  I'm going to mark this thread 'Waiting for Response'.  If I don't hear back from you after a while, I'll go ahead and close the issue.  I hope to hear from you soon!

FormerLurker, 2020-02-26:
This issue has been open for a while, so i'm going to go ahead and close it.  If you still need help here, please respond back and I will reopen the issue.  Thanks for your post!

closed 2020-02-26 by FormerLurker