FormerLurker/Octolapse #406: Exception thrown when Printer is not set.

buddydvd, 2019-11-14:
#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
1. Install Octolapse (**without** specifying a printer in the settings page)
2. Start a print job
3. Inspect `plugin_octolapse.log`

#### What should have happened?
Expect to not see any errors in `plugin_octolapse.log`.

#### What happened instead?
Sees the following error on repeat every 5 seconds in `plugin_octolapse.log`:

    ....
    2019-11-12 17:31:00,521 Traceback (most recent call last):
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 1318, in on_gcode_queuing
        return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 568, in on_gcode_queuing
        if self._is_snapshot_command(self.Settings.current_printer(), command_string):
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 1012, in _is_snapshot_command
        return command_string.lower() == printer.snapshot_command.lower()
    AttributeError: 'NoneType' object has no attribute 'snapshot_command'
    ....

This issue causes the the `plugin_octolapse.log` files to balloon to 100MB+ (depends on active print time).

#### Operating System running OctoPrint and Octolapse
OctoPi 0.16.0; Linux octo-1 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

#### Notes ####
Specifying a Printer in the settings page stops Octolapse from throwing the exception.

FormerLurker, 2019-11-14:
Wow, it isn't even supposed to start without a printer selected.  Was octolapse disabled?  I doubt this is an issue in V0.4, but I will try hard to replicate this to make sure it doesn't happen in the next release.  Thanks for the report!  I will let you know what i find.

buddydvd, 2019-11-15:
Octolapse — the plugin — is **enabled** in *OctoPrint Settings* -> *Plugin Manager* -> *Installed Plugins*.

Octolapse — the timelapse functionality — is **disabled** in the *Octolapse Tab*.

I tested it again just now. I found out you don't even need to start a print job to trigger the error.

- Install Octolapse
- Press the **Connect** button in the **Connection** panel.
- Observe the `plugin_octolapse.log` (e.g., `tail -f plugin_octolapse.log`)

Octolapse just throws the same exception every five seconds:

```
2019-11-14 15:52:38,731 Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 1318, in on_gcode_queuing
    return self.Timelapse.on_gcode_queuing(cmd, cmd_type, gcode, kwargs["tags"])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 568, in on_gcode_queuing
    if self._is_snapshot_command(self.Settings.current_printer(), command_string):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/timelapse.py", line 1012, in _is_snapshot_command
    return command_string.lower() == printer.snapshot_command.lower()
AttributeError: 'NoneType' object has no attribute 'snapshot_command'
```

Hope the extra information helps.

FormerLurker, 2019-11-15:
Yes it does.  So, no printer selected, but Octolapse functionality is disabled (not the entire plugin).  That is specific enough that it should be easy to fix!  I'm looking into it now.

buddydvd, 2019-11-15:
https://github.com/FormerLurker/Octolapse/blob/617269a870aa1bbca088874e7cf6e10657346773/octoprint_octolapse/__init__.py#L1312-L1323

It seems like following line (along with a few others):
```python
         if self.Timelapse is not None: 
```
should be replaced with:
```python
         if self.Timelapse is not None and self.Timelapse.is_timelapse_active(): 
```
since `self.Timelapse` always gets assigned in `on_after_startup()` via `self.create_timelapse_object()`:
https://github.com/FormerLurker/Octolapse/blob/617269a870aa1bbca088874e7cf6e10657346773/octoprint_octolapse/__init__.py#L804-L817

Also, I think it would make sense to inspect all the `self.Timelapse is not None` conditional expressions to make sure they make sense and that they can standalone without the additional `self.Timelapse.is_timelapse_active()` check.

FormerLurker, 2019-11-15:
Hey, thanks a TON for your review!  However, timelapse.py is a LOT different in v0.4 (check the devel branch), and I just verified that the bug does not exist in that branch.

However, it looks like I took your advice mostly.  The check for ```if self._timelapse is not None``` is indeed sufficient.  The timelapse.py call to  '''on_gcode_queuing''' now must proceed even if the timelapse isn't running because of a new option to suppress the pseudo-gcode snapshot command even when Octolapse is disabled.  I did just push an update to clean up that section somewhat:

https://github.com/FormerLurker/Octolapse/blob/a5489b8ce1b3d3ca533e8a4d24c69b9d99027ba6/octoprint_octolapse/timelapse.py#L623-L637

FormerLurker, 2019-11-15:
FYI, your bug report here may have been the most thorough I've ever seen.  I love it!  It's SOOO different than what I'm used to seeing :)

buddydvd, 2019-11-15:
Thanks. Great job to your project. Look forward to the new version!

closed 2019-11-15 by buddydvd

FormerLurker, 2019-11-15:
You are welcome to try out the devel version.  I'm hoping to push v0.4.0rc1.dev3 very soon.  You can use this url to install via the plugin manager: https://github.com/FormerLurker/Octolapse/archive/devel.zip

It is pretty stable at this point.  [Here is a guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) you might want to look at if you decide to give it a go.

Let me know if you ever have any questions or problems!