FormerLurker/Octolapse #408: Internal Server Error (No guid)

shaven, 2019-11-18:
After installing Octolapse, and I restart Octoprint I get this error:

- Octolapse was unable to load the current settings.  Status: error.  Error: INTERNAL SERVER ERROR

When I go into settings to add my printer the button doesn't work and I am seeing this in the console:

- packed_plugins.js?12d99356:5403 Uncaught TypeError: Cannot read property 'guid' of undefined

    at new Octolapse.DebugProfileViewModel (packed_plugins.js?12d99356:5403)
    at Octolapse.ProfilesViewModel.self.getNewProfile (packed_plugins.js?12d99356:3011)
    at Octolapse.ProfilesViewModel.self.showAddEditDialog (packed_plugins.js?12d99356:3079)
    at Octolapse.ProfilesViewModel.eval (eval at parseBindingsString (packed_libs.js?ef1e2987:11374), <anonymous>:1:82)
    at HTMLButtonElement.<anonymous> (packed_libs.js?ef1e2987:11395)
    at HTMLButtonElement.dispatch (packed_libs.js?ef1e2987:5184)
    at HTMLButtonElement.elemData.handle (packed_libs.js?ef1e2987:4992)
Octolapse.DebugProfileViewModel @ packed_plugins.js?12d99356:5403
self.getNewProfile @ packed_plugins.js?12d99356:3011
self.showAddEditDialog @ packed_plugins.js?12d99356:3079
eval @ VM11791:1
(anonymous) @ packed_libs.js?ef1e2987:11395
dispatch @ packed_libs.js?ef1e2987:5184
elemData.handle @ packed_libs.js?ef1e2987:4992
packed_libs.js?ef1e2987:9601 POST https://shoctopi2.haven.local/plugin/octolapse/loadState 500 (INTERNAL SERVER ERROR)
send @ packed_libs.js?ef1e2987:9601
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
setTimeout (async)
error @ packed_plugins.js?12d99356:1393
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
done @ packed_libs.js?ef1e2987:9308
(anonymous) @ packed_libs.js?ef1e2987:9549
load (async)
send @ packed_libs.js?ef1e2987:9568
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
packed_libs.js?ef1e2987:9601 POST https://shoctopi2.haven.local/plugin/octolapse/loadState 500 (INTERNAL SERVER ERROR)
send @ packed_libs.js?ef1e2987:9601
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
setTimeout (async)
error @ packed_plugins.js?12d99356:1393
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
done @ packed_libs.js?ef1e2987:9308
(anonymous) @ packed_libs.js?ef1e2987:9549
load (async)
send @ packed_libs.js?ef1e2987:9568
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
setTimeout (async)
error @ packed_plugins.js?12d99356:1393
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
done @ packed_libs.js?ef1e2987:9308
(anonymous) @ packed_libs.js?ef1e2987:9549
load (async)
send @ packed_libs.js?ef1e2987:9568
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
packed_libs.js?ef1e2987:9601 POST https://shoctopi2.haven.local/plugin/octolapse/loadState 500 (INTERNAL SERVER ERROR)
send @ packed_libs.js?ef1e2987:9601
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
setTimeout (async)
error @ packed_plugins.js?12d99356:1393
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
done @ packed_libs.js?ef1e2987:9308
(anonymous) @ packed_libs.js?ef1e2987:9549
load (async)
send @ packed_libs.js?ef1e2987:9568
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
setTimeout (async)
error @ packed_plugins.js?12d99356:1393
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
done @ packed_libs.js?ef1e2987:9308
(anonymous) @ packed_libs.js?ef1e2987:9549
load (async)
send @ packed_libs.js?ef1e2987:9568
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370
setTimeout (async)
error @ packed_plugins.js?12d99356:1393
fire @ packed_libs.js?ef1e2987:3269
fireWith @ packed_libs.js?ef1e2987:3399
done @ packed_libs.js?ef1e2987:9308
(anonymous) @ packed_libs.js?ef1e2987:9549
load (async)
send @ packed_libs.js?ef1e2987:9568
ajax @ packed_libs.js?ef1e2987:9207
OctolapseViewModel.self.loadState @ packed_plugins.js?12d99356:1379
OctolapseViewModel.self.getInitialState @ packed_plugins.js?12d99356:1370

#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.12 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

This happens right after I restart Octoprint after installing.

#### What should have happened?
I'm hoping to be able to set my printer in the config

#### What happened instead?
Nothing visually. Error in web browser console.

#### Operating System running OctoPrint and Octolapse

OS Name: MAC (Mojave)
Os Version: 10.14.6

#### Printer model & used firmware incl. version

Printer Model: Reprapguru Prusa i3
Printer Firmware Version: Marlin v4

#### Browser and version of browser, operating system running browser

Browser: 78.0.3904.87
Browser OS: Chrome on Mac

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Not related to Gcode File

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  The file is blank. Guessing it isn't initing enough to put anything in the file.

#### Link to octoprint.log

Link to octoprint.log:  http://www.piuid.com/octoprint.log

#### Link to contents of Javascript console in the browser

Link to javascript console output: Unrelated

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: None

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-11-18:
Hmm, looks like there is a 500 error while trying to load the settings from the server.  I am traveling right now, but will take a look at the log when im in front of my laptop.  Thanks for the report!

FormerLurker, 2019-11-30:
Sorry for the long delay.  I was out of commission for the last two weeks.  Can you try restoring the default settings in Octolapse?  You can do this by opening the OctoPrint settings (Wrench/Spanner Icon in top right corner of OctoPrint), clicking on the Octolapse link in the left menu.  Then click 'Restore Default Settings' in the Octolapse settings tab.  FYI, the button may have a slightly different title (I haven't ran 0.3.4 in a while, and the text may have changed for this button).

Let me know if that helps, and sorry for the delay!

FormerLurker, 2020-02-26:
So it's been a while since this issue has been updated, so I'm going to assume you've been able to solve it.  If you still need help debugging, please post back and I will reopen it.  Thanks!

closed 2020-02-26 by FormerLurker