FormerLurker/Octolapse #409: Can't run bash script to reach my websever

Nightowl3D, 2019-11-18:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4 

#### Version of OctoPrint

OctoPrint Version: 1.3.11 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

On the one hand I make pictures with my webcam on the other hand I wanted to reach my websever with my correct written bash script (is running without errors in the command line of octopi with sudo ./BeforeSnapshotScript.sh) before every snapshot.

#### What should have happened?

My websever should be reached

#### What happened instead?
I get this error, while printing:

snapshot_script_error: The snapshot script failed with the following error message: The process does not exist

or this one if I Test the Script with the Test Script button:

There were 1 errors while applying custom camera settings/scripts:
camera_initialization_error: The custom camera initialization script failed with the following error message: The process does not exist 

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 
Os Version:  0.17.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3s
Printer Firmware Version: 3.8.0 - 2684

#### Browser and version of browser, operating system running browser

Browser: Firefox
Browser OS: Windows 7

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
https://www.bilder-upload.eu/bild-82c3a4-1574082052.png.html

https://www.bilder-upload.eu/bild-f7f67f-1574082085.png.html

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-11-18:
Hey, I didn't have a chance to get to this yet, but Im guessing you figured it out.  Can you let me know how you solved this?

Nightowl3D, 2019-11-20:
I have just written the bash script into a new file. Now it is running alright, but I really don't know the difference :D

Nightowl3D, 2019-11-20:
Perhaps you can help me with another question:
I wan't to adjust the Snapshot Timeout more than 25000.
I have already found the line in your source code under /templates/octolapse_profiles_camera.jinja2, where I have to change something.
I want to change this local on my octopi - is this possible?
I cant find the whole octolapse folders..
I cant open the ocotprint folder under /home/pi/oprint/bin there.. 
Is there a other way to do this? With the plugins maybe?
(https://docs.octoprint.org/en/master/plugins/gettingstarted.html#frontend-fun-how-to-add-functionality-to-octoprint-s-web-interface)

Thanks a lot!

Thanks alot!

FormerLurker, 2019-11-20:
/home/pi/.octoprint/data/octolapse/settings.json

However..  25 seconds is a ton!  Why such a long timeout?  That will severely affect print quality.

Nightowl3D, 2019-11-26:
I need this time because I want to try something with my webserver..
I have already tried this, but it is not working - the system himself adjust the snapshot timeout into 25000 again.. where can I change this permanent?
How can I edit the source code under /templates/octolapse_profiles_camera.jinja2 on my raspberry?
Thanks a lot!

Nightowl3D, 2019-11-26:
Ok it doesnt adjust the snapshot time into 25000 again, however it doesnt wait more than 25 seconds.
I get this error again:

snapshot_script_error: The snapshot script failed with the following error message: - The snapshot script timed out in 25.0 seconds.

FormerLurker, 2019-11-26:
Hey there!  I am traveling at the moment, but will try to point you to the correct jinja2 file asap.  Also, i will drastically increase the timeout max for the next release.

FormerLurker, 2019-11-26:
Lol, found it right away.  Here you go:https://github.com/FormerLurker/Octolapse/blob/617269a870aa1bbca088874e7cf6e10657346773/octoprint_octolapse/templates/octolapse_profiles_camera.jinja2#L152

Nightowl3D, 2019-11-26:
yes this i have also found.
But i dont find it on my raspberry.. is it even possible to change the code on the raspberry?

FormerLurker, 2019-11-26:
It is buried somewhere (NOT in home/pi/.octoprint/data/octolapse).  I recommend running a recursive search from /home/pi for octolapse_profiles_camera.jinja2

Let me know if/where you end up finding it.  I should probably know these things.  If you cant locate it, I will try to find it as soon as I can, prob Saturday (am out of town).

Nightowl3D, 2019-11-26:
ok i try it again.. now i know that it is possible to find :D 

Thank you so far! Dont worry, you cant know everything!
I keep you updated.

Nightowl3D, 2019-11-26:
ok i did the recursive search and found it under:

/home/pi/oprint/lib/python2.7/site-packages/octoprint_octolapse/templates/

thank you!

FormerLurker, 2019-11-26:
Excellent 👍. Thanks for letting me know, and let me know if the timeout works after you increase it.

FormerLurker, 2019-11-30:
@Nightowl3D, I'm going to mark this thread waiting for response.  Just let me know if the timeout worked and I'll close the issue.  If not, I'm about to increase the timeout in the current devel branch, which you are welcome to try.  I'll post back here when this is complete.

FormerLurker, 2019-11-30:
Ok, I've removed the max timeout completely in https://github.com/FormerLurker/Octolapse/commit/af5b405cc48af822360eaee67f0f8124616015a3.

Nightowl3D, 2019-12-02:
Everything was working fine after increasing it. Thanks again!

closed 2019-12-02 by Nightowl3D