FormerLurker/Octolapse #410: Octolapse Startup Failed

leebyungjun1583, 2019-11-20:
Unable to start the timelapse. Cancelling print. Error: Some required slicer settings are missing from your gcode file. Unable to proceed. Missing Settings: Retraction Length,Retraction Speed,Deretraction Speed,X/Y Travel Speed,Z Lift Height,Z Travel Speed,Retract Before Move,Lift When Retracted

FormerLurker, 2019-11-20:
Please post your gcode file to gist.github.com, and let me know what version of octolapse you are using.  Thanks!

FormerLurker, 2019-11-30:
@leebyungjun1583, just checking in.  When you get a chance be sure to post your gcode file and your Octolapse version, and I'll take a look.  The error you are seeing is usually pretty easy to correct.  If I had to guess, I'd say you are using Cura and are missing the settings gcode script in cura's start gcode for your machine.  If that's the case, [try this guide](https://github.com/FormerLurker/Octolapse/wiki/Automatic-Slicer-Settings#install-the-cura-settings-script).

FormerLurker, 2020-02-26:
This issue has been open and waiting for a response for some time, so I'm going to assume you were able to resolve your issue and am going to close this thread.  If you want me to reopen it, just respond back.

Thanks!

closed 2020-02-26 by FormerLurker

FormerLurker, 2020-09-03:
@the-dbp, please see the [Automatic Slicer Settings guide within the Octolapse Wiki](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration).  Be sure to follow the [extra steps for Cura]() mentioned within the guide.