FormerLurker/Octolapse #411: Is this an error

Muscats133, 2019-11-30:
On this Wiki page 
https://github.com/FormerLurker/Octolapse/wiki/Troubleshooting#for-xyz-choose-one-of-the-following-options

regarding extruder settings it say to choose from gcode M84(Absolute) or M83(Relative), is this an error, should it be M82(Absolute) or M83(Relative)

For Extruder (E), choose one of the following options:
Set your extruder axis mode in your start gcode. This usually involves adding a M84(Absolute) or M83(Relative) command to the start GCode. Which code you need depends on your slicer settings. See 'Modifying your start gcode' below for more information.
Set your E axis mode to 'Absolute' or 'Relative' within the Octolapse printer profile. If you are SURE that your printer's E axis will always be in either relative or absolute mode, you can set this here without adding M84/M83 to your start gcode. I don't recommend this option and believe that requiring an explicit gcode to set the mode and adding the proper GCode to the start GCode is a better and safer option.

FormerLurker, 2019-11-30:
Yes, this was definitely a typo.  I have corrected it in the wiki.  Let me know if my edits look good, or if I've missed anything.  

Thank you for reporting this!

Muscats133, 2019-12-01:
yes, looks good now. Thanks for developing a great plugin.
I did have another issue. After confirming that I had the the correct codes I continued to get an error message about the requirement for the codes, the only way to stop the error was to change to "Default To Absolute", so it seems it is not seeing the Gcodes, you can see in the attached file that the codes are actually in there twice because I added them through Slic3r after receiving the error 

[Test bolt 2.txt](https://github.com/FormerLurker/Octolapse/files/3907950/Test.bolt.2.txt)

FormerLurker, 2019-12-01:
Move the G90 and M82 above the G28 command and it will stop complaining :)

FormerLurker, 2019-12-01:
Oh, fyi, those errors have been removed from v0.4.  Now the error will create a new popup with a help button that shows suggests several  resolutions.  Also, all printer.profiles are set to default.to absolute.  I also added another option for XYZ and E axis modes to force absolute/relative modes which automatically sends the proper gcodes before the print starts.

FormerLurker, 2019-12-16:
Did my earlier suggestion help?

FormerLurker, 2020-02-26:
So, I'm going to assume that this is all resolved now, and will close the issue.  Feel free to post here again if you want this reopened, or if you find any other bugs/issues!

closed 2020-02-26 by FormerLurker