FormerLurker/Octolapse #412: [Request] Support for M118 to trigger a snapshot

RomRider, 2019-11-30:
#### If this is a feature request describe it here

I print a lot from the virtual SD Card using klipper because Octoprint makes my printer stutter for high poly objects and I was wondering if there would be a way to use the `M118 //action:xxxxx` to trigger a snapshot from octoprint.

The idea is that my slicer would add the proper gcode to park the head in the corner on every layer change, send the M118 to the host, Octoprint would then take the snapshot and the printer would resume.
Obviously, in this mode, it wouldn't be possible to use all the settings from Octoprint but it would be a nice way to trigger snapshots for prints from the sdcard.

What do you think?

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

FormerLurker, 2019-12-07:
Interesting idea!  I will look into this possibility.  It would also be possible to use Octolapse to generate the necessary gcode to park the head, and to create a new gcode file including these motions and the M118 command.  I will think about this and may run some tests.

However, my ability to add new features is somewhat hampered by the fact that I've been working on a new release for over a year now, lol!  About once a month for the past six months I've thought I'd have a new release, but things keep popping up (the release of OctoPrint 1.4.0 and python 3 compatibility was the most recent complication).  I REALLY feel like I'm getting close now.

I'm going to mark this as an enhancement and will update as things progress.  Let me know in the meanwhile if you have any thoughts or new ideas about this enhancement!

Thank you for your contribution!

darkporpoise, 2019-12-28:
This sounds like a great idea as I also prefer to print from SDCard for quality.

Would you pause the print in the GCODE before sending the `M118` and have OctoLapse handle the resume?

I'm trying to think how this could be done in such a way that the print works even if Octolapse isn't listening.