FormerLurker/Octolapse #413: Octolapse cancelling print at start

AnduriI, 2019-12-01:
#### If this is a feature request describe it here
FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
Octolapse Version: Octolapse 0.4.0rc1.dev2

#### Version of OctoPrint
OctoPrint Version: OctoPrint 1.4.0rc1 auf OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Trying to start a print
2. was cancelled by octolapse

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
plugin_octolapse.log said:
```
2019-11-24 19:47:45,443 - octolapse.timelapse - INFO - Print Start Detected.  Command: M110 N0, Tags:set(['trigger:comm.start_print', 'source:job', 'trigger:comm.reset_line_numbers'])
2019-11-24 19:47:45,446 - octolapse.__init__ - INFO - Print start detected, attempting to start timelapse.
2019-11-24 19:47:45,716 - octolapse.__init__ - ERROR - Unable to start the timelapse.  Cancelling print.  Error:  Unable to start the print.  Some required slicer settings are missing or corrupt: Retract Before Move,Lift When Retracted
2019-11-24 19:47:46,103 - octolapse.__init__ - INFO - Print cancelling.
2019-11-24 19:47:46,453 - octolapse.__init__ - INFO - Print cancelled.
2019-11-24 19:47:46,511 - octolapse.__init__ - INFO - homing to payload:%s.
2019-11-24 19:47:46,886 - octolapse.__init__ - INFO - Print failed.
```

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: CR-10mini
Printer Firmware Version: Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/AnduriI/085ed05764b5da9b16391030297191c4

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  already under what happened instead

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2019-12-01:
This is the problem:

> Some required slicer settings are missing or corrupt: Retract Before Move,Lift When Retracted

My guess is that you are using Cura to slice your gcode file, and you have your Octolapse printer profile set to 'Automatic' slicer configuration, right?  What version of cura are you using exactly?  Take a look at [this guide](https://github.com/FormerLurker/Octolapse/wiki/Automatic-Slicer-Settings#install-the-cura-settings-script) and pay attention to the cura versions.  Let me know if that helps.

Thanks for posting!

AnduriI, 2019-12-01:
hey FormerLurker,

thanks for your instant reply. You are correct about the file I uploaded, but I also tried with the same model sliced in prusaslicer. I uploaded it to gist as well: https://gist.github.com/AnduriI/4576d37851865889b9939b011cc28037
I will try your starting gcode and report back asap.

AnduriI, 2019-12-01:
still the same error with the updated starting gcode.
I think this happens since I updated to the devel rc of octoprint. Do you have an instance of the devel rc running for testing purpose yourself?

FormerLurker, 2019-12-01:
Do you mean v1.4.0rc1?  I have been debugging that all day :)  Mostly python 3 compatibility stuff. 
 

Double check your Octolapse printer profile and make sure it's set to automatic.  If not, you might well be missing settings in your manual profile.  If that doesn't fix things, please export and post your octolapse settings (click the gear icon in the octolapse tab, and click export settings in the popup).

AnduriI, 2019-12-08:
Well I think I forgott to change the setting in Octolapse to cura 1.4.2 and above after my last upgrade of cura. But I don't know why this did not caused any problems on upgrades to cura 1.4.2 and 1.4.3... well at least it's working now :) thanks for your help

closed 2019-12-08 by AnduriI

FormerLurker, 2019-12-08:
Well, glad you got it to work, but have you tried the 'Automatic' slicer type (as opposed to cura 1.4.2)?  I'm curious if that worked for you.  Please again take a look at [this guide](https://github.com/FormerLurker/Octolapse/wiki/Automatic-Slicer-Settings#install-the-cura-settings-script) and make sure you have the appropriate gcode script added to your cura start gcode!

AnduriI, 2019-12-12:
Hey FormerLurker,
I have now set it to 'Automatic' and it seems to work. I have added the gcode from the guide.
Thank you very much for your help.

FormerLurker, 2019-12-12:
Perfect!  Let me know if you have any other issues, and thanks for helping me test the rc-devel version!  I will be dropping a new one hopefully soon.