FormerLurker/Octolapse #414: Live View Mode Canon 77D

akku3D, 2019-12-07:
I have set the Live View mode on my camera (Canon 77D), so that the mirror does not fold up with every picture.
However, if octolapse has taken a picture, the camera will exit Live View mode.
Can I make a setting somewhere that preserves Live View mode?

I've added the config file from gphoto. 
Please can you have a look at it?

[canon 77D.txt](https://github.com/FormerLurker/Octolapse/files/3934845/canon.77D.txt)

FormerLurker, 2019-12-07:
Ok, I think there is a way to do this, but you will have to test it.  Create another script file called ```live-preview.sh```.  Make it look like this:

```
#!/bin/sh
# Turn on live view script
# Written by: Formerlurker@pm.me

gphoto2 --show-preview
```

Save the script, and add the necessary permissions just like in the original DSLR guide you watched via the ```chmod +x live-preview.sh``` command.

Next execute the live-preview.sh command manually and make sure it turns on live view.  It might be helpful for debug purposes to run it a second time and see if there are any errors.  Let me know how this step goes please.

Once you've tested the live-preview script, go into your DSLR camera profile in Octolapse and add the newly created script as an 'After Snapshot Script'.  It should be set to something like the following (may vary depending on where your scripts are located): ```/home/pi/scripts/live-preview.sh```

In theory you should now be good to go, but I think we can do better.  If you stopped here you would need to make sure that your camera is set to live view BEFORE starting Octolapse, else your shutter wouldn't be open for the very first snapshot.  However, you could also add the ```live-preview.sh``` script to the Before Print Start Script, which should switch your camera to live view at the beginning of the print.  If you choose to do this (and I recommend that you do), your 'Before Print Start Script' should be close tot the following (again, depending on where the script is located): ```/home/pi/scripts/live-preview.sh```

Let me know if that helps.  I have NOT tested this, but intend to at some point soon.

Thanks for posting your issue!

FormerLurker, 2020-02-15:
This issue has been open for a long while without a response, so I'm going to close it now.  If you want me to reopen it, please let me know.  Thanks!

closed 2020-02-15 by FormerLurker