FormerLurker/Octolapse #415: Compatibility with OctoPrint V1.4.0

FormerLurker, 2019-12-07:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

pebsi, 2020-02-08:
Hi, I am not sure if this is the right place for this, but I have downloaded octolapse for the first time with the dev version for 1.4.
I am having problems setting up a new profile for my printer. I am using the Cura 4.2 and above setting and even after filling in all the fields it keeps telling me that "The selected printer profile has not been configured. Please edit the profile and copy in your slicer settings."

FormerLurker, 2020-02-08:
@pebsi, there is no Octolapse version 1.4.  I'm guessing you mean 0.4.  What version exactly did you install?  If you didn't install from the plugin manager with this url (in the get more->from url... box):  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Let me know how it goes.

pebsi, 2020-02-08:
Sorry, I meant Octoprint 1.4 rc4 and octolapse 0.4 rc1dev2.
I still couldn't get either of the cura profiles to work even when filling in all required info.
But I have now downloaded the script for the auto setting and placed in Cura and will try it tomorrow with a new print. If that works then I guess it wont really matter for me if the Cura profiles don't ;)

FormerLurker, 2020-02-09:
Ok, so you installed the last devel rc, and the issues you encountered are known in that build.  Please re-install using this URL exactly:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Note that it is ever so slightly different than the one you used (/rc/devel).  You may well encounter some settings migration errors, so BEFORE you upgrade, please export your settings by clicking going to OctoPrint settings (gear/spanner icon)->Octolapse->Export Settings.  Then upgrade Octolapse with the URL above by opening the plugin manager and pasting it into this box:

[image]

After it installs, and you reboot, if you have any errors (Octolapse doesn't load, settings are wrong, etc), please post your exported settings to gist.github.com and paste a link into this issue.  I should be able to figure out what went wrong if I have your settings.

ASMfreaK, 2020-02-18:
I have issues installing Octolapse. 
First time I had th following issue:
```
2020-02-18 08:56:59,992 - octoprint.plugin.core - ERROR - Error loading plugin octolapse
Traceback (most recent call last):
  File "[VENV]/lib/python3.7/site-packages/octoprint/plugin/core.py", line 937, in _import_plugin
    module = _load_module(module_name, spec)
  File "[VENV]/lib/python3.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "[VENV]/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 241, in load_module
    return load_package(name, filename)
  File "[VENV]/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 215, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 44, in <module>
    from flask_principal import Permission, RoleNeed
ModuleNotFoundError: No module named 'flask_principal'
```

I've entered the venv and installed `flask_principal`. After a restart the Octolapse tab showed up.

ASMfreaK, 2020-02-18:
Tried printing and got the following error:
```
2020-02-18 09:14:58,700 - octolapse.__init__ - ERROR - Unable to start the timelapse.
Traceback (most recent call last):
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 2397, in on_print_start
    timelapse_settings = self.get_timelapse_settings()
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 2624, in get_timelapse_settings
    success, error_type, error_list = current_printer_clone.get_gcode_settings_from_file(gcode_file_path)
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 682, in get_gcode_settings_from_file
    missing_settings = new_slicer_settings.get_missing_gcode_generation_settings(slicer_type=self.slicer_type)
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 2621, in get_missing_gcode_generation_settings
    settings = self.get_gcode_generation_settings(slicer_type=slicer_type)
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 2791, in get_gcode_generation_settings
    settings.extruders = self.get_extruders()
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 2780, in get_extruders
    extruders.append(extruder.get_extruder(self))
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 2689, in get_extruder
    extruder.z_lift_speed = self.get_speed_travel_z()
  File "[VENV]/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 2734, in get_speed_travel_z
    LooseVersion(self.version) >= LooseVersion("4.2.0")
  File "/usr/lib/python3.7/distutils/version.py", line 70, in __ge__
    c = self._cmp(other)
  File "/usr/lib/python3.7/distutils/version.py", line 337, in _cmp
    if self.version < other.version:
TypeError: '<' not supported between instances of 'str' and 'int'
```

FormerLurker, 2020-02-18:
@ASMfreaK, what version of Octolapse are you using?

ASMfreaK, 2020-02-18:
I'm using Octolapse 0.4.0rc1.dev3

FormerLurker, 2020-02-18:
@ASMfreaK, I fixed the first issue (missing flask_principal).  The second one has me stumped.  Can you post your gcode to gist.github.com, as well as your settings.json file?  You can export the settings file by clicking on the gear icon on the Octolapse tab, then clicking *Export Settings*.

Also, you might want to re-install octolapse from the rc/devel branch if you haven't already.  I just officially released that yesterday.  Here is a URL you can use to install from the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/0.4.0rc1.dev3.zip

willhoh, 2020-02-19:
Hi, I am using Octoprint 1.4.0rc5 and this on a Pi with Python3. Now I wanted to install Octolaps to make nice videos of my prints. From the repository I get the message that Octolaps is incompatible (not surprising), but when I use the archives (links) mentioned here I always get an error after a long time of installation. I hope to get help here and if necessary to be helpful myself to fix this error for others. 

`Requirement already satisfied: pytz>=2015.7 in /home/pi/OctoPrint/venv/lib/python3.7/site-packages (from Babel>=2.3->Flask-Babel<0.13,>=0.12->OctoPrint>1.3.8->Octolapse==0.4.0rc1.dev3) (2019.3)
Building wheels for collected packages: Octolapse
Building wheel for Octolapse (setup.py): started
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): still running...
Building wheel for Octolapse (setup.py): finished with status 'done'
Created wheel for Octolapse: filename=Octolapse-0.4.0rc1.dev3-cp37-cp37m-linux_armv6l.whl size=3235525 sha256=8081f5110763f5e5604b6ea80db076de11278265b5b01e4edb4f00ad8bed602b
Stored in directory: /tmp/pip-ephem-wheel-cache-8m_bsy85/wheels/ce/e8/36/c5d2003f955525518ffa06c5164d724ed616c34ee65b92c637
Successfully built Octolapse
Looks like the plugin was already installed. Forcing a reinstall.
/home/pi/OctoPrint/venv/bin/python3 -m pip --disable-pip-version-check install https://github.com/FormerLurker/Octolapse/archive/0.4.0rc1.dev3.zip --no-cache-dir --ignore-installed --force-reinstall --no-deps
Error!
Could not install plugin, unknown error, please consult octoprint.log for details`

FormerLurker, 2020-02-19:
@willhoh, this is a known issue, but I've never been able to recreate it.  You might want to try rebooting.  Other users have said that despite this error message, the plugin actually installs.  I suspect it is a timeout issue somewhere, since the new version take quite a while to install (maybe 3 min) it has to compile a bunch of c++).  I will try to dig in more.  I'd really like to get this fixed.

willhoh, 2020-02-19:
Thanks for the answer. I must point out that I use a Pi 1 B+. Here the timeout could be very likely. Because it is not the first time I try to install Octolaps, I always get the message 'was already installed, forcing a reinstall'. Can this be a reason? How do I uninstall a plugin via the console? Octoprint does not show me Octolaps.

FormerLurker, 2020-02-19:
@willhoh, no problem!  Octolapse will eat your 1B+ alive I'm afraid :(  I think it will kill your print quality to use Octolapse on that board.

Regarding the 'was already installed' message, you can ignore that.  Doesn't hurt a think, and you do not need to uninstall Octolapse to upgrade.

Regarding installing it from the console, that's a good question.  I will look into it.

FormerLurker, 2020-02-19:
@ASMfreaK, I believe I fixed your issue.  It was caused by the version number reported by cura being 'master' and not the expected semantic version number (4.x.x).  I found a way to remove the necessity of the version number, so long as that line exists somewhere like so:

''';Generated with Cura_SteamEngine {any version number here}''

I'm going to see if I can make it work even without that line.  Will report back.

FormerLurker, 2020-02-19:
@ASMfreaK, ok, I got it working even if the version number line is removed for all known versions of cura.  It was a rather simple change.  Pushed to the devel branch at https://github.com/FormerLurker/Octolapse/archive/0.4.0rc1.dev3.zip

Also, I'm going to open up a ticket for v0.4.0rc1.dev3.  Most of these discussions belong there actually.  Just forgot to create that issue :(

FormerLurker, 2020-02-19:
[Here is an issue](https://github.com/FormerLurker/Octolapse/issues/452) for reporting any feedback regarding V0.4.0rc1.dev3.  I've added all issues here that were related to that release that were not listed as *Known Issues* in the release notes.  I omitted any issues that were fixed between v0.4.0rc1.dev2 and dev3.  

Unless it relates specifically to OctoPrint 1.4.0 (or if you don't know that's fine too), please post in the new issue.  Thanks!

FormerLurker, 2020-02-22:
A new devlopment rc has been released that includes all bugs found and fixed from V0.4.0rc1.dev3.  Please see [this issue](https://github.com/FormerLurker/Octolapse/issues/455) regarding the new version.  You can read the release notes [here](https://github.com/FormerLurker/Octolapse/releases/tag/0.4.0rc1.dev4).

FormerLurker, 2020-05-09:
This has been tested pretty completely now.  Going to close!

closed 2020-05-09 by FormerLurker