FormerLurker/Octolapse #416: Camera Test Failed

Vexatuspl, 2019-12-08:
#### Version of Octolapse
 Octolapse Version: 0.4.0rc1.dev2

#### Version of OctoPrint
OctoPrint Version: 1.3.12

#### What should have happened?
Snapshot should have been taken

#### What happened instead?
When I hit "test" button to test the webcam an error pops-up:
Camera Test Failed
Errors were detected - An unknown webcam server type 'IP Webcam Server 0.4' is being used for the 'New Camera' camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences.

That has been working, but I damaged SD card and after new install it doesn't. Webcam is accessible, the original Timelapse does access it and is able to take snapshots. If I try to change profile from Webcam to anything else I get the same error.
Already tried to uninstall and re-install Octolapse.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi 
Os Version: 0.17.0

#### Printer model & used firmware incl. version
Printer Model: Anet A8
Printer Firmware Version: Marlin 1.19

#### Browser and version of browser, operating system running browser
Browser: Chrome
Browser OS: Windows 10

FormerLurker, 2019-12-08:
Do you have custom image preferences enabled?  If so, turn it off in the camera profile and try again.  Let me know how it goes!

Vexatuspl, 2019-12-08:
No, it's disabled. When I try to enable it, the following error pops:

Unable To Enable Custom Preferences
An unknown webcam server type 'IP Webcam Server 0.4' is being used for the 'New Camera' camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences.

I've even tried to change the line mentioned in the help:
Change the following lines:

#camera_http_webroot="./www-octopi"

#camera_http_options="-n"
to this:

camera_http_webroot="./www"

camera_http_options=""

But nothing has changed.

FormerLurker, 2019-12-08:
So I dug into the camera test code and did find a bug.  There is a test to make sure that custom image preferences can be applied if they are enabled, but it was only supposed to run on print start, and only if they are enabled.  I fixed the code, and you can try it out by installing from the devel branch using the plugin manager with the following url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Before you upgrade it would be a good idea to export your octolapse settings (Octolapse Tab->Gear Icon->Export Settings) in case there are migration problems.

Fyi the new devel version is compatible with the version of OctoPrint you have installed, and 1.4 (including Python 3).

Let me know how it goes, and thanks for posting!

Vexatuspl, 2019-12-08:
Yessss, that works! Thank you!

Vexatuspl, 2019-12-08:
Ok, well - there's no error but it ain't taking snapshots now. When I click the "test" button it takes the pic, but while printing - it just moves extruder to the snapshot position and goes back to printing right away.

FormerLurker, 2019-12-08:
Can you switch your Octolapse debug profile to 'debug - prevent extrusion'?  If you don't see that option, click on the 'Debug' link shown below:

[image]

Add a new debug profile by clicking 'Add Profile...'

Then in the 'Import Debug Profile' drop down select 'Debug - Prevent Extrusion' as the 'Profile to Import'.  Then save your changes, exit the settings menu, and select the 'Debug - Prevent Extrusion' profile and run a few layers from a test print and cancel (no plastic will be extruded, and your printer will not warm up'.

Finally, open the OctoPrint settings (wrench/spanner icon), select 'Logs' from the left menu, and download plugin_octolapse.log and send it to me please.

Vexatuspl, 2019-12-08:
Here's the file:
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/3937044/plugin_octolapse.log)
Seems to be exactly the same error as before but at another stage of the process :)

FormerLurker, 2019-12-08:
Hmm, the errors there regarding the server type have a timestamp that is very different than the timestamp of the last print.  I'm not seeing any logging of octolapse.snapshot in your log.  Try editing your debug profile and make sure that octolapse.snapshot has been added with a logging level of 'Debug'.  Then run another print and send me your log file again.

Also, can you export your settings (octolapse tab->gear icon->export settings) and attach them?  Something must be preventing the snapshots from being downloaded.  Also send me your gcode file please.

FYI, my debug environment uses a remote camera, similar to your setup, so it's baffling to me why this isn't working since remote cameras are more tested than local ones!  I'm sure it must be something simple though.  If you can respond quickly, I should be able to figure this out and post a patch that you can install.

FormerLurker, 2019-12-08:
OH, also, I did fix a few things in the gcode trigger (I think they are unrelated, but better safe than sorry) so you might want to upgrade again using this URL:   https://github.com/FormerLurker/Octolapse/archive/devel.zip

Vexatuspl, 2019-12-08:
Ok, so I wanted to update first, but that failed. Here's the log:
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/3937117/octoprint.log)
Then I wanted to export the settings but insted of the Settings.json my browser is trying to save downloadSettings.html, so here's the one that I've exported before I've update to the previous devel:
[Settings.zip](https://github.com/FormerLurker/Octolapse/files/3937121/Settings.zip)
Finally I did one more test print, so here's the log again (I did checked and the snapshot module log level is set to Debug:
[plugin_octolapse (2).log](https://github.com/FormerLurker/Octolapse/files/3937123/plugin_octolapse.2.log)
And the gcode file:
[762x54r.zip](https://github.com/FormerLurker/Octolapse/files/3937127/762x54r.zip)

FormerLurker, 2019-12-08:
Ug.  Can you send the plug-in manager log?  Id like to see why the upgrade failed, but that info is not available in octoprint.log.

Not sure what is going on with the settings export.  I obviously need to do more regression testing with octoprint 1.3.12...

I will start debugging the rest, but i still don't see any snapshot.py logging :( maybe the code isn't reached at all?!

Vexatuspl, 2019-12-08:
Sure, here you go:
[plugin_pluginmanager_console.log](https://github.com/FormerLurker/Octolapse/files/3937143/plugin_pluginmanager_console.log)
It looks strange: moves extruder and then goes back right away. Even the delay is not applied (5 seconds that should be).

FormerLurker, 2019-12-09:
@Vexatuspl, it looks like the reinstall from the devel branch worked! There are no errors in the log.

I was able to confirm that there is an issue exporting settings.  I will fix that and will let you know when that is complete so that you can reinstall.

I will work on your other issues ASAP.

Vexatuspl, 2019-12-09:
Ok, thanks! Please let me know once you have the updated version so that I can send you the export file.

FormerLurker, 2019-12-09:
It has been updated :)  I have been running some tests, and so far so good.

FormerLurker, 2019-12-10:
@Vexatuspl, I think I found your issue and have fixed it.  Try updating to the latest devel branch and re-run your gcode trigger script.  I'm excited to hear how/if it works for you!

Vexatuspl, 2019-12-12:
@FormerLurker, I kinda broke down my printer so couldn't do the test for the last coupl of days. Sorry for that. Installing lastet devel right now and will post results in a couple of hours. Thatnk you!

Vexatuspl, 2019-12-12:
@FormerLurker, ok, so I got exactly the same error as last time:
[image]
And that wasn't visible in the logs that you've seen. Any ideas? I'll wait for your answer as when I click "close" button, the Pi will freeze.

Vexatuspl, 2019-12-12:
Ok, it seems thatit actually isntalled despite of the error as I can export settings now:
[Skull_Extruder_Button.zip](https://github.com/FormerLurker/Octolapse/files/3957499/Skull_Extruder_Button.zip)
[plugin_octolapse 001.log](https://github.com/FormerLurker/Octolapse/files/3957500/plugin_octolapse.001.log)
[Settings 001.zip](https://github.com/FormerLurker/Octolapse/files/3957504/Settings.001.zip)

Now it not even moves the extruder to the stabilize possition not mentioning taking a snapshot.

FormerLurker, 2019-12-12:
Well at least I now see some useful info in your log.  I'll work on this. 

In the meanwhile, can you also re-send me pluginmanager_console.log?  This could be a bug in OctoPrint somehow, or maybe something else.  So odd that it installs correctly but looks like it's failing!

FormerLurker, 2019-12-12:
So, the problem is subtle unfortunately.  Your printer IS priming at 0.28MM, but the priming detection is used to prevent priming that happens ABOVE the first layer.  Since you are priming below your first layer height (which is kind of unusual I think) Octolapse is tricked into thinking that your printer isn't finished priming.

Here is the solution:

1.  Change your priming height so that it is above your max layer height (say 0.5mm)

[image]

Now Octolapse will (hopefully) work, yeah!  

However, and this is relatively minor but will be somewhat annoying.  Octolapse will think your priming line is a separate layer, which is not ideal.  This won't be an issue for the gcode trigger, but will cause some annoying problems when using the layer trigger or the smart triggers (especially snap to print).  You have two options to fix:

1.  Restrict your Snapshot Area so that Octolapse will not take snapshots below a certain height like so:
[image]
[image]

2.  Use a new feature I just added to deal with this issue by adding some gcode to your start/end gcode.  The two new gcodes are '''@Octolapse stop-snapshots''' and '''@Octolapse start-snapshots'''.  They can be used to prevent snapshots from being taken in certain places, namely your start and end gcode.  For example, in your gcode (edited, do not copy and paste exactly!)

```
; Add the following as the VERY FIRST line of your start gcode!
@Octolapse stop-snapshots 
G90
M82
......
G1 X60.0 E9.0 F1200.0
G1 X200.0 E25.0 F1200.0 ; prime the nozzle
M900 K0.4 ; set Linear Advance K-factor
M117 I'm printing...
; This should be the VERY LAST line of your start gcode!
@Octolapse start-snapshots
```

And your end gcode (again, do not just paste this into your start/end gcode):
```
;This should be the VERY FIRST line of your end gcode
@Octolapse stop-snapshots
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
G92 E0
G1 E-1.0 F1200.0 ; retract
G91
G0 Y2.0 F9600
G0 X 200.0 Y200.0 Z50.0 F500.0 ; lift the nozzle
G90
M84 ; disable motors
M107; fan off
M117 I'm done!
```

I'll be creating a guide/video for this soon.  Also, I'm going to try to improve the error and status screen to make this easier to debug before the next release candidate comes out.

Let me know how it goes!

Vexatuspl, 2019-12-14:
Yes, yes, yes! That works! Thank you! I'll stick to the gcode trigger, did changed the priming height to 0,5mm and it's fine. 
My priming line is higher then a layer as I don't change it for every print - it's in my starting script.

Unfortunatelly I'm unable to use octolapse with my current setup. I think it's caused by the fact that I use Pi Zero - CPU as at 100% for most of the time and that impacts print quality dramaticaly. But it works and I need to move to Pi 3B I guess :)

FormerLurker, 2019-12-14:
Excellent!  Fyi, the 'smart' triggers use fewer resources while printing, and I am adding a 'smart gcode' trigger as we speak.  However, it probably isn't good enough for the zero sadly.

Vexatuspl, 2019-12-14:
Yup, I'm looking forward to switch over to 3B or 3B+ and be able to record octolapses - it's a great job man!

FormerLurker, 2019-12-16:
So, I'm going to close this issue since I believe it is resolved.  Let me know if I should reopen it, and Thanks!

closed 2019-12-16 by FormerLurker