FormerLurker/Octolapse #417: With Octolapse on layer change, the extruder is left in absolute mode after snapshot

damienmg, 2019-12-08:
#### If this is a feature request describe it here

N/A.

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Activate Octolapse with trigger on Layer Compatibility change and move of the extruder (here extruder back at left).
2. Do a 2 layer print
3. Observe huge retraction before the second layer, leading to no extrusion.

Upon investigation, it seems like the printer is left with a G90 but without sending the M83. Next extrusion is going to be 0.04 so a huge retraction. From what I understand from the log it sends M84/M83 but according to the terminal it does not.

I don't really know what happened, it used to work well.

I tried to play with the octoprint start gcode without success.

#### What should have happened?
M83 should have been sent after G91 to set the extruder to relative position

#### What happened instead?
No M83 / M84 seen in the terminal, leading to incorrect extrusion move after, cf gcode between the layers from the console:

```
G1 Z0.800 F10800.000
M104 S235
G1 X125.069 Y111.065
G1 Z0.400
G1 E0.80000 F2100.00000
G1 F1011
G1 X125.407 Y112.041 E0.03012
; Octolapse
M400 ; Wait for all move to finish
M114 . Get current position
G1 E-0.80000 F2100.000 ; Retract 0.8
G91 ; Relative position
G1 Z0.600 F10800.000 ; Up 0.6 in Z
G90 ; Absolute position
G1 X2.350 Y211.500 ; Camera position
M400 ; Wait for move to finish
M114 ; Get info - Probably also taking snapshot
G1 X125.407 Y112.041 ; Move back to poistion
M400 ; wait
M114 ; Get position
G91 ; relative pos
G1 Z-0.600 ; go back down
G1 E0.80000 F2100.000 ; extract 0.8

G90 ; Absolute position! Missing M83! now extruder is in absolute position!
G1 F1011.000 ; Restore feedrate
; /Octolapse
G1 X125.439 Y122.640 E0.30905
G1 X125.261 Y123.517 E0.02610
G1 X124.828 Y124.300 E0.02610
G1 X109.561 Y112.347 E0.29922
```

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian (not octopi)
Os Version: 9 (stretch)

#### Printer model & used firmware incl. version

Printer Model: Creality Ender-3 with SKR1.3 / TMC2209 / Marlin
Printer Firmware Version: Marlin 2.0 (config at https://github.com/damienmg/Marlin/tree/dmg-ender3)

#### Browser and version of browser, operating system running browser

Browser: Firefox
Browser OS: Linux Ubuntu 18.04.3

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File:
[cube_0.2mm_PET_.gcode.txt](https://github.com/FormerLurker/Octolapse/files/3936620/cube_0.2mm_PET_.gcode.txt)

#### Link to settings.json

Link to settings.json with all passwords removed: [settings.json.txt](https://github.com/FormerLurker/Octolapse/files/3936622/settings.json.txt)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/3936623/plugin_octolapse.log)

#### Link to octoprint.log

Link to octoprint.log:  [octoprint.log](https://github.com/FormerLurker/Octolapse/files/3936626/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: N/A.

#### Screenshots and/or videos of the problem:

Screenshot/Video Links:  N/A.

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-12-08:
That is strange!  Would you be willing to test the new alpha version?  It's changed so much since V0.3.4, and I very much doubt this error exists in the new version.  I recommend that you download your settings (octoprint settings->Octolapse->download settings) before you upgrade, then you can refer to [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse), making sure that you use the https://github.com/FormerLurker/Octolapse/archive/devel.zip link to install from the plugin manager.

Please make sure you read the guide all the way, because things have changed (especially regarding the new automatic slicer settings detection, which is awesome).  If you have any questions, I will be available for the next few hours.

If you don't wish to upgrade, I will try a regression test as soon as I deploy Octolapse v0.4.0rc3.  This new release has taken me over a year to develop, and I'm hopeful I can push to master (v0.4.0) in the next couple of months.

damienmg, 2019-12-08:
Sure thing, I'll look into it tomorrow!

FormerLurker, 2019-12-08:
Awesome 👍 it is hard to find testers, so many thanks brave soul!  Fyi, I have run hundreds of hours of printing since rc1 dropped, so it is pretty stable, at least in environment.

damienmg, 2019-12-10:
Sorry I did not got time to do that yesterday so I only got to install it now. I unfortunately failed the install of octolapse without a clear error message. However on restart it now logs:

```
2019-12-10 21:10:32,570 - octoprint.plugin.core - ERROR - Error loading plugin octolapse
Traceback (most recent call last):
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 29, in <module>
    import octoprint_octolapse.log as log
AttributeError: 'module' object has no attribute 'log'
```

This sounds like this is due to the partial install. I'll try to install it from the virtualenv directly.

damienmg, 2019-12-10:
Ok pip install did suceeded and now I have the new version :)

damienmg, 2019-12-10:
I found a bug, or maybe it is not one :) The test mode does heat because Octolapse get confused because I have declared I have 2 extruders while the printer only have one (cf. https://medium.com/@damien.martin.guillerez/multi-filament-print-with-a-single-extruder-using-prusaslicer-2e0746348cdd?source=friends_link&sk=541dcda7cd469b502874ed76e266267e).

I get the following error in the logs:

```
2019-12-10 21:30:20,500 - octolapse.__init__ - ERROR - Your printer profile has fewer extruders (1) defined than your gcode file (2).
2019-12-10 21:38:11,367 - octolapse.__init__ - ERROR - Your printer profile has fewer extruders (1) defined than your gcode file (2).
```

I'll try to reslice the thing.

FormerLurker, 2019-12-10:
Hey, just go into Octolapse printer profile and set the number of extruders = 2, and make sure you select that you are using a shared extruder.  That should do it without a reslice.

FormerLurker, 2019-12-10:
Also
> The test mode does heat 

This should never happen UNLESS you have disabled 'Cancel print if startup errors are detected':

[image]

Please let me know, thanks!

damienmg, 2019-12-10:
Ok I'll try that tommorow. I'll also try to create a new profile from
scratch using feature detection.

I got it running using my single extruder profile in PrusaSlicer but I
still got the same issue. I'll continue testing tomorrow anyway.

damienmg, 2019-12-10:
And also I do have this option deactivated, I don't remember why I did
that, this was a long time ago.

damienmg, 2019-12-14:
I did a bunch of other tests (deactivating all other plugins, changing the settings for G90/G91, etc...) no success so far.

I however noticed that contrary to what I said earlier, octolapse does not even try to send M82/M83 despite the configuration being set to:

- X/Y/Z Axis Mode: Require explicit G90/G91
- E Axis Mode: Require explicit M82/M84
- G90/G91 Influences Extruder:  True

As far as I read the doc, this does not change how Octolapse is handling the retraction, but I haven't found any settings for changing that, I'll try to dive a bit into Octolapse code to understand.

FormerLurker, 2019-12-14:
Hold on a sec..  G90/G91 influences extruder MUST be false for your printer (ender 3 right?)!  That could definitely cause this error.  Set that to false.  I will look into the force absolute setting.

damienmg, 2019-12-14:
So I guess I understand what is happening.

From https://github.com/FormerLurker/Octolapse/blob/master/octoprint_octolapse/gcode.py#L524:
```py
        if self.ZhopBySnapshotStartGcode:
            if not self.IsRelativeCurrent:
                new_snapshot_gcode.append(
                    SnapshotGcode.END_GCODE,
                    self.get_gcode_axes_relative()
                )
                self.IsRelativeCurrent = True

            if self.ZHopSpeed != self.FCurrent:
                new_f = self.ZHopSpeed
                self.FCurrent = new_f
            else:
                new_f = None

            new_snapshot_gcode.append(
                SnapshotGcode.END_GCODE,
                self.get_gocde_z_lower_relative(self.ZLift, new_f)
            )

        # detract
        if self.RetractedBySnapshotStartGcode:
            if not self.IsExtruderRelativeCurrent:
                new_snapshot_gcode.append(
                    SnapshotGcode.END_GCODE,
                    self.get_gcode_extruder_relative()
                )
                self.IsExtruderRelativeCurrent = True
```

So  even though G90/G91 influence the extruder, the extruder status is not updated by the z-hop setting relative axis.

I'll bake a PR and send it after testing.

FormerLurker, 2019-12-14:
So G90/G91 DOES affect your extruder??  If so, there may be a big there.  I don't have a printer like that, so maybe I goofed and didn't test it properly via the debugger.  However, be sure you look at [this file from the devel branch](https://github.com/FormerLurker/Octolapse/blob/devel/octoprint_octolapse/gcode.py), because the other gcode file is super old and I probably will just fix this in the next version.  However, looking at the new version of gcode.py, it looks like I got this right:

https://github.com/FormerLurker/Octolapse/blob/477c90bc198b0800cae44a6350a7a881702f2db4/octoprint_octolapse/gcode.py#L485-L505

I will test this just to make sure.

FormerLurker, 2019-12-14:
OMG.  Well, I didn't look at the code hard enough.  I see a mistake here:

https://github.com/FormerLurker/Octolapse/blob/477c90bc198b0800cae44a6350a7a881702f2db4/octoprint_octolapse/gcode.py#L496-L505

Line 500 should be this:

```
self.is_extruder_relative_current = False
```

Whoops, lol!  I managed to mess that up in the original AND in a rewrite..  I will push an update to the devel branch shortly.

damienmg, 2019-12-14:
Oh master is not the develpment branch :) sorry my bad.

As for G90/G91, this is the default behavior in Marlin: http://marlinfw.org/docs/gcode/G090.html

I have a SKR1.3 with Marlin 2 so not the stock Ender-3.

FormerLurker, 2019-12-14:
> As for G90/G91, this is the default behavior in Marlin: http://marlinfw.org/docs/gcode/G090.html

Well, that's good to know!  It's a nightmare dealing with all of these different firmware behaviors.  I've always found that the G90/G91 affects extruder always seemed like a very odd design choice to me.  I can't think of a situation where you would want to change both of them at the same time, but maybe there is such a situation.

FormerLurker, 2019-12-14:
Found another issue here:

https://github.com/FormerLurker/Octolapse/blob/477c90bc198b0800cae44a6350a7a881702f2db4/octoprint_octolapse/gcode.py#L817-L833

This code needs to be added after line 835:
```
            if self.g90_influences_extruder:
                self.is_extruder_relative_current = is_relative_return
```

With that final change I'm getting what looks like good snapshot gcode:
```
	Init           - G1 X124.725 Y104.725 E0.01862
	Start          - G1 E-4.00000 F4800.000
	Start          - G91
	Start          - G1 Z0.600 F10800.000
	Snapshot       - G90
	Snapshot       - G0 X125.000 Y103.500
	Take Snapshot  - @OCTOLAPSE TAKE-SNAPSHOT
	Return         - G0 X124.725 Y104.725
	End            - G91
	End            - G1 Z-0.600
	End            - G1 E4.00000 F3000.000
	End            - G90
	End            - M83
	End            - G1 F1500.000
```

How does that look to you?

damienmg, 2019-12-14:
This looks good, this is what I expected.

Thanks!

FormerLurker, 2019-12-14:
Thank YOU for leading me to this!  I will post here once I push the changes.  Need to do just a bit more testing.

FormerLurker, 2019-12-15:
OK, I pushed an update to the devel branch.  I"m updating my printer and will be running some tests soon.  Let me know if your problems are fixed.

FYI, Logging settings within your debug profiles will be a bit off since you were running a pre-release build, and the migration won't run.  I was in the middle of some refactoring, and changed some file names, and since logging is module based in Octolapse some modules will no longer log without debug profile edits.  It shouldn't be a big deal, but you could restore the default settings if you want to.  At some point soon I will update the profile repository, so you could also just wait for that.

FormerLurker, 2019-12-15:
So, I tested the 'Force Absolute' and 'Force Absolute' printer profile settings, and they seem to be working for me.  Let me know if you still have a problem with those options after updating to the latest devel branch.

damienmg, 2019-12-15:
Thanks a lot for everything,

Unfortunately I cannot seems to install the new dev version. The octolapse pane just load forever and I am unable to install any settings / add any printer. Editing the configuration just not work, with an octoprint error.

My octoprint log is full of error in the like of :
```
2019-12-15 10:40:39,076 - octoprint - ERROR - Exception on /plugin/octolapse/loadState [POST]
Traceback (most recent call last):
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 428, in load_state_request
Exception: Unable to load values from Octolapse.Timelapse, it hasn't been initialized yet.  Please wait a few minutes and try again.  If the problem persists, please check plugin_octolapse.log for exceptions.
```

The root error seems to be a simple parse error (might be due to installation):
2019-12-15 10:15:38,629 - octoprint.plugin - ERROR - Error while calling plugin octolapse
```
Traceback (most recent call last):
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 1690, in on_startup
    raise e
AttributeError: 'dict' object has no attribute 'to_dict'
```

After downgrading to stable there is still an issue with the settings of octolapse. I guess it is time for me to cleanup the install

damienmg, 2019-12-15:
Ok I got back in a working state by removing all my octolapse configs and installing directly the devel version. The upgrade path from the stable version seems to be broken.

damienmg, 2019-12-15:
I can confirm the problem is solved now ! \o/

So the only issue with dev I see is doing the following:

- Install previous version, tweak CR-10 profile
- Upgrade to devel, and get in a mess, with the pane not loading and unable to load errors.

Anyway this is a diferrent issue than this one so I am closing this bug and will recreate one for the upgrade path.

closed 2019-12-15 by damienmg

FormerLurker, 2019-12-15:
So glad this horrible bug was found.  It was easy to fix once identified, and was no doubt causing many people quality issues.  Thank you so much for sticking to it and helping me solve it.  I REALLY appreciate it!!

jessegrunert, 2020-01-08:
Thanks for working this one out folks, I just ran into this error when trying to print a timelapse with a Palette 2 Pro. Have to run the extruder in relative mode and can confirm the exact same problem only present when trying to run in relative mode. I am going to upgrade to the development branch right away.