FormerLurker/Octolapse #419: Upgrade path to devel from 0.3.4 is leaving octolapse in an inconsistent state

damienmg, 2019-12-15:
cf #417 for more details.

To reproduce:

    Install previous version, tweak CR-10 profile
    Upgrade to devel, and get in a mess, with the pane not loading and unable to load errors.

Anyway this is a different issue than this one so I am closing this bug and will recreate one for the upgrade path.

FormerLurker, 2019-12-15:
I am not surprised that you found some migration issues.  I will play with it today and will try to reproduce.  It is likely something in settings.json, so if I have trouble perhaps you could recreate your old settings file and send it?  Octolapse backs up the old settings before upgrading, so maybe you still have the old file in /home/pi/.octoprint/octolapse??

FormerLurker, 2019-12-15:
So, I have been able to recreate the issue.  I will let you know when I think it is solved.  The good thing about this version of Octolapse is that you can import settings, even from previous versions, so it should be much easier to confirm that the migration is fixed.

FormerLurker, 2019-12-15:
I think it is solved.  Let me know if you have problem migrating to the devel branch.  Thanks!

damienmg, 2019-12-19:
Hi sorry for not reporting earlier

Again the zip install did not worked, I got the following stack trace from octoprint.log:
```
2019-12-19 23:12:42,121 - octoprint - ERROR - Exception on /plugin/octolapse/loadState [POST]
Traceback (most recent call last):
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 221, in load_state_request
    "Unable to load values from Octolapse.Timelapse, it hasn't been initialized yet.  Please wait a few "
Exception: Unable to load values from Octolapse.Timelapse, it hasn't been initialized yet.  Please wait a few minutes and try again.  If the problem persists, please check plugin_octolapse.log for exceptions.
```

This lead to some kind of install loops, with the following errors showing up in octoprint.log:
```
2019-12-19 23:17:33,332 - octoprint.plugin.core - ERROR - Error loading plugin octolapse
Traceback (most recent call last):
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 29, in <module>
    import octoprint_octolapse.log as log
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/log.py", line 29, in <module>
    from octoprint_octolapse.utility import Singleton
ImportError: cannot import name Singleton
```

Octoprint did not restored by themselve until I force restarted it. Somehow this actually fixed the install and now Octolapse is installed nicely.

Other test I did was run a few print, and getting good timelapse. So from my point of view, except this update path that is a bit problematic, the devel branch looks pretty stable.

FormerLurker, 2019-12-19:
Thanks for letting me know!  I have no idea what could cause that import issue.  It doesn't look like there are any circular imports or anything.  Perhaps the imports from the current install are messing with the upgrade somehow?

In any case, I moved the 'Singleton' class definition inside of the log.py module, since that is the only module that uses that class.  It doesn't seem like it belongs there, but if you would try upgrading again I'd appreciate it.  If I can verify that this is fixed, I"m going to push 0.4.0rc1.dev3!

Also, where you upgrading from 0.4.0rc1dev2, or something else, like 0.3.4?

damienmg, 2019-12-20:
I was trying the upgrade from 0.3.4 as this is the path most people will
do. I'll retry updating this evening.

damienmg, 2019-12-20:
So first test failed, still have the upgrade issue. I'll try to dig a bit to understand what is happening.

FormerLurker, 2019-12-20:
Can you see if it's still the same log message?  Also, did you install from this url: https://github.com/FormerLurker/Octolapse/archive/devel.zip

If not, the 'fix' (if it is a fix, I still have no idea what is causing that issue) won't be included.

Also, what version of OctoPrint are you running?

FormerLurker, 2019-12-20:
So, I upgraded from 0.3.4 to 0.4.0rc1.dev3 using both OctoPrint 1.3.4 and the dev version of 1.4 and didn't see your issue.  Either it has been fixed (if you installed from the devel branch), you are using a different version of octoprint, or it is something specific to your installation.

Let me know what version of OctoPi (if any) you are using, what hardware/OS you are running on, what version of OctoPrint you are using, and which version of Octolapse (a link would suffice) you are upgrading to.

Thanks!

damienmg, 2019-12-20:
Ok this is really fancy.

When I install I get an error from the installer, but the log for the installation is all good (the installer shown an error before the end of the installation):

```
$ tail -1 logs/plugin_pluginmanager_console.log 
2019-12-20 20:45:26,529 > Successfully installed Octolapse-0.4.0rc1.dev3
```

However the install did not worked, I get some weird stack trace:

```
2019-12-20 20:45:29,449 - octoprint.plugin.core - ERROR - Error loading plugin octolapse
Traceback (most recent call last):
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 56, in <module>
    import octoprint_octolapse.stabilization_preprocessing
  File "/home/octoprint/venv/local/lib/python2.7/site-packages/octoprint_octolapse/stabilization_preprocessing.py", line 27, in <module>
    from octoprint_octolapse.settings import PrinterProfile, TriggerProfile, StabilizationProfile
ImportError: cannot import name PrinterProfile
```

That's indeed point out to a problem with my own install of octoprint (maybe something is corrupted in the venv).

Another error point to the absence of `/home/octoprint/venv/lib/python2.7/site-packages/octoprint_octolapse/static/js/octolapse.profiles.snapshot.js` which indeed does not exists. Which is consistent with the install not being done correctly.

I am using Octoprint version 1.3.12 (latest stable) by the way.

My guess is that the install is too slow for some reason. I definitely not see anything that would point out to the plugin itself. Maybe a bug in Octoprint itself?

damienmg, 2019-12-20:
Another test, same result but waiting for the full install to be well done and restarting octoprint lead to a perfect install.

I am closing this bug as this is definitely not an Octolapse issue.

Thanks a lot for your work on Octolapse, if you ever need more testing for future version, feel free to ping me!

closed 2019-12-20 by damienmg