FormerLurker/Octolapse #420: Snaps shot with DSLR but It pauses...

homescienceideas, 2019-12-19:
So hey bro, I made a video to show you what this is doing. It’s in test mode and it runs through but when it moves to the right side it’s naps a shot. I don’t know where the shot goes though. Before it goes to the right that I set it to go to It’s almost like it freezes up and I can hear a motor buzzing then it moves to the specified location and snaps a shot, then it moves back and does another long pause when it resumes the print. 

Before when I was using my DSLR I had batteries in the camera but now I have a power supply. This also pops up right when it snaps a shot and it doesn’t specify what the problem is.  

[image]

Video here: https://youtu.be/SX-A3fn__PA 

#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-12-19:
First, LOVE the video!  That is without a doubt the best issue report video I have seen.  

So, i will probably need some more info to figure this out.  Is the buzzing coming from your extruder, or the XYZ axes?  Also, if you could post a copy of plugin_octolapse.log, that would help.

Fyi, i just added much better DSLR logging and testing buttons if you are into beta testing.  Let me know.

Thanks for posting!  I will be able to give this more attention in the AM.

homescienceideas, 2019-12-19:
I would definitely be interested in the beta testing. Right now though I’m trying to get some videos out for some clients of mine. I’m sponsored by different 3D printing companies and so I’d like to get a couple videos out to keep them at bay. 

Not sure where the buzzing is coming from so I will have to get back to you on that as soon as I can.

homescienceideas, 2019-12-19:
I hope I did this right... [https://gist.github.com/homescienceideas/021a5986e9bdd9d58d5120b9ab3816cd] See here!(url)

FormerLurker, 2019-12-19:
In the meantime, see of you can grab a log for me.  I will take a look and see what i can find.  Maybe z hop speed is ultra slow.  You could also disable z hop and see what that does.  Download and post your settings.json too, and I will review the slicer settings.  If you are using the automatic settings extraction, could be a bug in that too.

FormerLurker, 2019-12-19:
Our posts crossed like ships in the night, lol.  So, for whatever reason that log is pretty bare.  I can figure out why that is with your settings.json file (export from octoprint settings->octolapse).  About to turn in for the evening, but will look first thing in the AM.  Also post your gcode so that i can run a test through the debugger using your settings.  Thanks!!

homescienceideas, 2019-12-19:
I disabled Z hop so that's not it. Here is my .json (https://gist.github.com/homescienceideas/011ce401a60dbd47517a373fdf2c7932)

homescienceideas, 2019-12-19:
Here is my gcode... (https://gist.github.com/homescienceideas/9c0225a38c109cca64f33e81603f0d45)

FormerLurker, 2019-12-19:
Try turning 'show real snapshot time' off in the main settings.  Could be that your printer doesn't appreciate many m400s over a shortish  time period.  I removed that feature from the next version cause it can cause problems on some printers.

FormerLurker, 2019-12-19:
Also, not sure why logging is so messed up..  your settings look correct there.

homescienceideas, 2019-12-19:
I turned 'show real snapshot time' off and still the same issue. I then canceled the print and it will not generate the "plugin_octolapse.log" file.

homescienceideas, 2019-12-19:
I tried a smaller gcode and let it run and still no plugin_octolapse.log file. It looks like it disconnected from my printer and spit out a octoprint.log and this Error. https://gist.github.com/homescienceideas/56d85e790bdb46c91f088623203c182f

FormerLurker, 2019-12-20:
Sorry it took so long to look at this, but work has been very busy (it usually is at this time of the year).  OK, so Octoprint isn't reporting any error, and I don't have any issues running the gcode :(  Somehow this must be printer related.  I don't mean to say it isn't a bug in Octolapse, by no means.  In fact, I encountered a similar problem a while back, though I can't seem to locate the ticket at the moment.  It is possible this has been fixed already.  Since for whatever reason logging seems to not be working, I think our best option is to try the devel version.  It can be installed by using this URL in the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

I recommend restoring the default settings after the install (I plan to add a popup after install for anyone upgrading), but if you could run a 'no extrusion' test with logging enabled first, that would be great.  Be sure to edit your debug profile, and import the 'Debug - Prevent Extrusion' profile first.  To do this, just edit your debug profile and select 'Debug - Prevent Extrusion'  as the 'Profile to Import' under the 'Import Debug Profile' heading.

Also, fyi, I just pushed some additional logging and error message enhancements for script cameras to the devel branch and fixed an upgrade issue (package import problem).

I can make myself available on discord tomorrow is you need some more hands on help.  Saturday I'm leaving for Christams, so tomorrow will be the last day I'm available for the next week..

homescienceideas, 2019-12-20:
I have no idea how to do a no extrusion whatever you were saying up there. Can you send me some detailed instructions and I’m hoping it’s soon because I got to get this done:)

FormerLurker, 2019-12-20:
Sorry about your deadline...

Test mode is what i mean.  Pretty much the same way you did it when using version 0.3.4.  I renamed it because people didn't know what test mode was supposed to be.

homescienceideas, 2019-12-20:
So, install the devel version then do no extrusion thing?

FormerLurker, 2019-12-20:
Yeah, sorry, you keep hitting me at bedtime :)  I snuck up real quick to get you these instructions, but then I'm out for the evening.

1.  Install from this link:  https://github.com/FormerLurker/Octolapse/archive/devel.zip
2.  Reboot.
3.  Click on the 'debug' link next to the drop down box:
[image]
4 Click 'Add profile' and import the 'Debug - Prevent Extrudion' profile.
5. In that same profile, click 'Clear Log' to empty the log for a new test (makes it more readable):
[image]

6. Click Save and select the new debug profile:
[image]
7.  Run a test print, and stop after a couple layers.   If you see a problem, download the log by editing your debug profile and then clicking here here (I added a shortcut):
[image]
8.  If things start working you will want to run a live print, just edit your debug profle and inport the 'Log Errors' profile:
[image]

You might want to take a look at [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) if you haven't already.  It explains how to set up some of the features in the new version, like automatic settings detection.  Pay close attention to the cura section if you want to use that (and you will).

Good luck!

homescienceideas, 2019-12-20:
I cannot even get Octolapse to load now. 
[image]

FormerLurker, 2019-12-20:
Darnit...  Ok, something died in the settings migration process.  Easy to fix.  Click the gear icon in the octolapse tab and restore the default settings.  

Next go back to the octolapse tab and add a printer profile.  Your printer may be in the make/model list (fingers crossed).  Then edit your camera profile, switch it to a script camera in the drop down and enter the path to your snapshot script and save.

Look at that guide i sent for getting automatic slicer settings working for cura.  It is pretty fast and easy.

That should do it.

FormerLurker, 2019-12-20:
I will try to migrate your settings from 0.3.4 and will figure out what broke...  Sorrt about that.

homescienceideas, 2019-12-20:
The settings you are saying to check is not there. There is no "Click the gear icon" on the octolapse tab. It just says Loading Octolapse.

FormerLurker, 2019-12-20:
Ok, navigate there the long way.  Go into octoprint settings and find octolapse in the left menu.  You can restore the settings there too.

FormerLurker, 2019-12-20:
So, I did two things this AM for you:

1.  I debugged your settings.json file and figured out why it wasn't working.  I added lots of safety checks to make sure this doesn't happen again.  I was able to successfully migrate your settings.  [Here is a file](https://gist.github.com/FormerLurker/1bbb58327478c4776ec74f54784cd85a) you can use to import your previous settings (OctoPrint settings->Octolapse->Import Settings) if you want to for some reason.

2.  I noticed that you are using an Ender 5, and I had no profile for this in the profile repository.  I have added the Ender 5 and Ender 5 Pro, so if you want you can import that profile (Printer Profile-> select make (creality3d) and model (ender 5 or 5 pro).  I have never printed with those profiles, so keep that in mind.  They appear to match your profile except I added build volume details directly to the profile, instead of using the OctoPrint settings.

I will be here during the day, so if you need help let me know.

FormerLurker, 2019-12-20:
Oh, also, I pushed those changes (settings migration fix) to the devel branch, by the way, if you want to reinstall.  I'll be testing fixing any bugs I find all day.

homescienceideas, 2019-12-20:
I’m not sure how to get the Ender 5 profiles to show up? I did a video so you can see what I’m saying and I hope it helps my friend. I’m assuming I need to re-upload from the link that you gave above to update my Octolapse? https://youtu.be/Hc-C4n5kYQ8

FormerLurker, 2019-12-20:
Ok, so Octolapse now fetches new profiles and checks for updates on a periodic basis (configurable within the main settings tab in the Octolapse settings page you showed in your video).  I definitely need to add a way to force an update check after install or upgrade.  Octolapse did not check in your case, because it already had a cached copy, and the default update period is 7 days.

To force a check for new profiles click this button:

[image]

Let me know if that works!

homescienceideas, 2019-12-20:
In the meantime I tried to run it under test only and I got this pop up. I just wanted you to see how it’s working. I will do as you said above. 
[image]

homescienceideas, 2019-12-20:
OK, I can see the ender 5 and pro profiles. I put it on your Ender 5 profile and I get this popup. 
[image]
[image]

homescienceideas, 2019-12-20:
I’m assuming I should follow this for Cura? 
[image]

FormerLurker, 2019-12-20:
Ok, see if that help button can assist you (i spent a ton of time adding those and would greatly appreciate any feedback).  If you look at that guide i linked, it explains how to get cura running so you dont need to ever copy your slicer settings in, which is nice.

In the meantime i will see why you were missing settings in your profile after upgrading....  Settings migrations are SUCH a pain, and octolapse has lots of (too many) settings...

FormerLurker, 2019-12-20:
Yes!  Automatic slicer settings work for Slic3r and Simplify, but cura requires a script since it does not output slicer settings in the gcode by default :(

homescienceideas, 2019-12-20:
So I clicked on the help button and it took me to this information page. Honestly it’s very helpful and I read through the whole thing. I’m using Cura Took me to the automatic enabling guide and so far the guide is helping. 

Do I have to add the script at the beginning of my start G code or can I add it at the bottom? I just don’t want that to be the first thing I see in the start g code. I like seeing my G codes that way when I want to I can just edit them real quick. 

I would also suggest that maybe you have people make a separate profile specifically for their Octolapse settings in Cura. See image below for what I’m talking about.
[image]

[image]

FormerLurker, 2019-12-20:
Yes, not a bad suggestion about creating an octolapse specific profile.

Regarding the script location, it can be anywhere in your start OR end gcode.  They are just comments, so it doesn't really matter where you put them at all.  Putting it in the end gcode will cause the the settings extraction to be a bit slower (it searches the first 1000 and last 1000 lines and quits when it finds all the settings Octolapse needs), but it shouldn't be too bad.  Try both if you have time and see if there is any difference.  I've got some plans to speed the extraction up (takes 5 sec max in my experience).

FormerLurker, 2019-12-20:
OH wow, I also noticed looking at your image that some of the info there is now out-of-date!  I added beta support for multi-extruder/multi-material+shared nozzle printers.  It is DEFINITELY beta, but I've tested it quite extensively on my MMU2.  However, I do NOT own a multi-extruder printer, so I'm unable to fully test those.  I have a feeling extruder offsets won't work 100%, though I have simulated this, and the results match my (very limited) understanding of how multi-extruder printers should work.

homescienceideas, 2019-12-20:
> Regarding the script location, it can be anywhere in your start OR end gcode. They are just comments, so it doesn't really matter where you put them at all. Putting it in the end gcode will cause [...]

So I’m going to put the script under the G code in the start G code:) Thanks for that information it gives me access to my G code and that script could kind of hide in the background. 
[image]

homescienceideas, 2019-12-20:
So I went to run a test print and this popped up. I’m just updating you as to the process:) 
[image]

FormerLurker, 2019-12-20:
Sweet!  I'd love to hear all about what you think about these new features.  I have no idea what is confusing/appreciated/helpful/unhelpful, etc, since I'm buried in this so much.  Thanks for giving me a play-by-play, since a fresh perspective is EXTREMELY helpful and helps make Octolapse better.

homescienceideas, 2019-12-20:
OK Brad, so I let the print go all the way through and it had no errors. The only thing is it seems to be taking a snapshot mid layer. Basically it does the shell, snaps a shot, then does the infill. It can make for a really interesting looking unique Octolapse but I need to understand how and why it did this. Check out the video. https://youtu.be/oUcvKgaUHOI

FormerLurker, 2019-12-20:
Yes, that's on of the new features of the 'smart' layer trigger, and one of my favorite new additions to Octolapse v4.0.

The smart layer trigger will attempt to take a snapshot as close to the stabilization point as possible, while avoiding things like exterior perimeters, bridges, or other areas that could cause quality impacts.  It reduces the time added by the stabilization by reducing travel and reduces print quality issues caused by the stabilization time.

You might want to also check out the 'snap to print' stabilization, which can dramatically reduce the time/quality impact.  It prevents the extruder from leaving the print at all, and produces what I'm calling a 'quasi-stabilized' timelapse.  However, I don't recommend you use that setting while using your DSLR, since the added time can actually INCREASE some quality problems when using snap to print (though there is a high quality version of this trigger that can mitigate this issue somewhat with the trade off of less stable timelapses).

There are several variants of the smart layer trigger, and I suggest you look at the settings and the help links to read about what is available.

Also, you posted a pic of the popup that appears after starting your print when using the smart trigger.  Did you try running the preview animation (the play button)?  Also, did you notice the 'Total' and 'Saved' travel distance?  That shows you how much travel was saved by using the smart layer trigger vs the old version (now call the 'Classic Layer Trigger').  It is just an estimate, and sometimes the smart trigger can increase the travel distance, since it is also factoring in quality concerns when choosing when to take a snapshot.

I still need to create some guides/videos explaining all of this stuff.  I've been racing to get this version deployed since v0.3.4 is NOT compatible with OctoPrint v1.4, so some things have taken a back seat :(

homescienceideas, 2019-12-20:
Yes, I will read up on everything and also that play button finally works! I just finished a video where I actually printed the small part so I’m going to have a look and see how it went. Do you want me to post a video so you can have a look?

FormerLurker, 2019-12-20:
YES please :)  On your next print, you might want to try the snapshot preview animation.  It's in that 'Snapshot Plan Preview' popup that appears after you start a print when using the 'Smart Layer' trigger.  Let me know if I can add anything to that pop-up that makes the purpose and capability more clear.  My intention there was to let users see where the extruder is going to be before, during, and after a snapshot takes place.  This is really important when using the 'Snap To Print' feature, which doesn't perfectly stabilize the timelapse.  Also, think about trying snap-to-print when using vase mode.  Disabling stabilization when using 'Snap To Print' has some neat effects, especially in vase mode, and you can preview the results in the 'Snapshot Plan Preview' popup :)

FormerLurker, 2019-12-20:
>  that play button finally works

Yeah, that thing was a pain in the neck.  It works by caching images locally, so it was tricky to implement.  I also improved it (it only worked in some situations before) by having it do a really fast 'rewind' before playing the most recent frames, which results in a smooth preview.

Also, check out the drop down box under the preview.  You can switch which camera you are previewing there.  One negative is that you lose the animation from the other cameras when you switch.  I may eliminate that problem later, but it will come at the cost of using more memory/bandwidth.

homescienceideas, 2019-12-20:
Is the Play Button supposed to work after the print is rendered? I know it works while it is running but it would be cool if it showed the preview once the Octolapse is done:) Here is the best Octolapse I have got yet! I will be working on the focus:) https://www.youtube.com/watch?v=jeCQvYCZrNU

FormerLurker, 2019-12-20:
Yes, it will work until you switch tabs, or switch cameras.  It only stores tue last N frames (where N is configurable within the main settings, and defaults to 30 i believe).  I may change the behavior so that the last N frames are always stored, but will need to test memory usage and browser performance first.

FormerLurker, 2019-12-20:
Also, nice video!! How is the part quality?  Hopefully it is better than you got from 0.3.4, which was my entire motivation for the smart triggers, which took nearly a year to develop.

homescienceideas, 2019-12-20:
> YES please :) On your next print, you might want to try the snapshot preview animation. It's in that 'Snapshot Plan Preview' popup that appears after you start a print when using the 'Smart Layer' [...]

I’m not ignoring this by the way. I will get to it it’s just I am on my honey do list now. My wife bought a new mattress for us so I need to get that done. I’m going to get to this.

FormerLurker, 2019-12-20:
OOOh.. a new mattress.  We need one of those too.  Been flipping it around every month or two, and it's becoming a futile exercise.  Good luck with your prints and timelapses, and keep me posted!

I'm marking this issue as resolved, but will leave it open for updates.

homescienceideas, 2019-12-21:
> YES please :) On your next print, you might want to try the snapshot preview animation. It's in that 'Snapshot Plan Preview' popup that appears after you start a print when using the 'Smart Layer' [...]

If I where you I would tell the user just that on the popup:) Something like "This feature lets you see where the extruder is going to be before, during, and after a snapshot takes place." Then add something like "Additionally, you can enable or disable the snapshot plan preview and configure auto-close within the "Main Settings" (<--Link). There is also more info on both of these features there that you can read on."

FormerLurker, 2019-12-21:
I will try just that!

homescienceideas, 2019-12-24:
Hey my friend! Well I ran a longer Octolapse and it never finished the timelapse. At least it never showed up in the timelapse configuration area. It also said that there was supposed to be more snapshots than it was actually taken. I have no idea where these photos are🤷🏻‍♂️
[image]

homescienceideas, 2019-12-24:
I just found this and it looks like it’s on rendered. I will wait for a while to see if the 267 frames start to go up.
[image]

homescienceideas, 2019-12-24:
Not sure why that came out upside down. Well, it rendered out of the blue and I am not sure what happened. The video came out bad because I suck at using my camera!

homescienceideas, 2019-12-24:
So, I think there is a new issue bro. After I did this video I tried to print again with only 1 camera selected. Snaps like 3 shots per layer. https://youtu.be/1fL2PYSQ9T0

FormerLurker, 2019-12-24:
So, as for the first issue (missing then found time lapse).  Probably a UI indicator bug.  It was likely rendering, buy didn't show it was rendering.  If that happens again, refresh (F5), and see if the rendering icon shows up .

As for the new issue, is this a vase mode print?

FormerLurker, 2019-12-24:
Send me your gcode and settings file for the print in your most recent video please.

FormerLurker, 2020-02-15:
Hey, I wanted to let you know that some major enhancements to the rendering routines and UI features were just added to the devel branch.  You might want to try a reinstall from the devel branch: 
 https://github.com/FormerLurker/Octolapse/archive/devel.zip

I'm not sure what could have caused 3 snapshots per layer, but you might want to delete your camera profiles and start over.  Your issue could have been a regression caused by an older version of the default settings.

Anyway, sorry for the long delay in responding.

FormerLurker, 2020-03-26:
Since it's been about a month, I'm going to go ahead and close this issue.  Let me know if you would like the issue reopened.  I'm happy to do so!

Thank you for your post!

closed 2020-03-26 by FormerLurker