FormerLurker/Octolapse #421: Timelapse file missing

sterniasty, 2019-12-19:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse 
 
Octolapse Version:  0.3.4

#### Version of OctoPrint

OctoPrint Version:  1.3.12 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

The is no error at all.

#### What should have happened?
File with time-lapse should be visible in Time-lapse tab.

#### What happened instead?
Everything look fine, shots are being done, when print is done where is an information about rendering time-lapse as usual. But since few days there is no output of this rendering.
I checked files on the SD card and its exactly the same amount of files as visible on time-lapse tab. There is a plenty of free space on a card, so this is not the issue.

#### Operating System running OctoPrint and Octolapse
MacOS Catalina
Os Version: 10.15.1

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3S
Printer Firmware Version: 3.8.1-2869

#### Browser and version of browser, operating system running browser

Browser: Safari

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://we.tl/t-CLIzV76iDM  but happens with different gcodes

#### Link to settings.json

Link to settings.json with all passwords removed: https://we.tl/t-oC6Wy9YqbL 

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://we.tl/t-MbuIps6QBB 

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/sterniasty/907aee083c08f2400923ad06f6601b1c

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://we.tl/t-vFULs9fLQM 

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-12-20:
Wow, that is a massive logfile!  It will take me some time to look through it.

If you are willing, maybe you could try this on the latest devel branch?  Octolapse 0.4 is quite a bit different than v0.3.4, so there will be a bit of a learning curve.  Lots of things have been simplified, many bugs fixed (maybe yours???), many features added, etc...  Anyway, if you're interested I'll send you a link to install and some instructions for use.

FormerLurker, 2019-12-20:
I saw this in the log:

2019-12-18 06:19:30,739 Running ffmpeg with command string: /usr/bin/ffmpeg -framerate 120.0 -loglevel error -i "/tmp/octolapse_renderlOPufj/MK3_PLA_char%06d.jpg" -threads 3 -r 25 -y -b 64M -vcodec mpeg4 -vf '[f0] format=yuv420p [out]' "/home/pi/.octoprint/data/octolapse/timelapse/MK3_PLA_char_20191218054712.mp4"
2019-12-18 06:56:31,553 return-code: Could not render movie, got return code -9: 

You are rendering at 64Mbit, so I suspect that has something to do with it.  Might try backing off of that a bit and see if that helps.

sterniasty, 2019-12-20:
Thank you for sucha  quick response!
I'll start with update to 0.4, since I've been rendering 64M for weeks and with much longer prints, with no issue. If that  would not help, I'll change the quality. If you have a manual how to do the update, I would appreciate it.

sterniasty, 2019-12-23:
update helped and time lapses are now visible in the tab, thank you for your support.
Now I have a new problem with the quality but it's probably subject for another ticket - first I will try different settings :)

closed 2020-01-10 by sterniasty