FormerLurker/Octolapse #422: Nikon D3100 can't capture image

Kesiunas, 2019-12-27:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Hello,

I can not capture image with Nikon D3100. I get this error:
gphoto2 --capture-image

*** Error ***
PTP Operation Not Supported

*** Error ***
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture.

Using octoprint with raspberry pi4.
Done everything by FormerLurker tutorial.

Output of gphoto2 --capture-image --stdout --debug is at https://pastebin.com/qpg0W965

Can't find a solution to the problem. Maybe someone could help me out? Or there is no chance getting D3100 working with gphoto2?

FormerLurker, 2019-12-27:
Hmm.  I looked in the remote controllable camera section at the gphoto2 website, and this camera is listed.  I saw that you posted on the gphoto2 github, which was a good idea, as I'm not an expert, especially with specific makes/models of camera.

Did you try the --capture-image-and-download or --trigger-capture flags and see if those work?  I'm not sure how much I can help since I don't have a camera like yours to test :(

Kesiunas, 2019-12-27:
Yes, I've tried those with no success.

FormerLurker, 2020-01-06:
So, can you try running the following command from a terminal:

```
gphoto2 -v
```
That should list the version of ghpoto2 and libgphoto2.  Perhaps there is a more recent version available?  I have modified an existing update script that can update gphoto2 to the latest build or latest dev build on a pi.  Here is a [link to that script](https://gist.github.com/FormerLurker/d379cf5bd520b63fba26bd2f9846c7a5), along with attribution

Kesiunas, 2020-01-07:
I have updated to latest already

`This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking`

After some fiddling around I think I figured it out one step closer.
The thing is with DSLR's (I'm totally new to using DSLR cameras) that sometimes it doesn't focus (if object is too dark, autofocus can not get the object in focus with that kind of lens or any other problem that I don't know), when you get it to focus and take picture normally (without connecting it to USB), then it works with SSH commands. So I got my D3100 to take picutures and save them but it still gives error and after that it makes photos and saves them. Because of that error, octolapse doesn't count those photos as "taken snapshots" and shows errors without any explanation.

`gphoto2 --capture-image

*** Error ***
PTP Operation Not Supported
New file is in location /store_00010001/DCIM/220D3100/DSC_0008.JPG on the camera`

` gphoto2 --capture-image-and-download

*** Error ***
PTP Operation Not Supported
New file is in location /store_00010001/DCIM/220D3100/DSC_0007.JPG on the camera
Saving file as DSC_0007.JPG
Deleting file /store_00010001/DCIM/220D3100/DSC_0007.JPG on the camera`

It still shows PTP operation not supported.
With Nikon D7000 no problems at all. Both of these cameras after auto detect are shown as (PTP mode), but D3100 gives PTP error and D7000 does not.

[image]

FormerLurker, 2020-02-15:
Sorry it's been so long since I last replied.  Have you made any progress here?  If not, you may want to add this to your gphoto2 arguments:  --auto-detect

I had another person with another camera where that command was necessary.  Might be worth a shot?

FormerLurker, 2020-03-26:
It's been over a month, so I'm going to assume that you have this figured out.  If you want me to re-open the issue, please let me know and I'll be happy to do so.

Thanks for posting!

Kesiunas, 2020-03-26:
Hello,
Sorry that I forgot to reply. Last timelapses that I've done was with D7000. D3100 with gphoto2 is not solved yet. I haven't spent more time on this.
Maybe someone will have the same problem and will be able to do more research on this.

cheers

edm00se, 2020-03-26:
I've also been having this issue w/ D3100. My last attempts didn't include tacking on `--auto-detect`, which I'll try out in the hopefully near future; likely this weekend.

FormerLurker, 2020-03-26:
Ok, will reopen soon.  Thank you both for responding!!!

edm00se, 2020-03-30:
Since my usual octolapse setup (and octopi) are currently busy, I installed gphoto2 on my MacBook Pro (version 2.5.23, via homebrew). Connecting the same camera I was able to observe success adding the `--auto-detect` flag to the `--capture-image` command.

The addition of `--auto-detect` to the `--capture-image` appears to have taken care of some issue, the image successfully saves to the camera, and it no longer throws the:

> *** Error ***              
> PTP General Error
> ERROR: Could not capture image.
> ERROR: Could not capture.

jsnguyen, 2020-04-11:
Using `--auto-detect` did nothing for me. However, I believe I found a workaround as noted here:

https://github.com/gphoto/libgphoto2/issues/140#issuecomment-612299629

This does however, require building and installing from source.

FormerLurker, 2020-04-11:
You could try [this modified upgrade script](https://gist.github.com/FormerLurker/d379cf5bd520b63fba26bd2f9846c7a5) I adapted for Octopi to update gphoto2.  It also handles dev installs.  The gist i linked to includes the original authors notes.

FormerLurker, 2020-05-09:
Hey, I see that there have been changes to gphoto2 for this.  I'm going to assume this is closed, but let me know if you want the ticket reopened, and if the changes to gphoto2 helped!

closed 2020-05-09 by FormerLurker

edm00se, 2021-01-23:
Coming back to this on a new pi, new octoprint instance, but same camera. Apparently picking up where I left off I had to update to the latest development release of gphoto2 (2.5.26.1 as of today) and now `--trigger-capture` works as expected, from my pi, with octolapse.

FormerLurker, 2021-01-23:
@edm00se, thanks for getting back to me!  So glad you got your issue fixed!