FormerLurker/Octolapse #423: Issue with expecting ending with fi

iskyguy, 2019-12-29:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. I followed the quide on setting up an external camera
2. I was doing a test pint with logging on
3. each time it would take a picture it would give me this error `snapshot_script_error: The snapshot script failed with the following error message: ERROR: Could not get image.
/home/pi/scripts/take-snapshot.sh: 24: /home/pi/scripts/take-snapshot.sh: Syntax error: end of file unexpected (expecting "fi")`

#### What should have happened?
Image should have been taken and stored untill the print was done 

#### What happened instead?
Error message occurs and doesn't save file when print is canceled error message occurs saying that the rendering failed 

#### Operating System running OctoPrint and Octolapse

OS Name: octopi ,
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Original Prusa I3 MK3s
Printer Firmware Version: 3.8.1-2869

#### Browser and version of browser, operating system running browser

Browser: latest version of chrome 
Browser OS: chrome 

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://we.tl/t-kHLll3LgFG

#### Link to settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://we.tl/t-PzoTBngkwe

#### Link to octoprint.log

Link to octoprint.log: https://we.tl/t-xk5y4XACQa  

#### Link to contents of Javascript console in the browser

#### Screenshots and/or videos of the problem:

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2019-12-29:
Your script file is probably missing the last line, or has extra lines.  Double check that and post it if you can't find the problem.

iskyguy, 2019-12-29:
I've double checked everything and it looks okay. I followed the video step by step. I'm new so this is an obvious mistake on my part, but I still can't find the problem.

FormerLurker, 2020-01-06:
So, can you maybe try pasting in your entire script?  I will take a look just in case you overlooked something.

FormerLurker, 2020-02-15:
Hey, just checking back in.  Are you willing to copy your snapshot script into a file and paste it somewhere (preferably gist.github.com) so that I can take a look?  I'm going to mark this thread 'waiting for response', and will give you some time to respond.  Thanks!

iskyguy, 2020-02-23:
i figured out the ending with fi issue, but am now having an issue with the video not playing after render. i opened a new case, then saw that you had replied. Sorry for my huge delay.

FormerLurker, 2020-02-26:
Not a problem, and thanks for getting back with me!  I'm very happy you got the script figured out.  Looking forward to working on your other issues, and thanks for being a regular contributor here!

Closed!

closed 2020-02-26 by FormerLurker