FormerLurker/Octolapse #424: The Spaghetti Detective broke Custom Image Prefs

dereklowrey, 2020-01-03:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Updated TSD to 1.0.8 
2. Attempted to apply Custom Image Preferences
3. The Logs attached represent an attempted print job, but the error is the same.
4. Disabling TSD fixes issue.

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
Error:
Unable To Enable Custom Preferences
unknown-server-type: An unknown webcam server type 'Werkzeug/0.15.4 Python/2.7.16' is being used for the '920c' camera profile. Currently only MJPEGStreamer is supported. Unable to apply custom image preferences.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/dereklowrey/281099709dd5430e2372549e073ce9ca

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/dereklowrey/281099709dd5430e2372549e073ce9ca

#### Link to octoprint.log

Link to octoprint.log:  
https://gist.github.com/dereklowrey/281099709dd5430e2372549e073ce9ca

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-01-06:
So, this is a known issue for both Octoprint Anywhere and The Spaghetti Detective.  I rely on mjpgstreamer to control the settings, so if a custom streaming server is used, as is for both of those plugins, custom image preferences won't work.

I have been thinking of a way to control the camera settings by running bash commands, but am not very close to implementing that since I've been really swamped getting things ready for python 3 and the new version of OctoPrint coming out (V0.3.4 is not compatible with it unfortunately, so there is a bit of a time crunch).  I will create an issue for this so I don't forget.

Let me know if there's anything else I can help you with!

dereklowrey, 2020-01-06:
Makes perfect sense. Appreciate the response and love the app. Thanks.

closed 2020-01-06 by dereklowrey

FormerLurker, 2020-01-06:
No problem!  Thanks for posting!