FormerLurker/Octolapse #426: Find Alternative Camera Image Control Mechanism

FormerLurker, 2020-01-06:
Custom image preferences are great, but rely on mjpgstreamer's control.htm page to receive camera control commands.  Find an alternative method for both detecting the camera's control ability and for updating each control that will work more generally for cameras directly attached via USB to the host (hopefully for Linux, Windows and Mac).

Would solve #424.

RaymondHimle, 2020-02-04:
@FormerLurker Some input of the referenced issue would be very helpful. We could use some help figuring out a good way to allow users to use both OctoLapse and TSD's 25FPS streaming

FormerLurker, 2020-02-04:
I have posted a comment.  Once v0.4 is released I'd be interested in cooperating on this.  I think implementing H.264 in TSD is a great feature, and I'd love for these two plugins to play nicely with each other!

bjhiltbrand, 2020-06-29:
Is there an update on this? I just re-imaged my Raspberry Pi, and as such, installed the latest Octolapse and Spaghetti Detective plugins. However, Octolapse fails to load settings or work at all as long as The Spaghetti Detective is active. Now that v0.4 is released, has there been any progress made on making the 2 plugins play nice together? Thanks.

FormerLurker, 2020-06-29:
Yes, TSD has included a compatibility mode that should work.  Check your TSD settings and give it a go.  Let me know how it goes!

bjhiltbrand, 2020-06-29:
Wow! That was a fast response! Thank you for that. I re-enabled the TSD plugin took a look at the settings. I think I found the setting that you mentioned. It was set to the default option of "Automatic". I changed it to always in compatibility mode and restarted, but still seem to have issues with Octolapse being unable to load when TSD is active.

[image]
[image]

RaymondHimle, 2020-06-29:
Make sure you completely reboot your raspberry pi after changing to compatibly mode

FormerLurker, 2020-06-29:
That is an odd error message for this situation, and i am a bit baffled by it.  As the other poster said, do a full reboot first.  If you still see the same error, please post your plugin_octolapse.log file.

RaymondHimle, 2020-06-29:
Also feel free to ask for help on the TSD Discord as it should be working on compatibly mode.

bjhiltbrand, 2020-06-29:
I did a reboot, and full shutdown/power off. The issue still remains.

Here is my log file. Doesn't look like there's anything out of the ordinary.
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/4848264/plugin_octolapse.log)

bjhiltbrand, 2020-06-29:
@RaymondHimle, I'm just catching up on the discord group, as I haven't been in there for a while. I'll see if others have been in this situation.

FormerLurker, 2020-06-29:
Fyi, i am in the process of updating the plugin documents for TSD and Octoprint Anywhere to reflect the changes.  Things have been a bit crazy lately, and I have been delayed over and over.  I WILL get to it ASAP.

FormerLurker, 2020-06-29:
Check the browser console for errors please.  F12 usually opens the console.

bjhiltbrand, 2020-06-29:
[image]

FormerLurker, 2020-06-29:
Lots of 503 (permission errors) responses there.  Can you export and post settings.json (octoprint settings->Octolapse->export settings)?  I can review it tomorrow.

bjhiltbrand, 2020-06-29:
[settings.zip](https://github.com/FormerLurker/Octolapse/files/4848353/settings.zip)

Apparently, github won't let you upload json files on their own, please unzip to find the file you're after. Thank you so much for being so responsive. I really appreciate it. Have a good night.

FormerLurker, 2020-06-29:
For future reference, upload these to gist.github.com.  thanks!