FormerLurker/Octolapse #427: Camera

samd12, 2020-01-10:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.12 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Installed Octolapse
2. Test camera, came back OK
3. Noticed no camera view
4. Tried to hit Play button

#### What should have happened?
Video should be seen in box.  I can see video in the control tab and when I go to octoprint.local:8080 as well as a snapshot picture so camera is working otherwise
[image]
[image]

#### What happened instead?
Was taken back to main octoprint tab, octolapse closes

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-01-10:
Sorry, that area doesn't show the camera stream, but rather the most recent snapshot taken by octolapse.  If you haven't ran a print with octolapse enabled, there is no most recent snapshot to display.

The play button a preview of your timelapse (most recent 5 frames by default i believe), but is somewhat broken in v0.3.4.  I have made this much more clear in v0.4, and have fixed the play button too.

FormerLurker, 2020-02-07:
Since I believe your issue is related to poor UX regarding the function of the latest snapshot view in Octolapse, and I believe it's been taken care of in V0.4 (though I'd be VERY happy if you would like to try it out and let me know what you think.. let me know), I'm going to go ahead and close this issue.  I will reopen it if you want me to.

FYI, here is what the Octolapse tab will look like in the next release:

[image]

I'm considering adding the ability to switch back and forth from the live camera stream to the most recent snapshot.

closed 2020-02-07 by FormerLurker