FormerLurker/Octolapse #428: dev3 update fail?

keawepoo, 2020-01-13:
#### Version of Octolapse
Octolapse Version: Octolapse 0.4.0rc1.dev3

#### Version of OctoPrint
OctoPrint Version: Version 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
1. Update Plugin Manager
2. "Get more"
3.  "From URL"
4.  Paste in the given URL...

#### What should have happened?
Not error up... 

#### What happened instead?
 It forced me into rebooting my pi, and then it booted octoprint into safe mode. from there i had to reboot octo again to get it to work again... 

#### Operating System running OctoPrint and Octolapse
OctoPrint 1.3.12 running on OctoPi 0.16.0

#### Printer model & used firmware incl. version
Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser
Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json
Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/keawepoo/d809edd420282b64794caf30af93575d

#### Link to contents of Javascript console in the browser
Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2020-01-15:
I'm still looking into this, FYI.  As of now, I have no idea what would make it work after a reboot :(

FormerLurker, 2020-05-08:
So this seems to happen pretty much randomly.  Only a few users experience this and I've not been able to figure it out at all.  I'm going to close this for now and see if some OctoPrint updates fix it :(

Thank you for posting!  I hope you have tried 0.4.0rc1!  Rc2 dropping soon.

closed 2020-05-08 by FormerLurker