FormerLurker/Octolapse #429: In Stabilization dropdown I got duplicates and one item was removed

stefan57, 2020-01-13:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0RC1.DEV2

#### Version of OctoPrint

OctoPrint Version: 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. I was watching the Youtube videos regarding what was coming in the 0.4 version.
2. I created my printer Tronxy XY-2 Pro printer profile
3. I selected "Fixed - Extruder at Back Left"
4. I clicked the Edit Pen
5. Since I'm new to this I edited some values, saved and then I went back to edit more, and there I selected Profile to Import: "Back Left"

I repeated 4 and 5 a few times, and suddenly I had one "Back Left" in the Stabilization dropdown. I selected that, clicked the pen to edit, probably selected profile to import again: "Back Left", and saved. And I repeated this a few times, not really knowing what I did.

In the Stabilization dropdown, I now I have 3 items saying "Back Left" and the "Fixed - Extruder at Back Left" is gone and missing.

I have now realized maybe I should not have done the Profile to Import: "Back Left" but at the same time, since I did it, this problem showed up and hopefully can be fixed :-)

#### What should have happened?
I should not have got these duplicate "Back Left" items in the Stabilization dropdown, and the "Fixed - Extruder at Back Left" should not have disappeared...

#### What happened instead?
See above

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 10
Os Version: 1909

#### Printer model & used firmware incl. version

Printer Model: Tronxy XY-2
Printer Firmware Version: Marlin

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-01-13:
This is a known issue with 0.4.0rc1.dev2.  There were some incorrect GUIDs shipped with the default settings, unfortunately, leading to problems updating and removing.  I recommend deleting the offending stabilizations, and re-importing them.

This is fixed in dev3, I believe.  Except for memory problems when downloading large files (work on that now) dev3 is nearly ready to rock if you want to update from the devel branch.

stefan57, 2020-01-15:
Thanks, I deleted the bad items in the list, and then reinstalled the default, and all looks fine again.

 

Any idea when dev 4 will be available?

 

/Stefan

FormerLurker, 2020-01-15:
Glad you got it fixed.  That error with the bad profile GUIDs was a simple copy/paste error, and was a bit embarrassing :)

I've been furiously working on the next release.  Little things keep popping up that take time to deal with.  I'll post here when it is finished.

FormerLurker, 2020-02-02:
I'm going to go ahead and close this issue since your original issue was solved.  Prepping v0.4.rc1.dev3 for official release.  FYI, I change some settings (renamed debug profiles to 'logging' and moved test mode to the main settings to reduce confusion), and since you are already on dev 3, your migration won't run when you upgrade.  It shouldn't be a problem really, but you could either restore the defaults, or just import the new logging profiles to remedy the issue.

closed 2020-02-02 by FormerLurker